solvespace/solvespace #295: Circle rendering is very conservative

Timmmm, 2017-09-28:
SolveSpace version: 2.3
Operating system: MacOS 10.13

When sketching a circle it seems like the rendering accuracy (i.e. how many line segments it is split into) depends on the physical size of the circle, rather than its on-screen size. The result is if you draw a small circle and zoom in it looks like this:

[image]

I.e. not very circular. Nobody is running this on a 386 so I would suggest as a simple hacky fix just increase the minimum number of segments from 8 to something a bit more reasonable like 128.

Timmmm, 2017-09-28:
Hmm so I had a look in the source code and discovered the chord tolerance setting, which obviously helps. Reducing it from 0.5% to 0.01% basically fixes this for circles down to 1mm diameter, which is probably about the limit that I imagine most people would care about.

To be honest I would have expected the code to work by representing all the geometry as true curves, and then turning them into polygons for display. In which case it should be easy (well "easy") to set the chord accuracy dynamically based on how big the object is on screen. Or in other words, the chord accuracy should be measured in screen-space pixels, not world-space mm.

You know more than me though so maybe it isn't that simple. :-)

I just checked how Blender handles this. It seems like it uses a fixed number of divisions of each bezier irrespective of their size. Also not ideal.

Timmmm, 2017-09-28:
I guess [this](https://github.com/solvespace/solvespace/blob/master/src/srf/ratpoly.cpp#L260) is the relevant code?

```
void SBezier::MakePwlInto(List<Vector> *l, double chordTol) const {
    if(EXACT(chordTol == 0)) {
        // Use the default chord tolerance.
        chordTol = SS.ChordTolMm();
    }
    l->Add(&(ctrl[0]));
    if(deg == 1) {
        l->Add(&(ctrl[1]));
    } else {
        // Never do fewer than one intermediate point; people seem to get
        // unhappy when their circles turn into squares, but maybe less
        // unhappy with octagons.
        MakePwlInitialWorker(l, 0.0, 0.5, chordTol);
        MakePwlInitialWorker(l, 0.5, 1.0, chordTol);
    }
}
```

Maybe as a not-ideal fix it could just be changed so that the minimum number of segments is more than 2.

jwesthues, 2017-09-28:
Ideas:
1. Adjust default parameters until you have an equal number of complaints about speed vs. too few pwl segments.
    - But I think you get more about speed already?
2. Just make everything faster, duh.

You're obviously doing lots of little things that help (2). I don't see any one easy big thing, or else we'd have done it already...

@Timmmm This is the same tolerance used for the NURBS Boolean operations, so the computation isn't trivial. We could make the tolerances different for entities vs. the mesh, but then e.g. entities wouldn't line up with the edges of the mesh they were extruded to form.

Evil-Spirit, 2017-09-28:
I don't think if we should do something with this.
On the one side is the precision of visualization on the other is performance. We can increase minimum segments to 16, but I am not sure but what if we need to create a big array of cylindric shapes e.g. chip radiator. The other thing we can do - is use relative(fraction) tolerance from each bezier primitive bound, not for bound of whole model.

whitequark, 2017-09-28:
> The other thing we can do - is use relative(fraction) tolerance from each bezier primitive bound, not from bound of whole model.

This is worth exploring, I think.

Timmmm, 2017-09-29:
How does other CAD software handle this? I don't recall Solidworks or NX ever getting close to an octagon, even for 1mm circles.

(Sorry if it sounds like I'm complaining btw - this is by *far* the best open source CAD software in existence, and better in a few ways than the big commercial ones.)

whitequark, 2017-09-29:
I think they indeed calculate tolerance relative to primitive bound. The more I think about it, the more I become convinced that this is the only reasonable option.

@jwesthues, any opinion?

jwesthues, 2017-09-30:
Commercial CAD systems generally do a fixed chord tolerance, either absolute (like for export) or as a function of model size and/or zoom level and stuff. They presumably have lots of special case heuristics on top of that. If you've never seen the PWLs there, then I'd guess it's because they're just faster, making better use of the GPU, selectively re-meshing, etc. So they can set a finer tolerance without getting too slow. All of that could be done in SolveSpace, but it's a lot of work.

What's the rationale behind tolerance as a fraction of primitive bound? Like, if I represent the exact same geometric curve as one Bezier piece vs. two, why do I want finer PWLs for the two case? A single chord tolerance for the entire model is at least internally consistent. Note that the Booleans also use a chord tolerance, and rely on that geometric consistency.

whitequark, 2017-09-30:
Ack.

This seems a somewhat frequent UI papercut. What do we do here? Teach it better? Add some sort of heuristic that acts when you place the first sketch?

jwesthues, 2017-09-30:
Add a "chord angle" tolerance, on the maximum angle between the curve and the chord approximating it? So if you want big things meshed finer, then you make the chord distance tolerance smaller, but if you want little things meshed finer then you make the angle finer. That's how SolidWorks export does it, for example.

whitequark, 2017-09-30:
That seems very elegant.

jwesthues, 2017-09-30:
And easy to implement--just modify `MakePwlWorker` to compute some tangents and check the angle in the same place it checks distance. To make the Booleans consider angle tolerance would be a bigger project; but that's relevant only in cases where the chord distance tolerance is so big that the Booleans are failing entirely, and you can always fix that by dropping the chord distance tolerance, same as now.

jolshefsky, 2019-05-20:
For what it's worth, this also occurs during image export which is highly undesirable. My specific instance is to make a lineart isometric diagram of a small part; all the circles render as octagons even when >10% the size of the whole page.

baryluk, 2020-01-16:
Hi,

this is always a first thing I change in the freshly compiled, first time run solvespace. Then I always need to play with the max segments and tolerances. Relative tolerances would be way better I think.

I usually change my segments settings to 32 or 100, and only back it down to lower number if performance suffers, which is almost never.

8 is really super low.

Timmmm, 2020-03-28:
This is by far my biggest issue with Solvespace at the moment, and Solvespace is by far the best open source CAD software, so I had a look at the code to understand this more. This is on `master` as of a week or so ago.

## How it currently works

1. Initial default settings are loaded in `solvespace.cpp`:

```
    // Chord tolerance
    chordTol = settings->ThawFloat("ChordTolerancePct", 0.5);
    // Max pwl segments to generate
    maxSegments = settings->ThawInt("MaxSegments", 10);
    // Chord tolerance
    exportChordTol = settings->ThawFloat("ExportChordTolerance", 0.1);
    // Max pwl segments to generate
    exportMaxSegments = settings->ThawInt("ExportMaxSegments", 64);
```

(By the way is there a GUI for these settings? I can't see a way to edit them...)

2. When generating PWL (piecewise-linear) curves from beziers and circles (which are actually just 4 beziers), it calls `SolveSpaceUI::GenerateAll()` in `generate.cpp`. It calls it twice - first with `genForBBox = true` to get the bounding box for the entire scene, and then with `genForBBox = false` to generate the actual PWLs.

3. The second time it is called, it runs this code:

```
    // If we're generating entities for display, first we need to find
    // the bounding box to turn relative chord tolerance to absolute.
    if(!SS.exportMode && !genForBBox) {
        GenerateAll(type, andFindFree, /*genForBBox=*/true);
        BBox box = SK.CalculateEntityBBox(/*includeInvisibles=*/true);
        Vector size = box.maxp.Minus(box.minp);
        double maxSize = std::max({ size.x, size.y, size.z });
        chordTolCalculated = maxSize * chordTol / 100.0;
    }
```

So `chordTol` is the chord tolerance as a percentage of the bounding box size, and `chordTolCalculated` is the absolute chord tolerance in mm. This has the strange effect of changing the tolerance of some entities depending on what else is in the scene. For example, small circle inside a large circle:

[image]

Same small circle but with the large circle deleted: 

[image]

4. The calculated absolute `chordTol` is passed all the way down to `SBezier::MakePwlWorker()` in `ratpoly.cpp` where the actual check is:

```
    double d = pm.DistanceToLine(pa, pb.Minus(pa));

    double step = 1.0/SS.GetMaxSegments();
    if((tb - ta) < step || d < chordTol) {
```

Here it also checks the max segments.

Also, when exporting it just returns `exportChordTol` which is in mm, divided by `exportScale`:

```
double SolveSpaceUI::ChordTolMm() {
    if(exportMode) return ExportChordTolMm();
    return chordTolCalculated;
}
double SolveSpaceUI::ExportChordTolMm() {
    return exportChordTol / exportScale;
}
```

## Problems with this approach

1. The absolute chord tolerance depends on the size of your scene, which makes it behave in an unintuitive way.
2. Sometimes you really do have a large scene with small features that you care about. For example if I am CNCing some holes in a tabletop, it doesn't matter that the table is huge - I still want my small features to look nice when zooming in on them.

## Possible solutions

We have the existing constraints:

1. Maximum number of linear segments per bezier segment.
2. Distance tolerance, proportional to the scene size.

There are some other constraints that are possible.

3. Angle tolerance (as mentioned above).
4. Distance tolerance that is proportional to the size of the current bezier segment.
5. Absolute distance tolerance (like for export).

My thoughts on the constraints are:

1. Drop this. It means circles are never very smooth, even if they are huge and are well outside the distance tolerance.
2. Drop this too. It's just confusing that tolerance depends on the size of your scene, even if it sort of vaguely makes sense.
3. I think angle tolerance makes quite a lot of intuitive sense - it's the sharp corners that really look bad in the preview. But it might be hard to do just on its own because handling sharp beziers could be tricky (you'd end up with a load of tiny linear segments at the sharp corner). You could possible do something fancier, like a tolerance on the cross product of consecutive line segments?
4. This seems like it might work to me - if you've drawn small features you probably want the tolerance to be proportional to those features. This is kind of how actual manufacturing tolerance works too.
5. I think this would be useful as an additional constraint.

All these constraints can be combined in a boolean expression. Maybe one day the user could pick it. Anyway I am going to experiment with some different ones - I'll start with 4.

Timmmm, 2020-03-28:
Oh, also if you don't use a constraint that is proportional to the scene size, you don't need to call `GenerateAll(genForBBox=true)`.

whitequark, 2020-03-28:
I think your suggestion (4) makes a lot of sense. I'd also ask @jwesthues about it.

jwesthues, 2020-03-28:
You are correct that the max line segment count per Bezier segment (1) doesn't make any geometric sense. I intended it just as a sanity check, to stop the regeneration from becoming unusably slow if the user chooses a very fine distance tolerance, especially since there's no way to interrupt that. If you haven't hit such problems, then I suspect you don't make big solid models and would be happy greatly increasing (1) from the default.

Per above, distance as a fraction of Bezier length/bounding box/whatever (4) also doesn't make geometric sense--the Beziers will get chopped up by Boolean operations, and it doesn't make sense to represent the same physical curve finer just because it's now in two pieces. The Booleans assume a consistent distance tolerance, which is achieved only by (2) and (5). Perhaps (4) could replace line segment count as the sanity check, though it's perhaps more obvious to a user that "10000 segments" might get slow than that "0.0001 relative chord tolerance" will.

I'm not sure what you mean by a tolerance on the cross product? The magnitude of that has dimensions of length squared times angle, which does not seem useful. The combination of angle tolerance (for little things) and chord tolerance (for big things) is pretty common in professional CAD.

Evil-Spirit, 2020-05-10:
2. I spent so many time to implement this... 
[image]

whitequark, 2020-05-10:
@Evil-Spirit Your work has served us very well so far, and adding an angle chord tolerance would be building on top of it, not replacing it. Or am I missing something?

ruevs, 2020-05-10:
Personally I do not mnid tiny (compared to the model size) octagonal circles/holes at all. It is obvious to (this) user that it is just an optimization for the visual rendering. Real tiny octagons are "rare" (nuts are usually hexagonal :-) so there is little risk for confusion. As for visually checking alignment - constraints are for that.

So i would not trade "smooth tiny circles" for any performance loss. If it is possible to do it without performance loss AND - more importantly - very complicated (potentially buggy) code - then fine.

whitequark, 2020-05-16:
@phkahler This isn't really a mesh issue, is it? The Bezier curves from entities are turned into piecewise linear curves independently from the solid model.

baryluk, 2020-05-16:
@Timmmm and @Evil-Spirit Thank for putting a lot effort into your investigations.

I am not super familiar with solvespace internals, especially rendering, only some pieces of the solver and algebraic constraints, but I am pretty sure there is room for improvments, so the rendered can be more smart.

@ruevs octagonal ones bugs me a little bit. If it was 16-sided polygon I would be okish with it.

No commercial CAD program in the world really has pixel accurate rendering (which is actually very unfortunate), but I think there might be some interesting smarter options to look at.

How about when doing instantiation of the geometry for visualization, looking at the angles between consecutive line segments. And if it is too high (weighted by the zoom factor maybe?) backtrack and refine the approximation?

Another observation. People often edit the model, and then look at it, rotate, without changing any constraints. How about having a thread in a background asynchronously that do recalculations at higher level of detail and when it is ready update it. If new update comes in, cancel it.

Just some ideas.  I work with solvespace on desktop, with good CPU and good GPU, so I rearly see performance issues, but more interactive experience is worthwhile for everybody.

whitequark, 2020-05-16:
> How about when doing instantiation of the geometry for visualization, looking at the angles between consecutive line segments.

That was suggested in https://github.com/solvespace/solvespace/issues/295#issuecomment-333275504. It seems like a good route.

> How about having a thread in a background asynchronously that do recalculations at higher level of detail and when it is ready update it.

At the moment the internals of SolveSpace aren't well suited for parallel, asynchronous processing, so this is probably not viable.

Evil-Spirit, 2020-05-16:
Nurbs segments can be implemented on gpu. We can just create pre-baked mesh with fixed subdivisions (2,4,8..2^n). Parameters of curve can be passed to shader and curve points can be computed on gpu. Depending on screen size and type of primitive, we can choose a LOD (level of detil) from 2 to e.g. 1024. Rendering of curved primitives will be so fast as rendering regular lines

phkahler, 2020-05-16:
@whitequark  I believe the PWLs from the beziers ARE the edges of the polygons of the mesh. Changing them will also change the mesh.

I might be on board with bumping the minimum from 2 segment's to 4 for curves with degree higher than 1. That would make the tiniest holes 16 sided.

What is wrong with lowering chord tolerance until the small holes look good and setting the max segments to maintain performance?

whitequark, 2020-05-16:
> What is wrong with lowering chord tolerance until the small holes look good and setting the max segments to maintain performance?

The out of box experience we have is pretty poor and the need for fiddling with obscure settings isn't very nice either. The question of "why are all the circles polygonal" comes up very often with beginners IME.

whitequark, 2020-05-16:
Also, the other problem is that changing chord tolerance can make booleans fail, so if our out of box settings work well enough and people don't need to change them, it will reduce frustration with those.

Evil-Spirit, 2020-05-16:
> That would make the tiniest holes 16 sided.

I wrote about it. What if you want to make 256x256 cylinders? You need to look at it as fast as possible. BTW we can introduce "adaptive degradation". When we run at OK performance, we use small chord tolerance, if have a complex model and high time of computation, we just increase chord tolerance (for visualization only, ofc).  Anyway, export will work in precise way.

phkahler, 2020-05-16:
I think we need to revisit the defaults. Now that we have helix, a bunch of people are going to make screw threads just because they can ;-)  Those are going to have diameters down to a few mm.

It's also confusing to me that chord tolerance has a percent value and a mm value. Not sure how they are related. And the for export it's just the mm value, which makes more sense to me.

I'm not sure why the text window has a learning curve, but I do remember it taking me a while to really get it. The key there is that Blue text is clickable.

BTW: I just put the setting back to defaults now that I know what those are, and it's not bad. I'm really thinking raising the minimum segments from 2 to 4 (form octagon to 16-gon) will help with circles a lot. I mean, just draw a 3mm circle and press Shift-T  they turn into octagons very easily and it depends how far you spread them out. I don't know how common models are that would suffer from increasing circles from 8 to 16 segments minimum

phkahler, 2020-05-16:
My understanding was wrong. This change in SBezier::MakePwlInto() does bad things:

        MakePwlInitialWorker(l, 0.0, 0.25, chordTol);
        MakePwlInitialWorker(l, 0.25, 0.5, chordTol);
        MakePwlInitialWorker(l, 0.5, 0.75, chordTol);
        MakePwlInitialWorker(l, 0.5, 1.0, chordTol);

jwesthues, 2020-05-17:
If that's the exact change, I think the last line should be starting from 0.75 and not 0.5?

ruevs, 2020-05-17:
I was just about to write that it works just fine for me and was going to ask what "bad things" do you see :-) And it does - I simply did not make the copy-paste mistake ;-)
```
        MakePwlInitialWorker(l, 0.0, 0.25, chordTol);
        MakePwlInitialWorker(l, 0.25, 0.5, chordTol);
        MakePwlInitialWorker(l, 0.5, 0.75, chordTol);
        MakePwlInitialWorker(l, 0.75, 1.0, chordTol);
```
So 16 sided circles do look (twice as much :-D ) more like circles, and I think no one will complain about them. But then again so did Jonathan back when he wrote his comment ;-)
```
        // Never do fewer than one intermediate point; people seem to get
        // unhappy when their circles turn into squares, but maybe less
        // unhappy with octagons.
```
But do we want the performance penalty? I don't.

jwesthues, 2020-05-17:
An alternative would be to keep the current behavior for the solid model, but always render other curves to within some fraction of a display pixel. That would still have some cost in performance, but much smaller.

whitequark, 2020-05-17:
> But do we want the performance penalty? I don't.

I don't have such an absolutist view on performance. In particular, piecewise linear decomposition of beziers is cached these days (unlike the early implementation), so any penalty would only happen during regeneration. That said, *if* we take a performance hit, I would prefer a solution that renders circles exactly, perhaps using one of the techniques for rendering curves with a shader; Catmull-Rom splines, or [Keith Packard's take on them](https://keithp.com/blogs/more-iterative-splines/).

> but always render other curves to within some fraction of a display pixel

Wouldn't that mean recomputing Beziers each time you rotate the view? I think that'd be a pretty significant hit.

ruevs, 2020-05-17:
If I understand your proposal correctly I think this was discussed/proposed at some point in the past (I can not find the discussion). I think the complaint was that for circles represented by polygons with few sides it will be annoying that the edges of the "outline" and the "solid" will visibly not match and may cause even more confusion.

jwesthues, 2020-05-17:
@ruevs  Yeah, that's the ugly tradeoff.

@whitequark If you rendered non-solid-model stuff from Beziers to pixels in the GPU, then it would be really fast. Note that our current Beziers do represent circles exactly, since they're rational.

phkahler, 2020-05-17:
> If that's the exact change, I think the last line should be starting from 0.75 and not 0.5?

I think the "7" got deleted copying to here. I redid it and got the same strange result:
[image]

Nice curves but the extrusion is now square.

ruevs, 2020-05-17:
Ahh me too. I did not extrude my small circle :-) OK bad.
[image]

Evil-Spirit, 2020-05-18:
@whitequark 
> Wouldn't that mean recomputing Beziers each time you rotate the view? 

We just should choose LOD depend on size of primitive on screen (CPU side). Beziers computed on-a-fly inside shader - it is very fast for GPU (x10-x100 times faster than CPU) The correct way is use geometric shader, where we can generate as much segments as we need using chord tolerance (pixel size). But this approach will increase OpenGL version we use. If someone is common with OpenGL and CG, I can help in consulting to implement such feature.

whitequark, 2020-05-18:
@Evil-Spirit Looks like [we can actually use geometric shaders](https://caniuse.com/#feat=webgl2).

ruevs, 2020-05-18:
> 
> 
> @Evil-Spirit Looks like [we can actually use geometric shaders](https://caniuse.com/#feat=webgl2).

Well this usage of shaders will kill OpenGL1 and will make me sad. By default I still build and use (even with VS2019 under Windows 10 x64) the 32 bit OpenGL1 version. And at least once a month I build one with the XP tool chain of VS2019 for my laptop which has XP.

[Microsoft.VisualStudio.Component.WinXP](https://docs.microsoft.com/en-us/visualstudio/install/workload-component-id-vs-community?view=vs-2019) | C++ Windows XP Support for VS 2017 (v141) tools [Deprecated] | 16.1.28811.260
-- | -- | --

And it works just fine under Windows XP. From time to time I also build on my other XP machine with MinGW and (for now) that also works fine.

Timmmm, 2020-05-18:
I think "circles actually look like circles" should probably take precedence over "there's still one guy that wants to run this on a completely obsolete laptop", sorry Ruevs.

Evil-Spirit, 2020-05-18:
@ruevs don't be afraid, we already use shaders.

whitequark, 2020-05-18:
> Well this usage of shaders will kill OpenGL1 and will make me sad.

OpenGL 1 has been completely unsupported for a long time and produces incorrect renders. You're effectively building a fork of SolveSpace, even though the code happens to be upstream. You've been warned about this a long time ago, too.

If you want SolveSpace to run properly on Windows XP you should get ANGLE's DirectX 9 backend to build on the XP toolset. There's no inherent problem with it AFAIK, it's just looking for some DirectX 11 headers that are missing, which should be fixable with a few `#define`s. I briefly looked into it a long time ago but I cannot justify spending any time supporting an OS that has been long abandoned not only by its manufacturer but even by the last holdouts like Firefox.

whitequark, 2020-05-18:
That said, just like @Evil-Spirit says, starting to use geometric shaders will not directly impact the OpenGL 1 renderer, since it's separate from the OpenGL 3 renderer. It might after all be necessary to have a fallback in place for machines that only have DirectX 9 but not DirectX 11, or OpenGL 3 but not OpenGL 4. (How common are these? Does macOS provide geometric shaders via OpenGL, or only via Metal?)

It's just that I consider the OpenGL 1 renderer a pure liability and won't hesitate to remove it when it starts to cause any problems. That isn't going to happen until the 3.0 release, but is likely to happen the next time the rendering code needs to evolve.

ruevs, 2020-05-18:
Hi all,

Supporting XP has become eccentric - I understand. But I enjoy my machine booting to 165MB of RAM used. And as long as I have XP machines that are used and useful (right now I have two) I will keep a "fork" of SolveSpace that works on them.

I understand that `gl3shader.c/h` are not compiled when the OpenGL 1 renderer is configured and therefore any changes to shaders can not directly impact it. However the more that is done with shaders the more the  OpenGL 1 renderer breaks - as Whitequark already pointed out. That said I simply stated that this makes me "sad", I did not say "don't do it". It''s interesting why this annoys you so much.

Now back to the OpeGL **3** renderer and it's shaders. I know almost nothing about OpenGL programming but Google knows and from what I have read if you want to do
>
>
> Parameters of curve can be passed to shader and curve points can be computed on gpu.

This requires a [tessellation shader and not a geometric shader.](https://computeranimations.wordpress.com/2015/03/16/rasterization-of-parametric-curves-using-tessellation-shaders-in-glsl/) Tesselation shaders were [introduced in OenGL 4](https://www.khronos.org/opengl/wiki/Shader#Stages) and thus using them will require graphics drivers that support OpenGL 4.

It seems [Intel graphics](https://www.intel.com/content/www/us/en/support/articles/000005524/graphics-drivers.html) started [supprting OpenGL 4](https://en.wikipedia.org/wiki/List_of_Intel_graphics_processing_units#Gen7) in Gen7, which is Ivy Bridge. So for example people with a Sandy Bridge e.g [Core i7-2600K  4 cores/8 threads, 3.40 GHz/3.80 GHz Turbo, 8 MB  Cache](https://ark.intel.com/content/www/us/en/ark/products/52214/intel-core-i7-2600k-processor-8m-cache-up-to-3-80-ghz.html) will not be able to run SolveSpace. Really?!? 

And if a fallback is considered necessary anyway is it worth having the shader for performance, the fallback and the added complexity of "detecting and switching"?

whitequark, 2020-05-18:
> It's interesting why this annoys you so much.

There's nothing complex about that—the story of maintaining an open-source project is the story of dealing with boundless entitlement, passive aggression, and guilt trips from people benefiting from labor of volunteers. I maintain a lot more of them than you do (which is a wise choice on your part), am correspondingly exposed to more, and therefore have far lower tolerance for it.

> Tesselation shaders were [introduced in OpenGL 4](https://www.khronos.org/opengl/wiki/Shader#Stages) and thus using them will require graphics drivers that support OpenGL 4.

Yup, that's why I mentioned OpenGL 4 support.

> So for example people with a Sandy Bridge e.g [Core i7-2600K 4 cores/8 threads, 3.40 GHz/3.80 GHz Turbo, 8 MB Cache](https://ark.intel.com/content/www/us/en/ark/products/52214/intel-core-i7-2600k-processor-8m-cache-up-to-3-80-ghz.html) will not be able to run SolveSpace. Really?!?

Of course not. The reason I said that we "can" use geometric shaders is because the hardware and software support for them became *widely available*, in the sense that all of our target platforms, in principle, should be able to use them on recent hardware. The context you are probably missing here is that a few years ago, @Evil-Spirit suggested using OpenGL 4 features, and back then they were available only on select high-end hardware and on few platforms, so we *couldn't use them at all*, in contrast to the current situation. So at the time, I decided that we wouldn't spend any time trying to add OpenGL 4 dependent features at all, which has changed now.

> And if a fallback is considered necessary anyway is it worth having the shader for performance, the fallback and the added complexity of "detecting and switching"?

I don't understand how can you argue, in two statements that follow each other, simultaneously in favor of having a fallback OpenGL 1 renderer for machines without OpenGL 3, and against having a fallback OpenGL 3 renderer for machines without OpenGL 4.

So another reason why your attitude annoys me is that your position seems to be that the complexity that benefits you is necessary or at least tolerable, whereas the complexity that doesn't benefit you is a drawback. You're, of course, free to consider solely your personal convenience, but as a maintainer I don't have this option.

ruevs, 2020-05-18:
As for Angle - as we have discussed - I know it can work under XP - Firefox 52.9.0esr is a living proof. Maybe I will spend the time to figure out how one day.

ruevs, 2020-05-18:
@whitequark  "ACK" for all the technical points.
I apologize for annoying you. Last non-technical post from me:

> 
> I don't understand how can you argue, in two statements that follow each other, simultaneously in favor of having a fallback OpenGL 1 renderer for machines without OpenGL 3, and against having a fallback OpenGL 3 renderer for machines without OpenGL 4.

Just because the GL1 renderer exists, happens to still work OK and (as far as I am aware) causes no work or grief for anyone. So it is a zero overhead left-over that no one but me cares about and will bit-rot until removed.

> So another reason why your attitude annoys me is that your position seems to be that the complexity that benefits you is necessary or at least tolerable, whereas the complexity that doesn't benefit [...]

See above about the complexity. As for "my convenience" - I am not expecting or demanding anything from anyone - thus the "sad" statement above was useless and I should not have written it. I work on SolveSpace (less than I would like to) - when I have some time - because I like it, the architecture (mostly), the code, the UI - that's all.

Back to semi-technical arguments - [Ivy Bridge was  introduced in 2012](https://ark.intel.com/content/www/us/en/ark/products/codename/29902/ivy-bridge.html) so if using OpenGL 4 features and dropping OpenGL 3 (and 1) will make SolveSpace faster **and simpler** I think it is better to do it instead of adding "fallbacks".

It's not my intention to reduce the S/N ratio, thus this is my last non-technical post. But I reserve the "right" to annoy anyone and everyone with technical arguments if I think it's important :-D

whitequark, 2020-05-18:
@ruevs Thank you for being considerate, and I should note that I appreciate both your technical arguments and your other contributions. I do note that what counts as "too complex" is inherently social, and should be, perhaps, handled with more care than usually happens in technical discussions—but by no means I want to shut down that discussion either.

> if using OpenGL 4 features and dropping OpenGL 3 (and 1) will make SolveSpace faster **and simpler** I think it is better to do it instead of adding "fallbacks".

I think we are in agreement here: I also think that if dropping support for obsolete hardware simplifies the codebase, and if we can actually do it without impacting too many users (we'll never impact zero and that's just one of the unfortunate facts of life), we should do it. So the question is, what's the cost, and what's the impact?

We can never be sure of impact because we don't collect telemetry. We can only rely on hardware surveys by other projects, like Firefox or Steam, and those aren't necessarily representative.

Cost is a much easier consideration. I would not expect that a GPU-based Bezier renderer would result in major increase in complexity (wouldn't it be based on a pwl decomposition anyway?), but I could be wrong. In any case, that wouldn't require forking the renderer, and so it's much easier to test and maintain.

<hr>

While looking into this I discovered that WebGL 2 doesn't actually support geometry shaders, it only has transform feedback. So the question of whether we'll need a fallback if we use GPU-based Bezier rendering is moot: the answer is yes.

jwesthues, 2020-05-18:
I'm afraid we may be discussing this into exhaustion. But just to compound that, I think I see people doing Beziers just with vertex shaders, with a fixed number of line segments per Bezier? The fixed segment count is wasteful, but maybe the whole thing then gets fast enough we don't care, or we could use simple heuristics to choose from a small number of fixed segment counts?

Or just more line segments on the CPU of course (for loose curves only, not for the solid model). I suspect all of that is near-infinitely fast compared to the solid model.

whitequark, 2020-05-18:
Yeah, the discussion veered into geometry shaders without much consideration to whether we can already do it. I think an approach using a fixed segment count would be entirely acceptable while not requiring anything beyond existing OpenGL 3. I remember seeing quite a few options; besides the ones I've [mentioned earlier](https://github.com/solvespace/solvespace/issues/295#issuecomment-629859651) I think you can also do this with signed distance functions.

phkahler, 2020-05-19:
I'll just chime in again, and then I'm moving on to better things. IMHO:

1) We don't need geometry shaders. Drawing curves is not the bottleneck here.

2) If the curves are not rendered with the same number of segments as the surfaces they are attached to, we're going have all sorts of Z-buffer issues. And if those could somehow be fixed, people would complain about the other funny effects that will have.

3) When increasing the minimum from 2 to 4, why do the surfaces not follow?

4) the performance implications are not trivial. I have a design with a grid of holes. I made them square because SS choked on that many round holes. The performance is still bad with squares. This is not about tessellation, but about algorithmic complexity. There are areas in the code that have straight forward algorithms (good) that are not as efficient as they could be.

5) NURBS are easy. Trimmed NURBS are hard. NURBS booleans are even worse.

6) OP needs to lower chord tolerance for now.

whitequark, 2020-05-19:
> 2\. If the curves are not rendered with the same number of segments as the surfaces they are attached to, we're going have all sorts of Z-buffer issues. And if those could somehow be fixed, people would complain about the other funny effects that will have.

You're completely right, and I missed the Z-buffer implications. I suspect this rules out any solution that doesn't ultimately amount to a more sophisticated pwl decomposition.

whitequark, 2020-05-19:
> 4\. the performance implications are not trivial.

Not to disagree with you, but to refine this statement: once we tie the Beziers to the mesh, the performance implications become correctness implications as well. I've had a design with a combination of small holes and large holes, and it was not trivial to find a display chord tolerance that wouldn't cause booleans to fail. (Then again for export, but what we currently do for export is good.) 

This was after we tied display chord tolerance to a percentage of entity bounding boxes in 34a5d87011748dcf07943f29726a73cc04fbfaaf. I'd have to investigate but I suspect that is not sufficient in some cases, though it certainly makes things a lot better.

Evil-Spirit, 2020-05-19:
> I think I see people doing Beziers just with vertex shaders

I wrote about such approach - we can use it without geometrical shaders.

phkahler, 2020-05-19:
@jwesthues  Can you explain why changing circles to a minimum of 16 segments (4 per curve) results in the issues shown above by both @ruevs and myself?  Is there a mismatch between this and some other part of the code? and why does it not make squares from octagons? This seems important to understand for people working on the innards.

jwesthues, 2020-05-19:
@phkahler No idea without actually debugging, I'm afraid. What happens if you just modify the worker to keep recursing until `tb - ta <= 0.25`?

phkahler, 2020-06-02:
I found the problem with 4-segment arcs. At the top of function:

```c
void SCurve::RemoveShortSegments(SSurface *srfA, SSurface *srfB) {
    // Five, not two; curves are pwl'd to at least four edges (three points)
    // even if not necessary, to avoid square holes.
    if(pts.n <= 5) return;
```

This is my fix. It used to say "three, not two" because curves had minimum 2 segments "to avoid square holes".

Together with this change over in raypoly:

```c
        // unhappy with sixteen sides (formerly octagons).
        MakePwlInitialWorker(l, 0.0, 0.25, chordTol);
        MakePwlInitialWorker(l, 0.25, 0.5, chordTol);
        MakePwlInitialWorker(l, 0.5, 0.75, chordTol);
        MakePwlInitialWorker(l, 0.75, 1.0, chordTol);
```
We can have  all circles with minimum 16 segments (4 per curve). The function in curve.cpp was deleting short segments on the pwl, but only for one surface? Anyway, by forcing more edges that go below chord tolerance, that other function needs to be changed or it will delete edges.

 I have no idea if there are any negative consequences for doing this. The comment above RemoveShortSegments suggests that operation is necessary because short segments can break booleans. That suggests to me the possibility that some boolean errors may be caused by the early exit leaving small edges that break booleans. That would also mean that increasing the minimum edges to 4 there might make things worse (more early exits).

It seems like there should be a more elegant way to remove the short edges while not breaking booleans and not reducing circle edges. @jwesthues @ruevs @Evil-Spirit  Any ideas?

[9 more replies]

closed 2020-06-12 by phkahler