solvespace/solvespace #297: Suggest lowering chord tolerance when NURBS booleans fail

whitequark, 2017-09-30:
Here's the idea: NURBS booleans fail somewhat frequently, and this results in need for support. We could detect that, and show a screen, suggesting lowering chord tolerance (if that's the issue and we verified that it results in successful generation) or forcing to triangle mesh.

At least, we should indicate failures more prominently and explain what they are.

jonas-schievink, 2017-09-30:
+1, this is probably my biggest problem when using solvespace. I didn't even know chord tolerance affects boolean operations.

whitequark, 2017-09-30:
I'm the current maintainer, and I didn't realize it, too, so don't feel bad about it :)

rpavlik, 2017-12-16:
Is this why my attempts to fillet (by extrusion difference) one curved structure with another perpindicular one fails? Or is that just "Boolean ops between two nurbs surfaces is hard" in general?

whitequark, 2017-12-16:
Hard to say without at least a screenshot.

hokieengr, 2017-12-18:
Can someone upload a model to reproduce an example of this problem?

rpavlik, 2017-12-18:
Here's a sample of what I meant with my question, but as I've already decreased the chord tolerance, I think the issue I'm having is separate.

[image]
[image]
[image]

In any case, the zip contains the solvespace file that produces this output - it's a simplified re-creation of the scenario I was working on the other day when I commented.

[bug-repro.zip](https://github.com/solvespace/solvespace/files/1569707/bug-repro.zip)

phkahler, 2021-05-16:
Not sure if suggesting the user change the CT parameter is a good idea. It won't fix the issue in the model shown above. Keeping this open for now because of the NURBS issue shown.

phkahler, 2023-09-22:
I've been keeping the sketch above as a boolean test.  This one difference extrusion manages to hit many different bugs. I've had at least 2 or 3 fixes that helped this sketch a little bit, and it's still broken using master branch. As I see it the issues encountered are:

1) curve-curve intersection  (not creating a vertex there)
2) tangent surfaces (curved surface tangent at intersection in particular)
3) intersection of two curved surfaces
4) coincident surfaces

This will probably be one of the last broken sketches to get fixed because it fails in ALL the ways, but we will get there.

rpavlik, 2023-09-22:
glad it's useful - I have mostly just avoided doing these kinds of operations in my modeling, or saving them until the end so I can tick the "mesh first" box.

phkahler, 2026-11-03:
@rpavlik  FYIthis was still broken in v3.3 but there is a incoming batch of fixes and two of them were required to completely fix this. It is going to be in the regression tests ;-)

closed 2026-11-05 by phkahler