solvespace/solvespace #298: Cannot find "<OpenGL/GL.h>" when building on Mac OS X 10.11.x

u6006, 2017-10-24:
### System information

SolveSpace version: (e.g. 3.0~3dd2fc00; go to Help → About...)

Git checkout on 10/23/2017

Operating system: (e.g. Debian testing)

Mac OS X 10.11.x

### Expected behavior

Build should succeed

What should have happened?

### Actual behavior

Missing "<OpenGL/GL.h>"

[ 68%] Building CXX object src/CMakeFiles/solvespace.dir/render/rendergl.cpp.o
/Users/jingli/solvespace/src/render/rendergl.cpp:7:10: fatal error: 'OpenGL/GL.h' file not found
#include <OpenGL/GL.h>
         ^
1 error generated.
make[2]: *** [src/CMakeFiles/solvespace.dir/render/rendergl.cpp.o] Error 1
make[1]: *** [src/CMakeFiles/solvespace.dir/all] Error 2
make: *** [all] Error 2

What actually happened?

### Additional information

On my Mac, there's no "GL.h" in the found header dir by CMake "CMakeCache.txt:OPENGL_INCLUDE_DIR:PATH=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks/OpenGL.framework",  instead, it's "gl.h" (all in lowercases) 

I'm no MAC dev, so no idea if it's the case for all new versions of Mac OS X and Xcode.

It works after changing it to "gl.h"

$ git diff
diff --git a/src/render/rendergl.cpp b/src/render/rendergl.cpp
index 01ee8a5..38a8fbf 100644
--- a/src/render/rendergl.cpp
+++ b/src/render/rendergl.cpp
@@ -4,7 +4,7 @@
 // Copyright 2015-2016 whitequark
 //-----------------------------------------------------------------------------
 #ifdef __APPLE__
-#include <OpenGL/GL.h>
+#include <OpenGL/gl.h>
 #else
 #define GL_GLEXT_PROTOTYPES
 #include <GL/gl.h>

[100%] Building CXX object bench/CMakeFiles/solvespace-benchmark.dir/harness.cpp.o
[100%] Linking CXX executable ../bin/solvespace-benchmark
[100%] Built target solvespace-benchmark

For bugs, please attach a savefile that shows the problematic behavior.
You can attach `.slvs` files by archiving them into a `.zip` first.

u6006, 2017-10-24:
Didn't find this issue when building on another Mac with OS X 10.12.x, so it might be the combination of Xcode (v9) and OS X on that Mac. Considering this, I would say it need to be documented since some Macs might see this issue.

whitequark, 2018-03-13:
I think this is due to the new filesystem, which is case-sensitive (?)

whitequark, 2018-07-12:
Fixed in master, thanks!

closed 2018-07-12 by whitequark