solvespace/solvespace #299: nurbs

u6006, 2017-10-25:
### System information

SolveSpace v3.0-16540b1b built on top of Mac OS X 10.12.x & Xcode 9.x with dedicated Nvidia graphics adapter

SolveSpace version: (e.g. 3.0~3dd2fc00; go to Help → About...)

Operating system: (e.g. Debian testing)

### Expected behavior

What should have happened?

### Actual behavior

What actually happened?

Nurbs meshing failed (while triangle meshing works, however the file is too big in this mode)

It can be reproduced by
- sketching a rectangle and extruding it
- selecting a point and sketching in a new WS
- sketching another rectangle and extruding it
* nurbs meshing failed

- Builds v2.1, v2.2, & v2.3 have the same issue
- It did work on my another Mac without Nvidia graphic adapter
- I remember v2.1 worked on the same Mac before I upgraded it to Mac OS X 10.12.x

[image]

### Additional information

For bugs, please attach a savefile that shows the problematic behavior.
You can attach `.slvs` files by archiving them into a `.zip` first.

whitequark, 2017-10-25:
Sounds like your chord tolerance is set too low. Can you show me the file?

u6006, 2017-10-25:
Thanks whitequark! My bad. Yes, it's chord tolerance issue. It's wrongly set to 0%.

The other issue is: it will not take effect until I have reloaded the drawing after re-setting it. This is why I wrongly open this bug. It didn't work after setting to 5% previously without reload (How can I re-render after changing the global settings?).

So, it's not a bug.

closed 2017-10-25 by u6006

whitequark, 2017-10-25:
> The other issue is: it will not take effect until I have reloaded the drawing after re-setting it.

It did, it just didn't clear the error markings. You can press `Esc` to remove them.

u6006, 2017-10-25:
Thanks whitequark. Cool! Esc did the magic.