solvespace/solvespace #303: Tangential circles in a rectangle can break contour filling

Natronium, 2017-11-18:
### System information

SolveSpace version: 3.0~13695be0
Operating system: Gentoo Linux

### Expected behavior
The contour gets filled in

### Actual behavior
This happens:
[image]

Additionally, extruding the shape creates naked edges at the points where that same circle touches its neighbours. [image]

All of this is accompanied by the same error that whitequark pointed out in #296:
```
couldn't find an ear! fail
```
however, unlike #296, enabling 'force NURB surfaces to triangle mesh' doesn't fix the issue in either case.
### Additional information

The file in question: [tangent-circles-break-contour-fill.zip](https://github.com/solvespace/solvespace/files/1485089/tangent-circles-break-contour-fill.zip)

whitequark, 2017-11-19:
Yeah, needs a better triangulation algorithm.

Evil-Spirit, 2017-11-19:
Tangential should be handled as error like countour self-intersection. In case of tangential the result mesh will be non-manifold (this is bad for some reason)

whitequark, 2018-07-12:
> Tangential should be handled as error like countour self-intersection.

I agree. Do you think you can write the fix? I'm not good with numeric code...

Evil-Spirit, 2018-07-12:
This not look hard to me.

phkahler, 2020-06-25:
Initially could not reproduce using the attached .slvs file of similar construction.
[Tangent_Circle_Test.zip](https://github.com/solvespace/solvespace/files/4828794/Tangent_Circle_Test.zip)

But... It happens if the construction lines attached to the upper right circle-center are H and V constrained. In that configuration 4 Beziers end at a single point where the circles meet. Pinging @ruevs because he's got some interest in NURBS issues and this is a case where something 2D breaks down prior to any surface creation.

phkahler, 2020-08-14:
PR #675 only addresses the first part of this, so I'm not going to close this yet. The naked edges remain.

whitequark, 2020-08-15:
@phkahler Did you close this by accident?

phkahler, 2020-08-15:
@Whitequark Yes by accident, I thought I deleted the reference from the PR before I merged it but this still closed. It's still a good case of naked edge problems though.

phkahler, 2020-09-11:
I've submitted issue 696 which I believe covers the second problem reported here. I'd like to close this, as the title does not match the second issue which has nothing to do with the contour filling and everything to do with coincident edges.

commit bc4924ae (2020-08-14): Don't duplicate points when connecting contours with zero length bridge. Fixes #303
closed 2020-09-13 by phkahler