solvespace/solvespace #306: Naked edges only when exporting to triangle mesh

frafra, 2018-01-01:
### System information

SolveSpace version: 3.0~c6fc0125

Operating system: Fedora 26 Workstation x86_64

### Expected behavior

Naked edges check result should be consistent with the one used when exporting to triangle mesh.

### Actual behavior

Naked edges check gives 0 errors, but I cannot export to triangle mesh due to naked edges.

### Additional information

[naked-edges-mesh-bug.zip](https://github.com/solvespace/solvespace/files/1596418/naked-edges-mesh-bug.zip)

whitequark, 2018-01-02:
Not a bug. There are two chord tolerances, display chord tolerance and export chord tolerance. In your case, export chord tolerance causes naked edges to appear, which are promptly shown to you in export mode.

frafra, 2018-01-02:
Ops, I am sorry, thanks for your response. I am still learning how to use SolveSpace properly and I found these two messages incoherent. I fixed using bigger shapes when selecting union and difference (no naked edges errors in this case).

whitequark, 2018-01-03:
Yes, I think the errors should be improved.