solvespace/solvespace #310: Crashes on Mac OS High Sierra version 10.13.3 Beta (17D25b)

dwbird, 2018-01-04:
### System information

SolveSpace version: (e.g. 3.0~3dd2fc00; go to Help → About...)

Operating system: Mac OS 2.3

Should load and Run

Crashes when Started
[Solvespace dump.docx](https://github.com/solvespace/solvespace/files/1603175/Solvespace.dump.docx)

will not load and run

What actually happened?

### Additional information

For bugs, please attach a savefile that shows the problematic behavior.
You can attach `.slvs` files by archiving them into a `.zip` first.

dwbird, 2018-01-04:
Deleted libx.dylib from the package.   now loads and runs

whitequark, 2018-01-05:
`libx.dylib`? Do you mean `libz.dylib`?

dwbird, 2018-01-05:
Yes.     Typo

tcurdt, 2018-02-16:
I am seeing this
```
dyld: Symbol not found: _inflateValidate
  Referenced from: /System/Library/Frameworks/ImageIO.framework/Versions/A/Resources/libPng.dylib
  Expected in: /Applications/solvespace.app/Contents/MacOS/libz.dylib
 in /System/Library/Frameworks/ImageIO.framework/Versions/A/Resources/libPng.dylib
Abort trap: 6
```

whitequark, 2018-07-12:
SolveSpace now statically links all dependent libraries, which should fix this and any similar issues from now on. I've built a .dmg which you can download 
[here](https://github.com/solvespace/solvespace/files/2188410/solvespace.dmg.zip) for testing.

closed 2018-07-12 by whitequark

Foadsf, 2018-07-12:
@whitequark  maybe you can fix the homebrew cask formula too?

whitequark, 2018-07-12:
Sorry, I do not use Homebrew. (Actually, I don't use macOS at all other than for developing the SolveSpace port; I run it on a VM and it's ridiculously slow, so doing anything is a pain...)

Foadsf, 2018-07-12:
@whitequark I could fix it if I could find the HomeBrew cask ruby formula!
It is strange that when I search in the commandline the solvespace formula exists but I can't find via google. It shouldn't be difficult to add a delete command in the end.

Foadsf, 2018-07-12:
found the ruby formula: `https://github.com/Homebrew/homebrew-cask/blob/master/Casks/solvespace.rb`

It uses this address `https://github.com/solvespace/solvespace/releases/download/v#{version}/solvespace.dmg` , if you give me the new address I can update it.

whitequark, 2018-07-12:
There have been no releases for a good while, and I'm afraid won't be for a few months as the master branch has major ongoing refactoring.

Foadsf, 2018-07-12:
@whitequark If you give me the latest source code I will try compiling and give you feedback.

whitequark, 2018-07-12:
Please follow the [instructions in README](https://github.com/solvespace/solvespace#building-on-macos).