solvespace/solvespace #311: Crash when using wrong values for angles

frafra, 2018-01-09:
### System information

SolveSpace version: 3.0~c6fc0125

Operating system: Fedora 26 Workstation x86_64

### Expected behavior

SolveSpace should not show the "not a valid number of expression" display dialog error when using wrong values for angles, like it does with wrong values for line length.

### Actual behavior

SolveSpace crashes.

```
File /home/frafra/Downloads/solvespace/src/expr.cpp, line 342, function Eval:
Assertion 'Not supported yet' failed: ((false) == false).
Backtrace:
 0: /home/frafra/Downloads/solvespace/build/bin/solvespace(_ZN10SolveSpace14assert_failureEPKcjS1_S1_S1_+0x4d) [0x47ad2d]
 1: /home/frafra/Downloads/solvespace/build/bin/solvespace(_ZNK10SolveSpace4Expr4EvalEv+0x249) [0x4b0a29]
 2: /home/frafra/Downloads/solvespace/build/bin/solvespace(_ZN10SolveSpace14GraphicsWindow15EditControlDoneEPKc+0x1a8) [0x4fb138]
 3: /home/frafra/Downloads/solvespace/build/bin/solvespace(_ZN4sigc8internal9slot_callINS_18bound_mem_functor1IvN10SolveSpace17GraphicsWindowGtkEN4Glib7ustringEEEvJS6_EE7call_itEPNS0_8slot_repERKS6_+0x30) [0x467700]
 4: /home/frafra/Downloads/solvespace/build/bin/solvespace(_ZN10SolveSpace13EditorOverlay11on_activateEv+0xaf) [0x469cbf]
 5: /lib64/libglibmm-2.4.so.1(_ZN4Glib17SignalProxyNormal19slot0_void_callbackEP8_GObjectPv+0x28) [0x7fef447c3a38]
 6: /lib64/libgobject-2.0.so.0(g_closure_invoke+0x19d) [0x7fef413ff30d]
 7: /lib64/libgobject-2.0.so.0(+0x217ed) [0x7fef414117ed]
 8: /lib64/libgobject-2.0.so.0(g_signal_emitv+0x280) [0x7fef41419710]
 9: /lib64/libgtk-3.so.0(+0x116655) [0x7fef45023655]
10: /lib64/libgtk-3.so.0(+0x116b10) [0x7fef45023b10]
11: /lib64/libgtk-3.so.0(+0x116ca0) [0x7fef45023ca0]
12: /lib64/libgtk-3.so.0(gtk_bindings_activate_event+0x99) [0x7fef45024e79]
13: /lib64/libgtk-3.so.0(+0x1a3d0d) [0x7fef450b0d0d]
14: /lib64/libgtk-3.so.0(+0x22f3b7) [0x7fef4513c3b7]
15: /lib64/libgobject-2.0.so.0(+0xf450) [0x7fef413ff450]
16: /lib64/libgobject-2.0.so.0(g_signal_emit_valist+0x739) [0x7fef41419e69]
17: /lib64/libgobject-2.0.so.0(g_signal_emit+0x8f) [0x7fef4141ab0f]
18: /lib64/libgtk-3.so.0(+0x379f24) [0x7fef45286f24]
19: /lib64/libgtkmm-3.0.so.1(_ZN3Gtk6Widget5eventEP9_GdkEvent+0x15) [0x7fef4607ec55]
20: /home/frafra/Downloads/solvespace/build/bin/solvespace(_ZN10SolveSpace13EditorOverlay18on_key_press_eventEP12_GdkEventKey+0x37) [0x4679f7]
21: /lib64/libgtkmm-3.0.so.1(_ZN3Gtk12Widget_Class24key_press_event_callbackEP10_GtkWidgetP12_GdkEventKey+0x54) [0x7fef46079084]
22: /lib64/libgtk-3.so.0(+0x22f3b7) [0x7fef4513c3b7]
23: /lib64/libgobject-2.0.so.0(+0xf546) [0x7fef413ff546]
24: /lib64/libgobject-2.0.so.0(g_signal_emit_valist+0x739) [0x7fef41419e69]
25: /lib64/libgobject-2.0.so.0(g_signal_emit+0x8f) [0x7fef4141ab0f]
26: /lib64/libgtk-3.so.0(+0x379f24) [0x7fef45286f24]
27: /lib64/libgtk-3.so.0(+0x22c342) [0x7fef45139342]
28: /lib64/libgtk-3.so.0(gtk_main_do_event+0x800) [0x7fef4513b3b0]
29: /lib64/libgdk-3.so.0(+0x35685) [0x7fef44c4c685]
30: /lib64/libgdk-3.so.0(+0x66bb2) [0x7fef44c7dbb2]
31: /lib64/libglib-2.0.so.0(g_main_context_dispatch+0x157) [0x7fef44298247]
32: /lib64/libglib-2.0.so.0(+0x4a5e8) [0x7fef442985e8]
33: /lib64/libglib-2.0.so.0(g_main_loop_run+0xc2) [0x7fef44298902]
34: /lib64/libgtk-3.so.0(gtk_main+0x85) [0x7fef4513a4e5]
35: /lib64/libgtkmm-3.0.so.1(_ZN3Gtk4Main3runERNS_6WindowE+0x126) [0x7fef46013486]
36: /home/frafra/Downloads/solvespace/build/bin/solvespace(main+0x40a) [0x45805a]
37: /lib64/libc.so.6(__libc_start_main+0xea) [0x7fef42a9988a]
38: /home/frafra/Downloads/solvespace/build/bin/solvespace(_start+0x2a) [0x459e4a]
Aborted (core dumped)
```

ruevs, 2021-04-07:
I think this is fixed. Can anyone reproduce it on Linux? @phkahler ?

ruevs, 2022-07-05:
@rpavlik @ppd @phkahler does this still fail on Linux?

phkahler, 2022-07-05:
@ruevs what does it mean? How do I even try to reproduce it?

rpavlik, 2022-07-05:
yeah I don't know what "wrong angles" even are. "d2" seems wrong, it responds that it's invalid, which seems right to me.  -740 got turned into 740, which is more than 360 but not inherently wrong. (though the loss of the negative without changing the direction of the angle is interesting) 

The issue report says it should *not* show the dialog but I think that's a typo, the dialog seems to be the right thing to show and it shows up fine.

closed 2022-07-06 by rpavlik