solvespace/solvespace #313: Constraint point on a Bezier curve

frafra, 2018-01-20:
### System information

SolveSpace version: 3.0~13695be0

### Expected behaviour

Constraint point on a Bezier curve should work.

### Actual behaviour

It doesn't, but it would be a nice feature :-) It works on a circle, but in order to build an ellipsis I have to use Bezier curves and I can't place a point on it.

(typo in the default new issue test: behavior -> behaviour)

Evil-Spirit, 2018-01-21:
for more information, read here https://github.com/whitequark/solvespace/issues/52

fmagin, 2021-05-17:
The https://github.com/whitequark/solvespace repository doesn't exist anymore, and it's unclear if this issue is replicated/archived somewhere. I assume this is a non trivial thing to model as constraints? Or is this just a case of no one being motivated to implement it yet?

kk6mrp, 2021-05-17:
You can read the original issue here: http://web.archive.org/web/20200912191033/https://github.com/whitequark/solvespace/issues/52

phkahler, 2021-05-17:
Part 1:  The cubic spline with 4 points exists now, and that shouldn't be too hard to implement.  To constrain a point we'd need to create a new free parameter t and calculate the x,y,z equations for a point on the spline as a function of t. Then set those expressions equal to the point being constrained.  It is true that the point will be able to go past the end of the spline, much line point on line segment can go past the end of the segment. Keeping it within the span of the control points would require implementing "ranged constraint" which is something @Evil-Spirit has done already (not merged) and limiting that t-parameter to the range 0-1.  This still seems like a good idea even without the range limits.

Part 2:  Constraining to the larger splines is hard. I haven't looked at the implementation yet, but I believe they are internally represented as piecewise cubic splines. The problem there is having the solver switch from one (spline) set of constraints to another as the point is moved around what the user sees as one curve. The easy 4-point cubic spline implementation is probably a prerequisite to doing this, if it ever gets done at all.

If anyone wants to attempt part 1 above I can throw out a bit more information. You'll need to write a function in expr.cpp equivalent of the first function in the file /surf/ratpoly.cpp   then understand how to use the resulting expressions in constraint equations.

AjinkyaDahale, 2022-08-21:
Apologies in advance if this is an inappropriate place to post this.

I'm trying to implement this constraint in another solver, planeGCS within FreeCAD, and am looking for sponsorship for the same. There may be mathematical similarities with implementation for solvespace. Interested people can track at: https://forum.freecadweb.org/viewtopic.php?f=9&t=71130.

phkahler, 2022-09-29:
@Evil-Spirit  That is a good point about combining multiple sketch entities into a single "curve" for constraint.  A single curve needs to be composed of N entities and some entities are composed of M piecewise rational spline segments. Somehow this set of splines will need to be tracked so the constrained point can move between them.

AjinkyaDahale, 2022-10-13:
I managed to get this done in FreeCAD Sketcher. Here's the code if you're interested: https://github.com/AjinkyaDahale/FreeCAD/tree/sketcher-point-on-bspline. You primarily want to look at the changes in `...planegcs/Constraints.cpp`. I don't know enough about SolveSpace to code it in, but I'd be happy to help with the math.