solvespace/solvespace #314: Very slow rendering during moving parts manually in 3d space

Symbian9, 2018-01-24:
### System information

SolveSpace version: v2.3 (release)

Operating system: Debian stable

### Expected behavior

Normal rendering inside main window without lags during moving parts.

### Actual behavior

Very slow rendering inside main window for assembly + very slow during moving manualy free part of assembly.

As result, CPU & RAM overhead.

### Additional information

- Forum thread - http://solvespace.com/forum.pl?action=viewthread&parent=2148
  1. SLVS (testcase, buggy) - http://solvespace.com/forum.pl?action=attachment&id=2148
  2. SLVS (optimized & fixed) - http://solvespace.com/forum.pl?action=attachment&id=2197

Evil-Spirit, 2018-01-24:
> Very slow rendering inside main window for assembly + very slow during moving manualy free part of assembly.

You can't know about this. No one can know what's really hapens here without debugging/profiling. Just write what you do and what you get. Not what you think about it. Just step-by-step. And again, I can't know what you actually do to achieve this result. For my PC it can not be so critical as for yours, so you should describe the actions, and the moment when you get wrong behaviour or lags.

Symbian9, 2018-01-24:
> You can't know about this.

I just tell my feeling. Nothing more.

You should look inside fixed SLVS -- and fix SolveSpace, if possible. ;-)

Evil-Spirit, 2018-01-24:
@Symbian9 
https://testlio.com/blog/the-ideal-bug-report/
[image]

Evil-Spirit, 2018-01-24:
> You should look inside fixed SLVS -- and fix SolveSpace, if possible. ;-)

I actually do this and don't see any problems. You should just do step-by-step instruciton because I can't read your mind.

Symbian9, 2018-01-24:
> Just write what you do and what you get. Not what you think about it. Just step-by-step.

I just open "buggy SLVS" produced by *Eric Buijs* and found the way how reconstruct same assembly without lags ("fixed SLVS")

I'm not an author of "buggy SLVS", details about it described on linked forum thread.

Evil-Spirit, 2018-01-24:
@Symbian9 
Sorry, this not helps. You have solved the problem by hacking slvs files (and spend a much time), but you just can't write a few words about how I can reproduce the wrong behaviour using my computer and my feelings.

Symbian9, 2018-01-24:
*Eric Buijs* wrote

> I created a case consisting of several parts in total. When I assemble these parts in Solvespace the program becomes very slow when I add the four feet (simple cylinders in this case)

Symbian9, 2018-01-24:
I will try add detailed step-by-step guide as soon, as posdible.

Symbian9, 2018-01-24:
> You have solved the problem by hacking slvs files

@Evil-Spirit, I'm not hcking original "buggy SLVS" file! I just fully reconstruct same geometry in other method using only SolveSpace without hacking it!

Evil-Spirit, 2018-01-24:
@Symbian9 can you check it with solvespace_ranged_constraints.exe? It seems I've already fixed it.

Evil-Spirit, 2018-01-24:
> I will try add own detailed step-by-step guide on reproducing bug as soon, as possible.

> just open "buggy SLVS" and try move cylinders or any other parts inside it.

^^^This is enough for understanding what I am doing wrong way. The problem was in opening it with the last version, where the problem was already fixed.

Evil-Spirit, 2018-01-24:
For getting more speedup you should always switch off edges
[image]

Evil-Spirit, 2018-01-24:
@whitequark 
Likely it was fixed in https://github.com/Evil-Spirit/solvespace-master/commit/76f81b42b42c981e1f3fe6dafef99fae74059613

Evil-Spirit, 2018-01-24:
>  I don't hack original "buggy SLVS" file!
> Just fully reconstruct same geometry in other method/order using only SolveSpace without hacking it!

I guess, you just don't forced booleans to triangle mesh and this solved the problem. Am I right?

whitequark, 2018-07-12:
@Evil-Spirit Thanks!

I've merged this in master.

closed 2018-07-12 by whitequark

blinkenlight, 2018-09-12:
Is there any binary for Windows anywhere that incorporated this fix, even just a nightly or a test build...? I'm trying to get something done with Solvespace (the last release 2.3) and I'm clearly having the same problem (dragging assemblies around very very slowly) making it unusable...

Evil-Spirit, 2018-09-13:
@blinkenlight try this https://www.dropbox.com/s/ycleiza57hng0o8/solvespace_ranged_constraints.exe?dl=0

blinkenlight, 2018-09-13:
@Evil-Spirit unfortunately this version works even worse for me; loading only my first assembly object and nothing else (the bottom half of a plastic box) and dragging it around, I get render times of 3200ms per frame; I can only approximate for the standard release 2.3 binary, but it looks slightly under a second per frame. This is only when dragging the box of course, if I don't move it and only pan the view I get perfectly fluid motion with render times under ten milliseconds. I'm testing this on Windows 7 SP1 x64 with a quad core AMD Phenom II X4 995 at 3.2GHz, 16GB RAM and an AMD RX480, loading a single object, as an assembly; the other machine I work with, a single-core AMD Turion x64 laptop, is obviously faring even worse. I have to admit I'm quite puzzled why moving an assembly - which should be a rigid object with nothing to compute at this stage - would place such a burden on the CPU. Anyway, thanks for the link and the attempt to help, much appreciated!