solvespace/solvespace #316: Crash at startup on Mac OSX 10.9 Mavericks (build with XCode 6.2)

ghoss, 2018-03-03:
### System information

SolveSpace version: current clone from GitHub as of 03-MAR-2018

Operating system: Mac OSX 10.9.5 Mavericks

### Expected behavior

Normal startup as with precompiled MAC OSX image, which works on 10.9.5

### Actual behavior

Immediately crashes at startup

### Additional information

This is the application's error log:

en
SolveSpace!

ERROR: 0:13: Attempt to redeclare 'tan' as a variable
ERROR: 0:27: Attempt to use 'tan' as a variable
ERROR: 0:28: Use of undeclared identifier 'norm'
ERROR: 0:28: Use of undeclared identifier 'norm'
ERROR: 0:28: Use of undeclared identifier 'norm'
ERROR: 0:29: Use of undeclared identifier 'norm'
ERROR: 0:36: Use of undeclared identifier 'perp'
ERROR: 0:37: Use of undeclared identifier 'norm'

File /Users/guido/Desktop/solvespace/src/render/gl3shader.cpp, line 121, function CompileShader:
Assertion 'Cannot compile shader' failed: ((compiled) == false).
Backtrace:
 0: 0   solvespace                          0x0000000104221c72 _ZN10SolveSpace14assert_failureEPKcjS1_S1_S1_ + 98
 1: 1   solvespace                          0x0000000104213e7d _ZN10SolveSpaceL13CompileShaderERKNSt3__112basic_stringIcNS0_11char_traitsIcEENS0_9allocatorIcEEEEj + 621
 2: 2   solvespace                          0x0000000104213a9a _ZN10SolveSpace6Shader4InitERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEES9_RKNS1_6vectorINS1_4pairIjS7_EENS5_ISC_EEEE + 42
 3: 3   solvespace                          0x00000001042158aa _ZN10SolveSpace12EdgeRenderer4InitEPKNS_12StippleAtlasE + 362
 4: 4   solvespace                          0x000000010421c8f1 _ZN10SolveSpace15OpenGl2Renderer8NewFrameEv + 49
 5: 5   solvespace                          0x00000001042f7fbe _ZN10SolveSpace10TextWindow5PaintEv + 318
 6: 6   solvespace                          0x0000000104213285 _ZN10SolveSpace11GlOffscreen6RenderEiiNSt3__18functionIFvvEEE + 309
 7: 7   solvespace                          0x000000010420b9c1 -[GLViewWithEditor drawRect:] + 209
 8: 8   AppKit                              0x00007fff924bb557 -[NSView(NSInternal) _recursive:displayRectIgnoringOpacity:inGraphicsContext:CGContext:topView:shouldChangeFontReferenceColor:] + 1082
 9: 9   AppKit                              0x00007fff924bb00d __46-[NSView(NSLayerKitGlue) drawLayer:inContext:]_block_invoke + 186
10: 10  AppKit                              0x00007fff924bae03 -[NSView(NSLayerKitGlue) _drawViewBackingLayer:inContext:drawingHandler:] + 2297
11: 11  AppKit                              0x00007fff924ba4f8 -[NSView(NSLayerKitGlue) drawLayer:inContext:] + 108
12: 12  QuartzCore                          0x00007fff91851812 CABackingStoreUpdate_ + 2220
13: 13  QuartzCore                          0x00007fff91850f60 ___ZN2CA5Layer8display_Ev_block_invoke + 59
14: 14  QuartzCore                          0x00007fff91850f1c x_blame_allocations + 84
15: 15  QuartzCore                          0x00007fff91850a2b _ZN2CA5Layer8display_Ev + 1539
16: 16  AppKit                              0x00007fff924ba3c3 _NSBackingLayerDisplay + 235
17: 17  AppKit                              0x00007fff9249174b -[_NSViewBackingLayer display] + 811
18: 18  QuartzCore                          0x00007fff91850162 _ZN2CA5Layer17display_if_neededEPNS_11TransactionE + 590
19: 19  QuartzCore                          0x00007fff9184f8b1 _ZN2CA5Layer28layout_and_display_if_neededEPNS_11TransactionE + 35
20: 20  QuartzCore                          0x00007fff9184f33c _ZN2CA7Context18commit_transactionEPNS_11TransactionE + 236
21: 21  QuartzCore                          0x00007fff9184efd6 _ZN2CA11Transaction6commitEv + 388
22: 22  AppKit                              0x00007fff924b9652 -[NSView(NSLayerKitGlue) _drawRectAsLayerTree:] + 591
23: 23  AppKit                              0x00007fff924b8351 -[NSView _drawRect:clip:] + 422
24: 24  AppKit                              0x00007fff924b78c4 -[NSView _recursiveDisplayAllDirtyWithLockFocus:visRect:] + 1799
25: 25  AppKit                              0x00007fff924b7ca0 -[NSView _recursiveDisplayAllDirtyWithLockFocus:visRect:] + 2787
26: 26  AppKit                              0x00007fff924b7ca0 -[NSView _recursiveDisplayAllDirtyWithLockFocus:visRect:] + 2787
27: 27  AppKit                              0x00007fff924b7ca0 -[NSView _recursiveDisplayAllDirtyWithLockFocus:visRect:] + 2787
28: 28  AppKit                              0x00007fff924b5706 -[NSView _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 841
29: 29  AppKit                              0x00007fff924b4eb1 -[NSThemeFrame _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 314
30: 30  AppKit                              0x00007fff924b1e9f -[NSView _displayRectIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:] + 2828
31: 31  AppKit                              0x00007fff924912da -[NSView displayIfNeeded] + 1680
32: 32  AppKit                              0x00007fff924aaeb6 -[NSWindow _reallyDoOrderWindow:relativeTo:findKey:forCounter:force:isModal:] + 1972
33: 33  AppKit                              0x00007fff924aa460 -[NSWindow _doOrderWindow:relativeTo:findKey:forCounter:force:isModal:] + 786
34: 34  AppKit                              0x00007fff924aa0e0 -[NSWindow orderWindow:relativeTo:] + 162
35: 35  solvespace                          0x000000010428e292 _ZN10SolveSpace14GraphicsWindow4InitEv + 578
36: 36  solvespace                          0x000000010420529e _ZN10SolveSpace12SolveSpaceUI12AfterNewFileEv + 302
37: 37  solvespace                          0x0000000104204dbf _ZN10SolveSpace12SolveSpaceUI4InitEv + 3695
38: 38  solvespace                          0x000000010421242c main + 1132
39: 39  libdyld.dylib                       0x00007fff8c2465fd start + 1
Abort trap: 6

Evil-Spirit, 2018-03-03:
It seems shader haven't compiled because "tan" interpreted as tangent.

Evil-Spirit, 2018-03-03:
@ghoss  Can you open res/shaders/edge.vert and res/shaders/outline.vert and replace tan to shomething else?

ghoss, 2018-03-03:
@Evil-Spirit Thanks, you put me on the right track there. Replacing tan fixed the error and the application could launch. I still got the following warnings however:

```
WARNING: Could not find vertex shader attribute 'tan' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'tan' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'tex' to match BindAttributeLocation request.
WARNING: Output of vertex shader 'fragLoc' not read by fragment shader
WARNING: Could not find vertex shader attribute 'tan' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'tan' to match BindAttributeLocation request.
WARNING: Could not find vertex shader attribute 'tex' to match BindAttributeLocation request.
WARNING: Output of vertex shader 'fragLoc' not read by fragment shader
```

I fixed these warnings too by removing the (no longer used) references to tex and fragLoc in the shaders and in gl3shader.cpp. Now the application compiles and launches without errors or warnings.

I'd like to provide the fixes, but I'm not a contributor yet. Can I become a contributor, or should I simply send the changes to you?

whitequark, 2018-03-03:
I'd prefer to fix this myself so you don't need to sign a CLA (which we don't have yet).

ghoss, 2018-03-03:
@whitequark Ok, here's the git diff:

```
diff --git a/res/shaders/edge.vert b/res/shaders/edge.vert
index 2217ee8..d2084d5 100644
--- a/res/shaders/edge.vert
+++ b/res/shaders/edge.vert
@@ -7,7 +7,7 @@ const float feather = 0.5;
 
 attribute vec3 pos;
 attribute vec3 loc;
-attribute vec3 tan;
+attribute vec3 tanV;
 
 uniform mat4 modelview;
 uniform mat4 projection;
@@ -21,7 +21,7 @@ void main() {
     vec3 dir = vec3(modelview[0].z, modelview[1].z, modelview[2].z);
 
     // calculate line contour extension basis for constant width and caps
-    vec3 norm = normalize(cross(tan, dir));
+    vec3 norm = normalize(cross(tanV, dir));
     norm = normalize(norm - dir * dot(dir, norm));
     vec3 perp = normalize(cross(dir, norm));
 
diff --git a/res/shaders/imesh_point.frag b/res/shaders/imesh_point.frag
index ca887f7..3f9af78 100644
--- a/res/shaders/imesh_point.frag
+++ b/res/shaders/imesh_point.frag
@@ -9,7 +9,7 @@ uniform vec4 color;
 uniform float pixel;
 uniform float width;
 
-varying vec2 fragLoc;
+// varying vec2 fragLoc;
 
 void main() {
     // Rectangular points
diff --git a/res/shaders/imesh_point.vert b/res/shaders/imesh_point.vert
index bc92045..717d9a4 100644
--- a/res/shaders/imesh_point.vert
+++ b/res/shaders/imesh_point.vert
@@ -13,7 +13,7 @@ uniform mat4 projection;
 uniform float width;
 uniform float pixel;
 
-varying vec2 fragLoc;
+// varying vec2 fragLoc;
 
 void main() {
     // get camera vectors from modelview matrix
@@ -31,7 +31,7 @@ void main() {
     vertex += ext * loc.y * normalize(v);
 
     // write fragment location for calculating caps and antialiasing
-    fragLoc = loc;
+//  fragLoc = loc;
 
     // transform resulting vertex with modelview and projection matrices
     gl_Position = projection * modelview * vec4(vertex, 1.0);
diff --git a/res/shaders/outline.vert b/res/shaders/outline.vert
index 6bf1dfc..4b41f97 100644
--- a/res/shaders/outline.vert
+++ b/res/shaders/outline.vert
@@ -11,7 +11,7 @@ const float feather = 0.5;
 
 attribute vec3 pos;
 attribute vec4 loc;
-attribute vec3 tan;
+attribute vec3 tanV;
 attribute vec3 nol;
 attribute vec3 nor;
 
@@ -40,7 +40,7 @@ void main() {
                           (mode == EMPHASIZED_WITHOUT_CONTOUR && isTagged && !isOutline));
 
     // calculate line contour extension basis for constant width and caps
-    vec3 norm = normalize(cross(tan, dir));
+    vec3 norm = normalize(cross(tanV, dir));
     norm = normalize(norm - dir * dot(dir, norm));
     vec3 perp = normalize(cross(dir, norm));
 
diff --git a/src/render/gl3shader.cpp b/src/render/gl3shader.cpp
index 2ee8bb3..1807fed 100644
--- a/src/render/gl3shader.cpp
+++ b/src/render/gl3shader.cpp
@@ -453,7 +453,7 @@ void EdgeRenderer::Init(const StippleAtlas *a) {
         {
             { ATTRIB_POS, "pos" },
             { ATTRIB_LOC, "loc" },
-            { ATTRIB_TAN, "tan" }
+            { ATTRIB_TAN, "tanV" }
         }
     );
 }
@@ -638,7 +638,7 @@ void OutlineRenderer::Init(const StippleAtlas *a) {
         {
             { ATTRIB_POS, "pos" },
             { ATTRIB_LOC, "loc" },
-            { ATTRIB_TAN, "tan" },
+            { ATTRIB_TAN, "tanV" },
             { ATTRIB_NOL, "nol" },
             { ATTRIB_NOR, "nor" }
         }
@@ -927,7 +927,7 @@ void IndexedMeshRenderer::Init() {
         "shaders/imesh.vert", "shaders/imesh.frag",
         {
             { ATTRIB_POS, "pos" },
-            { ATTRIB_TEX, "tex" }
+//             { ATTRIB_TEX, "tex" }
         }
     );
     texShader.Init(
```

whitequark, 2018-07-12:
Fixed in master, thanks all!

closed 2018-07-12 by whitequark