solvespace/solvespace #317: Exported STEP file unreadable

AndyPhil, 2018-03-31:
### System information

SolveSpace version: 2.3
Operating system: MacOS High Sierra

### Expected behavior

Exported STEP file should be readable

### Actual behavior

STEP file was not readable by many sites

### Additional information

After completing the excellent "bracket tutorial", I tried loading the resulting "Export surfaces..." step file to the [Weerg  site](https://www.weerg.com/en/quote-and-order/193-aluminum-5083-t651-peraluman) which produce automated on-line quotes (very cool).  The part did not load and caused an error.  

Based on the statement that Solvespace STEP files have been successful in the past, I tried to submit incrementally more complex files to find out if something in particular causes the problem.    I redid the tutorial quickly and loaded an incomplete part and it loaded fine!  I went back to the part from the original tutorial and deactivated the later steps (so the product is just an angle with an inside radius: no gusset and no holes) and to my surprise this part still caused an error.  The two parts (try1.stp and try2.stp attached in zip) are similar (not identical) but only try1.stp causes an error.  

In trying to understand the difference I looked at the step code, but .... it is too hard to understand.  Last thing I did was to export the *.slt versions of these files in the hope that my slicer program would give me a hint of what is going on.  Indeed the stl model was also faulty for the first try and fine for the second.  The slicer program detected 12 non-manifold edges (see messages in lower right corner of two images below).  This may help understand what is going on.
[image]
[image]

[Archive.zip](https://github.com/solvespace/solvespace/files/1865582/Archive.zip)

whitequark, 2018-03-31:
OK, this is related to the issues with our NURBS backend. I'm afraid this may prove tricky to address as this likely not an issue with the STEP exporter itself.

AndyPhil, 2018-03-31:
Thanks anymway!

phkahler, 2020-08-19:
@AndyPhil is there any way you can try this again with the latest code in master? I am unable to reproduce the problem. SolveSpace doesn't show any bad edges, and I was able to load an exported .stl in Slic3r with no problems. The website that couldn't read the stp is foreign language to me, so can't test that.

AndyPhil, 2020-08-20:
Hi,

I am afraid that I don't have the files anymore.  Thanks anyway!

Andre

phkahler, 2020-08-20:
@AndyPhil the files are attached above.

closed 2020-10-07 by phkahler

ruevs, 2020-10-21:
I tried with this:
https://viewer.kisters.de/en/products/3dviewstation-desktop-version.html
And the exported STEP files from both try1 and try2 open and display just fine.

shful, 2024-04-15:
I've got another file causing such an issue in SolveSpace 3.1: An exported .STL file could not be read by two 3D-Printer sites.

After that, I tested the generated .STL files with the import function of FreeCAD.
Finally, I got a pair of files where one worked, while the other generated an invalid STL export.

Both differ only minimally in some 2D objects only. The defective file has an additional Sketch In Workplace and a Point and a Circle in it. (I expected the exported STL to be identical in both files because only 2D elements were added.)

Please tell me when the defective files would be useful.

ruevs, 2024-04-16:
@shful it is always interesting to see files that cause problems. Issue https://github.com/solvespace/solvespace/issues/738 has a long list of known ones. If you have time to look through them and find one that is similar - comment under it. Otherwise open a new one.

shful, 2024-04-16:
Thanks!  I'll investigate more, and then try to put the issue in the right place...