solvespace/solvespace #318: German Localization

ghoss, 2018-04-02:
I'm currently in the final stage of making a German localization for SolveSpace. Now this went beyond simply making a .po file with some translations, as hard-coded English strings and tabular formatting (padding with blanks) still exist all over the place in the code. I've made all of these localizable, including changes to the printf-style string padding and formatting where necessary. This also may be useful for others who wish to make a complete localized version of SolveSpace in their language.

If this is of any interest to you, I'd like to find out the best way to contribute these patches. Obviously, they are too extensive to be included as a patch snippet in an issue, as the majority of files are affected. I could publish them in my own fork, but I just thought I'd ask if there's an easier way first.

whitequark, 2018-04-02:
> I've made all of these localizable, including changes to the printf-style string padding and formatting where necessary. This also may be useful for others who wish to make a complete localized version of SolveSpace in their language.

This was a really unnecessary piece of work :/ All of the printf-style code is getting ripped out and replaced with external templates, so you'll have to do it over once that happens.

ghoss, 2018-04-03:
@whitequark Thanks for the feedback. Is there any place (like any roadmap, wishlist, or even the slightest casual mentioning in a file like CONTRIBUTING.md), where the planned change you mentioned is documented for the uninitiated among us to see? Otherwise, sorry if I stumbled into a closed user group by accident. Won't happen again; I thought this was open source.

closed 2018-04-03 by ghoss

whitequark, 2018-04-03:
> Is there any place (like any roadmap, wishlist, or even the slightest casual mentioning in a file like CONTRIBUTING.md), where the planned change you mentioned is documented for the uninitiated among us to see?

I'm really sorry. It looks like this was among the only three issues not migrated from the old repository. It was https://github.com/whitequark/solvespace/issues/190. I've re-filed it as https://github.com/solvespace/solvespace/issues/319.

> Otherwise, sorry if I stumbled into a closed user group by accident. Won't happen again; I thought this was open source.

What's with the hostility? This *is* open source. And one part of open source is collaboration, namely, trying to coordinate any kind of all-encompassing change with the upstream *before* doing a large amount of work, not after. You could've just asked a question on IRC or on the issue tracker.

ghoss, 2018-04-04:
> What's with the hostility?

The „really unnecessary piece of work“ part did come across a bit derisive in my perception, as if I didn’t heed the warnings written all over the software not to dare touch it without asking and, well, tough luck and stupid me if I did so nevertheless. As already mentioned, hints about planned changes to the UI weren’t exactly abundant, the section on „contributing translations“ didn’t seem too deterrent either, so I didn’t really sense an issue here. In hindsight, I would have asked on IRC, but… wait a minute, where on the website or on GitHub is there any mention that there is an IRC channel?? Anyway… I’m not here to argue, but just to inform that I’ve made a German localization. It may be useful for someone for the time being until your planned template system is in place (at least for me it is); some of the translations might be useful even after that, the code changes probably not, and that’s basically all there is to it.

traverseda, 2018-04-04:
Not sure the best place to bring this up, so I'll try this closed issue, but you might also make the CLA situation a bit clearer. As it stands, this is a really hard repository for people to contribute to, often in surprising ways. A bunch of PR's have been waiting on a CLA for a while now. Maybe just set up [clabot](https://github.com/clabot/clabot) with a default CLA, and see what whoever is checking legal thinks?

whitequark, 2018-04-06:
> Anyway… I’m not here to argue, but just to inform that I’ve made a German localization. It may be useful for someone for the time being until your planned template system is in place (at least for [...]

Sure. Please publish it somewhere!

whitequark, 2018-04-06:
@traverseda That's gated on @jwesthues for many months now. I can't do anything here.

whitequark, 2018-04-21:
Let me correct myself: the CLA is actually gated on me. I know it's been a long time, but all details have been figured out, and I'll get it up ASAP.

ghoss, 2018-07-12:
@whitequark I can do that. Do you have a preferred way for incorporating those changes?

whitequark, 2018-07-12:
I will update the `pot` files in a moment, and then a pull request with a `po` file for `de_DE` would be great. Alternatively, I want to try out [Zanata](http://zanata.org). Let me know which option would be more convenient for you.

ghoss, 2018-07-12:
The `po` file pull request would probably be the easiest option for this one time.

From past experience with a service similar to Zanata, I know that it can take some time to set up the workflow and the options correctly, but once running, I believe that it does make working with localisation files more convenient in the long term.

whitequark, 2018-07-12:
Since SolveSpace is in ongoing development there will be continual changes in localization. I guess I can do both options.

whitequark, 2018-07-12:
Please use the Zanata project I've set up via [this link](https://translate.zanata.org/iteration/view/solvespace/r1_ba26acb963/languages/de). I believe it should allow you to upload a .po file just as well, so you can do whatever is most convenient. Let me know if something doesn't work.

ghoss, 2018-07-13:
Following the link, I get this error message:

[image]

ghoss, 2018-07-19:
@whitequark It's been some time, but I finally got around to uploading the German .po file into Zanata. Please check and let me know in case there's a problem.

whitequark, 2018-07-20:
Thanks, applied in master. I'll figure something out with the text window...