solvespace/solvespace #321: "Split at Point" as special case of "Split at Intersection"

ghoss, 2018-04-07:
I have extended `SplitLinesOrCurves()` in [modify.cpp](https://github.com/ghoss/solvespace/blob/master/src/modify.cpp) in [my fork](https://github.com/ghoss/solvespace) so that splitting is no longer restricted to intersecting lines/curves only.

A split can now also be performed at a point constrained to the line/curve (i.e. a datum point, mid point, or the endpoint of another line/curve with a coincident constraint). The action will not be performed if the point actually lies outside of the endpoints of the line/curve to be split.

Therefore, cases like these will now work:

[image]

whitequark, 2018-04-07:
Thanks, I'll take a look!

whitequark, 2018-07-12:
Thank you for your contribution, and welcome. The SolveSpace project has a Contributor License Agreement; in order for me to be able to merge your changes, please sign it at https://cla-assistant.io/solvespace/solvespace. It will take only a small amount of your time.

ghoss, 2018-07-12:
@whitequark Done!

whitequark, 2018-07-12:
Merged in master, thanks. I had to clean up code style and fix a few bugs, e.g. splitting a circle or arc at point caused an assertion failure.

closed 2018-07-12 by whitequark