solvespace/solvespace #324: Smarter font selection for Command::TTF_TEXT

henriklaxhuber, 2018-05-29:
This is not a bug report but a feature request/discussion.

Currently, when creating a new TTF_TEXT object, the font defaults to `arial.ttf` (see [here](https://github.com/solvespace/solvespace/blob/5744d1d599dfaeaa768e1667509eee1fda3b8aea/src/mouse.cpp#L1139)). Especially on linux systems (such as mine), `arial.ttf` may not be present. This means that every time I create a new TTF_TEXT object, I need to manually scroll down the list and select a suitable sans-serif font, which is rather tedious.

To resolve this issue, the default font should be selected in such a way that it actually exists. I can see primarily three options here:

1.) Require `arial.ttf` as a dependency. This is a simple solution and enables cross-platform compatability. However, it is very much against the GPL philosophy (see the [arial licensing info](https://github.com/solvespace/solvespace/blob/5744d1d599dfaeaa768e1667509eee1fda3b8aea/src/mouse.cpp#L1139)).

1.1.) Require something like FreeSans as a dependency and use it by default.

2.) Use pango to perform automatic font selection. This mean the user gets to see their default sans-serif font being used. However, the font may not be present on other systems opening the save file, or, if only the 'sans-serif' directive is stored in the save file, may look different on other systems.

3.) Persist the last selected font and use it as default when creating a new TTF_TEXT object. The obvious downside is that if the user intents to change the font only for a specific object, he will need to go through the process of scrolling down the list again.

I'm very much willing to create a pull request solving this issue, but I was hoping for some guidance as to which direction to go. Personally, I would prefer the solution `1.1`, as it reduces the problem to a one-off issue when first installing solvespace.

whitequark, 2018-07-12:
Sure, let's add FreeSans (or maybe Bitstream Vera Sans) as a default TTF font.

henriklaxhuber, 2018-07-12:
I'm supposed to be on holiday, but since this seems like a quick fix I'll make a PR later this week or so.

whitequark, 2018-07-12:
Thank you! I suggest adding the font under `res/fonts` and then adding a separate item in the TTF font selector, say `Bitstream Vera Sans (built-in)`. This would clearly establish that this font will never go missing.

henriklaxhuber, 2018-07-12:
Ended up getting around to implementing it tonight.

closed 2018-07-12 by henriklaxhuber