solvespace/solvespace #326: Interest in replacing libdxfrw with librecad? Esp. for constraints and dwg

rurban, 2018-05-31:
Hi

I'm the maintainer of libredwg, and I was wondering if you are interested in better DXF and DWG support? libdxfrw is very limited in its DXF support, just the very basics. Reactors, xrecords and all the objects and relations to represent parametric constraints are missing. libredwg is able to that.
libdxfrw can read more versions, up to r2013, libredwg so far only up to r2007, and its DXF support is just in development. But libredwg can write DWG files and represent all advanced AutoCAD features.
https://github.com/LibreDWG/libredwg/
I'd just need a helping hand to rewrite and test the constraints parts for the start.

The first LibreDWG alpha release will come out soon, I'm just waiting for some GNU permissions to upload it at their site. And I'm finishing the DXF parts and the newer DWG versions.

whitequark, 2018-05-31:
GPLv3 means that it's going to be impossible to get a build of SolveSpace that uses libredwg on iPhoneOS. That's troublesome.

rurban, 2018-05-31:
But solvespace is also GPLv3, isn't it?

whitequark, 2018-05-31:
SolveSpace is dual licensed as GPLv3/commercial.

whitequark, 2018-07-12:
I don't think this is viable for the reason stated in https://github.com/solvespace/solvespace/issues/326#issuecomment-393534335, especially as various people have been showing ongoing interest in having an iOS port. Very unfortunately.

Symbian9, 2018-08-30:
> SolveSpace is dual licensed as GPLv3/commercial.

SolveSpace use libdxfrw (that is under GPLv2+)?

- https://sourceforge.net/p/libdxfrw/home/Home/
  - https://github.com/solvespace/libdxfrw

whitequark, 2018-09-22:
I thought about this again and with the restrictions on iOS apps, GPLv2 and GPLv3 are functionally equivalent here. Any iOS port would probably have to use a web service anyway for dxf conversion.

Thus, libredwg is potentially interesting.

whitequark, 2018-09-22:
@rurban I apologize for the back-and-forth on this, the right decision was not immediately obvious.

If you are still interested, please go ahead, I will provide any help you'll need for conversion.

rurban, 2018-09-22:
Great, will do

whitequark, 2018-09-22:
I would appreciate if you added the new (libredwg) code alongside old (libdxfrw) code, since neither of them subsumes the support matrix of the other. The intent is to eventually retire libdxfrw in favor of libredwg, but during migration it would be good to have both options in case of library bugs etc.

SolveSpace is a mature project which comes with its challenges; I try to be conservative with my decisions here to make it as painless as possible for our users.

Evil-Spirit, 2018-09-23:
@rurban I can help you with supporting constraints, just mention me.

rurban, 2020-09-18:
Started working on it at https://github.com/rurban/solvespace/tree/libredwg
Will need a few months I think.

I'm not entirely sure yet, how AutoCAD stores dynamic blocks and constraints, but I hope I will figure it out during the work.
libredwg supports 80% of the AutoCAD dynamic blocks and constraints already. Not sure how much is needed for solvespace.

ruevs, 2020-10-22:
Not to self. Here is another DXF library:
https://qcad.org/en/what-is-dxflib

DuncanLithgow, 2021-04-09:
Great news that this is started.

@ruevs there's a page here I try and work on which is to track support for DXF in libre software
https://wiki.osarch.org/index.php?title=Drawing_Exchange_Format_(DXF) 

The DWG page needs a bit of cleanup.
https://wiki.osarch.org/index.php?title=Drawing_(DWG)

@rurban maybe you'd like to check those pages for accuracy some time. The idea is to get people to concentrate on using fewer better libraries. Sorry I'm going off topic now ...

phkahler, 2022-01-25:
I've added this to the 4.0 milestone to show it as  a priority. The rurban fork has a lot of work done already but he's busy with other things. It would be nice if someone could evaluate how capable the libredwg version is compared to the libdxfrw version for both DXF and DWG files. If libredwg is at parity for both we should switch. If it's better only for dwg we should use both. If it needs more work, someone might want to pick it up. He really did a lot of work.