solvespace/solvespace #328: Main menu is not rendered in master branch, but works

nazar-pc, 2018-06-04:
### System information

SolveSpace version: 3.0~2b9ffd15 (latest master as of right now)

Operating system: Ubuntu 18.10

### Expected behavior

Main menu bar should be rendered

### Actual behavior

Main menu is not rendered, but when clicked in corresponding area, dropdown menu appears. It works properly in 2.3 (at least as packaged in Ubuntu's repository).

### Additional information

Here is how it looks like (tiling WM, no window decorations):
[image]

whitequark, 2018-06-06:
Yes. It's some issue with the way SolveSpace uses OpenGL. I could never get to the bottom of it, sorry.

nazar-pc, 2018-06-18:
It wouldn't be a big deal, but the same happens to any text field, for instance, when editing dimensions or comments I can't see the text I'm typing, just black rectangle where text field should be

whitequark, 2018-07-31:
I suspect recent changes to the way SolveSpace handles platforms may have fixed this bug. Can you please rebuild from the master branch and say if it helped?

nazar-pc, 2018-07-31:
Just compiled master at 88d59ea7658db15bfd0adb9dd38802f574ac7f9b, doesn't help.
When trying to edit dimensions get following in terminal:
```
(solvespace:15241): Gtk-WARNING **: 23:34:39.934: gtk_widget_size_allocate(): attempt to underallocate gtkmm__GtkFixed's child gtkmm__GtkEntry 0x562fbe8502c0. Allocation is 140x31, but minimum required size is 162x31.
```

When opening menu I don't get any warnings/errors.

And before closing application I'm getting this (yes, that many lines at once):
<details>

```
(solvespace:15241): glibmm-CRITICAL **: 23:36:30.547: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:15241): glibmm-WARNING **: 23:36:30.547: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:15241): glibmm-CRITICAL **: 23:36:30.547: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:15241): glibmm-WARNING **: 23:36:30.547: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:15241): glibmm-CRITICAL **: 23:36:30.548: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:15241): glibmm-WARNING **: 23:36:30.548: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:15241): glibmm-CRITICAL **: 23:36:30.548: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:15241): glibmm-WARNING **: 23:36:30.548: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:15241): glibmm-CRITICAL **: 23:36:30.548: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:15241): glibmm-WARNING **: 23:36:30.548: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:15241): glibmm-CRITICAL **: 23:36:30.548: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:15241): glibmm-WARNING **: 23:36:30.548: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:15241): glibmm-CRITICAL **: 23:36:30.548: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:15241): glibmm-WARNING **: 23:36:30.548: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:15241): glibmm-CRITICAL **: 23:36:30.548: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:15241): glibmm-WARNING **: 23:36:30.548: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:15241): glibmm-CRITICAL **: 23:36:30.548: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:15241): glibmm-WARNING **: 23:36:30.548: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:15241): glibmm-CRITICAL **: 23:36:30.548: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:15241): glibmm-WARNING **: 23:36:30.548: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:15241): glibmm-CRITICAL **: 23:36:30.548: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:15241): glibmm-WARNING **: 23:36:30.548: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:15241): glibmm-CRITICAL **: 23:36:30.548: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:15241): glibmm-WARNING **: 23:36:30.548: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:15241): glibmm-CRITICAL **: 23:36:30.548: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:15241): glibmm-WARNING **: 23:36:30.549: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:15241): glibmm-CRITICAL **: 23:36:30.549: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:15241): glibmm-WARNING **: 23:36:30.549: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:15241): glibmm-CRITICAL **: 23:36:30.549: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

[... 391 lines cut ...]

[... cut ...]
(solvespace:15241): glibmm-WARNING **: 23:36:30.555: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.

(solvespace:15241): glibmm-CRITICAL **: 23:36:30.555: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:15241): glibmm-WARNING **: 23:36:30.555: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.
```

</details>

whitequark, 2018-07-31:
Sigh. OK, still no idea then.

whitequark, 2018-08-08:
So, it's *definitely* not a bug in SolveSpace (well it better not be, I've spent enough time poring over every bit of rendering code), and it's a bug in the Linux graphics stack. Somewhere. In the entire thing.

I'm... probably going to keep this open, since this clearly impacts SolveSpace, but there isn't anything I can do about it, including even writing a proper bug report, because every single debugging tool I have either doesn't detect this bug or perturbs the environment sufficiently for it to disappear and make all the gathered data useless.

nazar-pc, 2018-08-08:
Thanks for the effort @whitequark, I'm using iGPU in CoffeeLake CPU, so driver support should be as good as it gets on Linux (I'm also on Ubuntu 18.10, so most of the newer fixes in drivers should already be present).
Maybe you can create an upstream issue in GNOME tracker and then close this issue with link to upstream ticket?

whitequark, 2018-08-09:
> Maybe you can create an upstream issue in GNOME tracker and then close this issue with link to upstream ticket?

But I can't. An issue that says "under these specific conditions, which I have no way of letting you reproduce, main menu becomes black" is completely unactionable.

nazar-pc, 2018-09-29:
I guess after recent Mesa update (18.1.5 > 18.2.1) it started to look fine, closing now

closed 2018-09-29 by nazar-pc