solvespace/solvespace #329: Property Browser Disappeared

alesnovak, 2018-06-10:
### System information

SolveSpace version: (2.3~7c1ca460)

Operating system: (win10)

### Expected behavior

What should have happened?
Property Browser be visible
### Actual behavior
No Property Browser
What actually happened?
After program installation the Property Browser was there
### Additional information
The only difference between installation and present:
-Win10 update
-Java update
-Using notebook without external monitor
For bugs, please attach a savefile that shows the problematic behavior.
You can attach `.slvs` files by archiving them into a `.zip` first.

asantoni, 2018-06-10:
Does it re-appear if you hit tab?

alesnovak, 2018-06-10:
Thanks Albert. 
But also hitting tab Browser wont appear.

whitequark, 2018-06-10:
I believe the cause is that the property browser appears, but on the external monitor. Try erasing the key `HKCU/Software/SolveSpace` in regedit.

alesnovak, 2018-06-10:
Thanks whitequark. You were right, erasing the key in regedit the Browser is back again on my notebook display. What will happen when I plug in the external monitor again (in 2 weeks when at home)?

whitequark, 2018-07-31:
@alesnovak Can you please recheck with a recent version (e.g. [this](https://ci.appveyor.com/api/buildjobs/5xs7a8hpapr29kg6/artifacts/build%2Fbin%2FDebug%2Fsolvespace.exe) binary) and say if the problem is fixed for you? I've adjusted some of the code responsible for positioning windows, but I'm not sure if those changes were enough.

alesnovak, 2018-07-31:
> <mention@noreply.github.com>
> nonexistent monitors (#329)

Dear, thanks a lot for attention!

But your suggestion "erasing the key HKCU/Software/SolveSpace in 
regedit" solved the problem.

Ales

whitequark, 2018-07-31:
OK, if you or someone else encounters this issue again, I'll reopen it.

closed 2018-07-31 by whitequark