solvespace/solvespace #330: Model library

tpimh, 2018-07-05:
It would be great to have a library of simple common solvespace models (like screws, nuts, gears, etc.), maybe even with simple model generator (entering variable values, getting the right model).
Anyone interested?
It would be nice to have it as an online gallery that anyone can browse, spin 3D models rendered with three.js and then download.

whitequark, 2018-07-12:
Sure. You could do this with solvespace-cli and some server-side glue once parametric models are in.

ceremcem, 2018-10-21:
> maybe even with simple model generator (entering variable values, getting the right model).

Wouldn't it need a scripting infrastructure, which Solvespace has no such thing AFAIK?

whitequark, 2018-10-21:
Not yet.

phkahler, 2019-06-01:
I was just thinking about a hole tool to create hole for predefined bolts and screws. It would have predefined countersunk or chamfered profiles. Perhaps just having a model library would allow this - hole models would just need to import as difference. A little UI help to position and orient them would go a long way.

altaic, 2019-06-02:
@phkahler You’d have to add clearance to the hole threads rather than a zero clearance difference using the screw. I’m not sure, but I imagine different thread profiles will have different clearances due to manufacturing tolerances, use case requirements, etc. A tool that allows the user to set the hole clearance value (with some sort of sane default) would probably do the trick.

phkahler, 2019-06-02:
@altaic I guess I was unclear. There would be standard size hole objects - unthreaded - that could be cut from objects. You select the "M3 countersunk" bolt-hole tool and click on a surface to make that hole. Ideally the location could be set afterward using constraints. This needs more thought...

altaic, 2019-06-02:
@phkaler Ah, I see; sorry for the confusion. Back to the topic, though, a user-driven parts library would be fantastic. It’d be especially cool if one could add GitHub or GitLab (or similar) repos as a parts source.

tpimh, 2019-06-03:
Git repos for modules in the same way KiCad does it would be great!

whitequark, 2019-06-04:
Note that the amount of integration work required to make this happen is enormous. Consider that git is a large and complicated dependency; even the *possibility* of using git repos requires integrating openssl, libssh and libgit, and interfacing with platform cryptography on three distinct platforms, just to be able to clone a remote repository at all. (And none of this would work on Emscripten, which I think is an important way to give people near-universal access to SolveSpace in the age of Fusion 360 and Onshape running in the browser.)

Symbian9, 2019-06-05:
> Git repos for modules in the same way KiCad does it would be great!

FTR, [LibrePCB](https://github.com/LibrePCB/LibrePCB) also has nice built-in "Library Manager" as interface to [LibrePCB-Libraries](https://github.com/LibrePCB-Libraries/) repos on GitHub:

[image]

ruevs, 2020-11-07:
https://github.com/solvespace/solvespace/files/5505412/SS_parts_bin.zip

ruevs, 2020-11-08:
A parts collection from
https://github.com/solvespace/solvespace/issues/793

https://github.com/solvespace/solvespace/files/5505412/SS_parts_bin.zip

hinell, 2022-01-11:
Thumbing up this request. It's paramount for any CAD to have a library interop. Would be best to couple it with existing software.