solvespace/solvespace #336: Adding SolveSpace to package managers

Foadsf, 2018-07-13:
Not an issue, more of a feature request.

Most of the professional users, use package managers to keep their software updated. It would be great if SolveSpace team could make sure that the latest stable and nightly builds of the software is available in most of them. My suggestions:

1. I have listed available package managers for Windows [here](https://www.reddit.com/r/windows/comments/8nq62b/freeopensource_package_managers_for_windows/). The most famous ones are Chocolatey and Scoop. It would be great to make sure the latest SolveSpace is available there. 
2. [repology.org](https://repology.org/) is a great website which keeps track of different package managers. maybe you could add [their badge](https://repology.org/static/demobadge.v1.svg) to the github repo. You could go to [this page](https://repology.org/metapackage/solvespace/versions) and make sure all the package managers which have the SolveSpace are listed there, otherwise report it.

whitequark, 2018-07-13:
SolveSpace is distributed as a single executable file for Windows (and another one for the few people who want the CLI), and of course for macOS as well. Is there really any point in going for e.g. Chocolatey? As far as I'm aware package managers like Chocolatey and NuGet are typically used by developers; SolveSpace doesn't need and won't _install_ anything per se.

As for Linux, well, I tried building `.deb` packages, but this resulted in a generally awful experience for users (`.deb` packages aren't really portable even between Debian and Ubuntu), and the Debian upstream eventually specifically asked me to remove the Debian package from the repository. I have decided that until a technology like AppImage, but one that allows to link to libc and GUI toolkit dynamically without redistributing them, is available, the SolveSpace project will not provide prebuilt Linux binaries.

Foadsf, 2018-07-13:
It is great to have single file binaries like .exe, .dmg, .deb , snap and flatpack but there are a couple of issues. first of all if there is something wrong with the dependencies they can't handle it well. secondly keeping the software updated manually is a hassle. I personally use different operating systems per day and use different package managers to keep my software updated across these machines. I think it would be a great benefit for the SolveSpace community if they could have the software through package managers.

whitequark, 2018-07-13:
It's been a few years since the last release and I don't think SolveSpace will ever release more often than every few months, so it seems to me that it isn't a lot of hassle. Anyway, let's keep this open, if there is substantial interest I'll revisit this topic.

Foadsf, 2018-07-13:
Some of the package managers like HomeBrew can compile from the source. and you can even specify some install options for example --latest, --nightly-build --stable --with-xcode --with-mingw .... this way professional users can follow the latest updates and also give you guys more feedback. It also accelerates the development process.

whitequark, 2018-07-13:
What I'm concerned about is the combinatorial explosion of build configurations. At some point in the past I've spent over half my time _just_ on packaging issues. I already have to test it on three OSes and five compilers, and adding more burden here is not trivial. Moreover, I'm not an expert at packaging, and I wouldn't be able to do a very good job at writing Homebrew formulas or even more so Debian packages.

Foadsf, 2018-07-13:
You should consider that you are not alone. The more package managers supporting the SolveSpace, the more users have access to the latest bug fixes and features, the more they test it and give you feedback and the more they contribute to the project. It is a positive loop stimulating itself after certain threshold. Consider it as a great investment. 
BTW I personally appreciate all your efforts. you are awesome! :)

Symbian9, 2018-07-14:
> You should consider that you are not alone. We will help you.

Of course!

@whitequark, years ago I ask @AlexanderP for setup his PPA for packaging SolveSpace and he did it on my request.

Currently he provide fresh 'nightly' builds for Ubuntu & Debian:

- https://notesalexp.org/packages/en/stretch/amd64/

So, he could help with packaging too.

Symbian9, 2018-07-14:
> The more package managers supporting the SolveSpace, the more users have access to the latest bug fixes and features, the more they test it and give you feedback and the more they contribute to the project.

@Foadsf, fully agree with you ;-)

While there no ready-to-use packages, number of user & testers will not increase.

probonopd, 2018-07-14:
> I have decided that until a technology like AppImage, but one that allows to link to libc and GUI toolkit dynamically without redistributing them, is available, the SolveSpace project will not provide prebuilt Linux binaries.

What do you mean by this? AppImage is just a self-mounting filesystem image. Whether what you put inside is statically or dynamically linked is entirely your decision. Most AppImages do not redistribute libc but merely link to the one in the target system. As for GUI toolkits, it makes sense to bundle it in many cases because you usually can't expect a certain GUI toolkit to be there in each target system (distribution) in a suitable version.

whitequark, 2018-07-14:
> AppImage is just a self-mounting filesystem image. Whether what you put inside is statically or dynamically linked is entirely your decision.

I know, I mean that in practice (e.g. if you follow most of the AppImage creation guides on the web), most AppImages either link dependencies statically or dynamically and pack them inside the AppImage. For software without heavy security requirements, like SolveSpace, this is 100% fine except for two cases:
  * the libc, which loads shared libraries from the host system to provide e.g. nsswitch services, and
  * the GUI toolkit, which loads shared libraries from the host system to provide theming.

In case of libc, even Go, which is probably _the_ language most focused on static linking, gives up on Linux and links to glibc. That part is solvable for SolveSpace, though I would likely need some sort of way to force it to link only to old enough glibc symbols (I think there's a way to do this with linker scripts, although symbol versioning is only documented for the case of building shared libraries...)

In case of the GUI toolkit, this is completely broken in case of GTK, because the GTK developers do not maintain a stable ABI since version 3 (which I can only explain either with incompetence or with complete disregard to the needs of developers distributing software other than through a distribution package manager, go figure which option is worse). Of course, this is something that Windows and macOS, and even other GUI toolkits like Qt, provide as a baseline expectation.

Short of switching to Qt, there is no way to fix this issue, and so it's not possible to make AppImages of high enough quality, and so I won't make official AppImages.

whitequark, 2018-07-15:
> What do you mean by "give up on Linux"?

They link to glibc if you `import "net"` because of nsswitch:
```
$ cat test.go
package main
import "net"
func main() {
        net.Dial("tcp", "golang.org:80")
}
$ go build test.go
$ ldd ./test
        linux-vdso.so.1 (0x00007fff7e2a4000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f292e664000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f292e2aa000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f292e882000)
```
> Why not bundle the GUI toolking, whichever one you use? I do this for Qt, for Electron, for WxWindows,... you name it. This is working well for me.

Because the GUI toolkit depends on the shared libraries from the system where it runs on, primarily theming. If the toolkit in question doesn't properly implement ABI compatibility, you can only turn that off or just rely on the loads to fail (like Skype did) and that results in an awful user experience, so I'm not going to do that.

probonopd, 2018-07-15:
@whitequark I think you are up to something here. I like your line of thinking, the desktop Linux space needs people like you to think about those platform issues and suggest on how to solve them. Would you like to work with me in the https://gitlab.com/probono/platformissues/ project?

whitequark, 2018-07-15:
@probonopd I'm not opposed to it per se, but my plate is _more_ than full. Consider that Linux is just one of three platforms SolveSpace runs on. Anyway, the Gtk situation is broken beyond all repair and I don't think anything can be meaningfully done about it, and with Qt, I believe, what I want already works if some conditions on the versions of Qt used are satisfied, so what needs to be adde is some guarantees about the version of Qt present on the host system.

Foadsf, 2018-07-16:
@whitequark following this discussion, I thought you should also take a look at [Zero Install](https://0install.de/) (or [this link](http://0install.net/)) which tries to offer a universal and cross platform app store.

Symbian9, 2018-07-30:
Thanks to @AlexanderP's PPA, fresh 'nightly' builds (SolveSpace 3.0~2b4c484d) available for:

- Debian 9 "stretch"
  - https://notesalexp.org/packages/en/stretch/amd64/solvespace/
- Debian 10 "buster"
  - https://notesalexp.org/packages/en/buster/amd64/solvespace
- Debian (unstable) "sid"
  - https://notesalexp.org/packages/en/sid/amd64/solvespace
- Ubuntu 16.04 LTS "Xenial"
  - https://notesalexp.org/packages/en/xenial/amd64/solvespace
- Ubuntu 18.04 LTS "Bionic"
  - https://notesalexp.org/packages/en/bionic/amd64/solvespace

Source package:

- https://notesalexp.org/packages/en/source/solvespace/

whitequark, 2018-07-31:
> @whitequark following this discussion, I thought you should also take a look at Zero Install (or this link) which tries to offer a universal and cross platform app store.

I know about 0install and have considered it in the past, but it isn't widely used enough to have e.g. GTK 3 I can link to. If this changes in the past, and the same issues as those relevant to AppImage are resolved, it could be used.

whitequark, 2018-07-31:
@Symbian9 I've added a [README section](https://github.com/solvespace/solvespace#via-third-party-binary-packages) with this repository.

whitequark, 2018-07-31:
> You should consider that you are not alone. We will help you. The more package managers supporting the SolveSpace, the more users have access to the latest bug fixes and features, the more they test it and give you feedback and the more they contribute to the project. It is a positive loop stimulating itself after certain threshold. Consider it as a great investment.

Okay, so here's what I'm going to do.

I definitely don't have the bandwidth to handle packaging. I know this from experience; just the normal maintenance tasks are well enough to keep my plate full.

I welcome both packagers and packaging requests. If you made a package (formula, etc) for some package manager, file an issue and I'll add it to README and website, as appopriate. If you yourself want a package for some package manager ("it would be good to have" doesn't count, I'm not interested in supporting hypothetical users), open an issue for that specific package manager, and we'll see if someone can do that.

As far as I'm concerned, there isn't anything else to do with this issue after formulating the plan above, because I won't be working on packaging myself. Thus, closing.

closed 2018-07-31 by whitequark