solvespace/solvespace #337: Main menu stops working

narical, 2018-07-14:
### System information

SolveSpace version: 3.0~2b9ffd15
Operating system: Archlinux (stable repos, up to date)

### What is going on

In some cases main menu drop-downs stop working.

### Steps to reproduce:

1. run solvespace
2. select property browser window (make it active)
3. press any main menu button (file, edit, etc.)

### Additional information

Every time solvespace gives the same error in terminal:
`(solvespace:27385): Gdk-CRITICAL **: 20:28:36.852: Window 0x55927d24fb00 has not been made visible in GdkSeatGrabPrepareFunc`

There was another error, but only once and I can't reproduce it:
`(solvespace:27385): Gtk-CRITICAL **: 20:28:55.003: gtk_window_set_mnemonics_visible: assertion 'GTK_IS_WINDOW (window)' failed`

narical, 2018-07-14:
Updated solvespace to latest git master, bug still presented.

whitequark, 2018-07-14:
Can you please build the `platform` branch from this repository and see if it helps?

narical, 2018-07-14:
I built it from platform branch, same error.
Difference is - when I close window I get ~20 similar pairs of errors:
```
(solvespace:27299): glibmm-CRITICAL **: 21:41:25.228: Glib::ObjectBase* Glib::wrap_create_new_wrapper(GObject*): assertion 'wrap_func_table != nullptr' failed

(solvespace:27299): glibmm-WARNING **: 21:41:25.228: Failed to wrap object of type 'GtkWindow'. Hint: this error is commonly caused by failing to call a library init() function.
```

whitequark, 2018-07-14:
That's really odd. Which GTK version?

narical, 2018-07-14:
```
local/gtk2 2.24.32-1
    GObject-based multi-platform GUI toolkit (legacy)
local/gtk3 3.22.30-1
    GObject-based multi-platform GUI toolkit
```

narical, 2018-07-14:
Reinstalled gtk packages, rebooted PC, changed gnome shell and gtk themes to default (adwaita). Same.

whitequark, 2018-07-18:
Sorry, no idea. I have the exact same GTK version and everything works.

narical, 2019-01-16:
Bug disappeared. Close issue?

closed 2019-01-16 by whitequark

a6zj6, 2020-01-06:
Can this issue be re-opened?  I am running Ubuntu 18.04 LTS and I have this exact same problem.  This is not specific to any one application for me, it's the entire operating system.  For example, if I have mail open and another application open, if mail is not the currently-focused window and I click on any menu in the mail window (evolution mail), the menu stops working.  The only way I can get the menus to again re-appear is to quit the program and re-open.  Very frustrating.

What is going on: When main window is inactive, pressing main menu breaks drop-downs, they stop working.

whitequark, 2020-01-06:
Why are you asking to reopen a SolveSpace issue if you say that your problem isn't SolveSpace-specific?

a6zj6, 2020-01-06:
My apologies, I wasn't aware that it was specific to SolveSpace; it is however the identical issue which, in my mind, means it is not specific to SolveSpace but is a bug within the operating system itself.
Should I open a new issue?

a6zj6, 2020-01-06:
I should also mention that the error I get if I launch from terminal and reproduce the issue is identical:
(evolution:12656): Gdk-CRITICAL **: 10:21:32.823: Window 0x5604a7d4d4b0 has not been made visible in GdkSeatGrabPrepareFunc

whitequark, 2020-01-06:
You're correct; it is not an issue with SolveSpace, although before now no one has reported that it affects other applications too. Unfortunately, *because* it is not an issue with SolveSpace, there is nothing I can do about it, so there is no point in keeping the bug open or opening a new one.

You could report it against GTK upstream. Though, if I recall, I did mention it on GTK IRC channel and we concluded it's probably not GTK either, but rather Mesa or i965.

a6zj6, 2020-01-06:
Ok, thanks for the help whitequark.
I've reported it in askubuntu.com here: https://askubuntu.com/questions/1201041/issue-with-menu-not-being-displayed-when-clicked-on-and-window-is-not-focused-be

So, based on your last post, you've seen this issue as well?  Do you still experience this issue or were you able to correct it?

SyntevoAlex, 2020-06-22:
I'm investigating a similar issue and I think this is how to reproduce:
1. In `GNOME settings | Region & Language`, select `Allow different sources for each window`.
2. Run an GNOME on XOrg session.
2. When window is on background, right-click the menubar.
3. Repeat a few times if needed.
4. Issue is present on Ubuntu 18.04, but no longer there on Ubuntu 20.04.

See also: https://gitlab.gnome.org/GNOME/gnome-shell/-/issues/1899