solvespace/solvespace #340: Increase dimension line offset from text

Symbian9, 2018-07-28:
### System information

SolveSpace version: 3.0~8426992f

Operating system: Debian 9 "Stretch"

### Expected behavior

Offset from dimension value text to dimension line.

### Actual behavior

Currently horizontal offset equal *1px*, that is too close.

In same time vertical offset equal *3 px*.

### Additional information

[image]

[image]

[image]

[image]

Evil-Spirit, 2018-07-30:
Also probably we should get rid of REF. Just use ISO ANSI GOST or some other approach.

whitequark, 2018-07-30:
ISO does not specify parametric sketches, so it's fine that we depart from ISO where parametricity is required.

Evil-Spirit, 2018-07-30:
I mean, drawings have common approach to specify reference dimensions. for ex. in GOST  "*" is used:
http://gk-drawing.ru/images/size-inquiry-01.jpg
[image]

whitequark, 2018-07-30:
@Evil-Spirit Oh, I didn't know that. Thanks.

Symbian9, 2018-07-30:
> in GOST "*" is used

Depending on used font, "*" should be also in `<sup>` position.

- `X*`: X*
- `X<sup>*</sup>`: X<sup>*</sup>

Symbian9, 2018-08-02:
> Fixed in eda294e, thanks for the report.

It's look like *Angle* constraint text still look buggy:

[image]

Can you fix it too?

whitequark, 2019-05-21:
Fixed in 25b6eba.

closed 2019-05-21 by whitequark