solvespace/solvespace #341: Angle constraint ignore digits after decimal point global settings

Symbian9, 2018-07-28:
### System information

SolveSpace version: 3.0~8426992f

Operating system: Debian 9 "Stretch"

### Expected behavior

*Angle* constraint should use global "digits after decimal point" settings. Otherwise need option to set it separately.

### Actual behavior

Currently *Angle* constraint fixed to 2 digits after decimal point.

### Additional information

[image]

[image]

whitequark, 2018-07-28:
No. The current behavior is intentional. There are essentially no cases where you want more than two digits after decimal point when constraining angles, so the extra digits would simply clutter up screen space.

closed 2018-07-28 by whitequark

Evil-Spirit, 2018-07-28:
@whitequark can't agree with you. There are cases where you need more digits to display.

whitequark, 2018-07-28:
@Evil-Spirit OK, you are right, there is a use case presented in #98. I will reopen that issue.