solvespace/solvespace #343: Image transparency/opacity

Symbian9, 2018-07-28:
### System information

SolveSpace version: 3.0~8426992f

Operating system: Debian 9 "Stretch"

### Expected behavior

After "Sketch Image" there should be a way how set image transparency/opacity.

Alternatively add support for `*.png` images in APNG format (with transparency), so user can prepare image with transparency before import it to SolveSpace.

### Actual behavior

Currently there no way how to set transparency/opacity for imported *Image*.

Also, actually SolveSpace not support `*.png` in APNG format (it import it as PNG format without transparency).

### Additional information

[image]

whitequark, 2018-07-28:
The PNG format, of course, supports transparency, here is a PNG with a transparent cutout:
[image]

There is no need for support of APNG, because the only thing APNG adds is support for animation.

Symbian9, 2018-07-28:
> There is no need for support of APNG

Oh, sorry. Yes, I mean PNG with ARGB colors.

Currently SolveSpace suport only PNG with RGB colors.

whitequark, 2018-07-28:
That's not true. SolveSpace supports ARGB PNGs just fine. The problem is that rendering transparent geometry is hard because it is order dependent, and proper support for transparent geometry is a very significant task that is outside of scope of what SolveSpace tries to achieve.

However, that doesn't mean that we can't support PNGs with holes in them, i.e. with pixels that have alpha set to zero exactly. This is now supported in 2b4c484:
[image]

To reiterate, *blending* PNGs with pixels that have alpha between 0.0 and 1.0 is not supported (and probably will not be supported), and any pixels with non-zero alpha will appear opaque.

closed 2018-07-28 by whitequark

Symbian9, 2018-07-28:
> However, that doesn't mean that we can't support PNGs with holes in them, i.e. with pixels that have alpha set to zero exactly. This is now supported
Thanks, this would be very useful! ;-)

But, is there a way to apply opacity to "extrude" group with *Image*? Or its fully impossible too?

[image]

whitequark, 2018-07-28:
You can't extrude images directly. Trace the image using SolveSpace entities and extrude that instead.

Symbian9, 2018-07-28:
> You can't extrude images directly. Trace the image using SolveSpace entities and extrude that instead.

No, I not try extrude image to 3D model (I now that it's impossible)

I mean that opacity option now shown only in "extrude"/"lathe"/"assemble" groups.

As "sketch-in-plane" has no opacity option, I just make "extrude" group from "sketch-in-plane" with *Image* (you can see it on screenshot).

whitequark, 2018-07-28:
The opacity option only affects geometry. It won't work on image requests.

Symbian9, 2018-07-28:
> It won't work on image requests.

Okay...