solvespace/solvespace #347: Build an AppImage targeting Xenial or Bionic

whitequark, 2018-07-31:
As discussed many times before, having a downloadable, distro-independent Linux binary would be useful. Per https://github.com/solvespace/solvespace/issues/344#issuecomment-408624727, this AppImage cannot be built against Trusty, and might not be possible to build against Xenial. This needs to be checked.

AppImages built against Bionic are definitely capable of upholding the standard of user experience I consider acceptable, so we should build at least that.

alexandrestein, 2018-12-28:
> having a downloadable, distro-independent Linux binary would be useful

Ho yes, I really think so.

I'm on Arch Linux.

I don't want to install on my system every needed libs.
I tried to build Solverspace with Docker but it looks like most of the libs are SHARED. So the building is not usable outside of the container.

I tried to make an AppImage from within Docker but it doesn't start.
It's the first time I use AppImage so I'm not sure to do it right.

Is there any progress on this from your side? 
I would like to help if possible. 

Flatpack is maybe a better choice but it looks more complex to setup. (I have no experience either)

Here is the [Dockerfile](https://github.com/solvespace/solvespace/files/2715001/Dockerfile.txt) I made but the extracted AppImage is not running on my system. 
The [linuxdeploy-x86_64.AppImage image](https://github.com/linuxdeploy/linuxdeploy/releases/download/continuous/linuxdeploy-x86_64.AppImage) the Dockerfile refers to.

Symbian9, 2019-01-15:
> Here is the [Dockerfile](https://github.com/solvespace/solvespace/files/2715001/Dockerfile.txt) I made but the extracted AppImage is not running on my system.
>
> The [linuxdeploy-x86_64.AppImage [...]

@probonopd, could you check what is wrong?

probonopd, 2019-01-15:
Please upload the non-working AppImage for analysis.

probonopd, 2019-01-15:
Do you know which resource it is trying to load?
To find out, runnnig `sudo strace -f ./solvespace-x86_64.AppImage 2>&1 | grep ENOENT` may be helpful.

alexandrestein, 2019-01-15:
I can't test it right now.
Is it working for you?

probonopd, 2019-01-15:
I also can't test it right now ;-)

alexandrestein, 2019-01-16:
I ran it but I can't read the output...
```
sudo strace -f ./solvespace-x86_64.AppImage 2>&1 | grep ENOENT
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... access resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... access resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... access resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... stat resumed> 0x7ffef525a5e0) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... stat resumed> 0x7ffef525a5e0) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... stat resumed> 0x7ffef525a5e0) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... stat resumed> 0x7ffef525a5e0) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... stat resumed> 0x7ffef525a5e0) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... stat resumed> 0x7ffef525a5e0) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... stat resumed> 0x7ffef525a5e0) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] <... openat resumed> )      = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] openat(AT_FDCWD, "lib/charset.alias", O_RDONLY) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] access("/etc/gtk-3.0/settings.ini", F_OK) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] access("/etc/xdg/gtk-3.0/settings.ini", F_OK) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] access("/root/.config/gtk-3.0/settings.ini", F_OK) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] stat("/root/.config/gtk-3.0/gtk.css", 0x7ffef525a620) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] openat(AT_FDCWD, "/usr/lib/gvfs/tls/x86_64/x86_64/libgvfscommon.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] stat("/usr/lib/gvfs/tls/x86_64/x86_64", 0x7ffef52594d0) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] openat(AT_FDCWD, "/usr/lib/gvfs/tls/x86_64/libgvfscommon.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] stat("/usr/lib/gvfs/tls/x86_64", 0x7ffef52594d0) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] openat(AT_FDCWD, "/usr/lib/gvfs/tls/x86_64/libgvfscommon.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] stat("/usr/lib/gvfs/tls/x86_64", 0x7ffef52594d0) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] openat(AT_FDCWD, "/usr/lib/gvfs/tls/libgvfscommon.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] stat("/usr/lib/gvfs/tls", 0x7ffef52594d0) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] openat(AT_FDCWD, "/usr/lib/gvfs/x86_64/x86_64/libgvfscommon.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] stat("/usr/lib/gvfs/x86_64/x86_64", 0x7ffef52594d0) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] openat(AT_FDCWD, "/usr/lib/gvfs/x86_64/libgvfscommon.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] stat("/usr/lib/gvfs/x86_64", 0x7ffef52594d0) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[... cut ...]
[pid 23413] <... stat resumed> 0x7ffef525ace0) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[pid 23413] openat(AT_FDCWD, "/usr/share/solvespace/locales.txt", O_RDONLY) = -1 ENOENT (Aucun fichier ou dossier de ce type)
[1]    23408 abort      sudo strace -f ./solvespace-x86_64.AppImage 2>&1 | 
       23409 done       grep --color=auto --exclude-dir={.bzr,CVS,.git,.hg,.svn} ENOENT
```

I think you should take a look when you have the time for this.
Ask me for testing if you want but I pretty lost on troubleshooting C programs... :tired_face:

:v:

probonopd, 2019-01-16:
> pretty lost on troubleshooting C programs...

It's always best to have the author of an application join the effort, because usually it's mcuh easier for them to know what might be going on.

whitequark, 2019-04-21:
Yes, it would crash if it cannot find the resources directory. It looks for either `/usr/share/solvespace` (or whichever absolute path is configured via CMake) or `../res` relative to the binary path.

probonopd, 2019-04-21:
> `../res` relative to the binary path

Since we have `usr/bin/solvespace` we would need `usr/res`, right? But it is missing in the AppImage. There is no `usr/share/solvespace` either. I assume the AppImage is missing the resources directory entirely @alexandrestein.

whitequark, 2019-04-21:
I'd say you don't need the usr/ hierarchy inside the AppImage at all.

probonopd, 2019-04-21:
Well, it's best practice to have a usr/ tree inside an AppImage so that we have things like `usr/share/applications/*.desktop` and `usr/share/icons/hicolor/*/apps/*` in a the correct place from where they can be picked up by desktop integration tools like `appimaged`.

whitequark, 2019-04-21:
If you added the lookup for `../share/solvespace` that works the same way as the `../res` lookup I would accept that PR.

probonopd, 2019-04-21:
Being worked on:

https://github.com/probonopd/solvespace/blob/AppImage/.travis/build-appimage.sh

whitequark, 2019-04-21:
Note that I still do not think that AppImage is a good idea in its current form, and I have no plans to distribute them. But I can't stop other people from doing it, and adding support for relative lookup in `/usr` is a generally useful behavior; I think it'll benefit NixOS for one.

probonopd, 2019-04-21:
Looks like we may be getting somewhere:
https://github.com/probonopd/solvespace/releases

Symbian9, 2019-04-23:
I try launch it under Debian 9.x, but it failed

```Shell
$ ./SolveSpace-98d9ac3-x86_64.AppImage

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.065: Theme parsing error: gtk-contained.css:15:21: '-gtk-outline-radius' is not a valid property name

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.066: Theme parsing error: gtk-contained.css:16:28: '-gtk-secondary-caret-color' is not a valid property name

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.066: Theme parsing error: gtk-contained.css:28:20: '-gtk-icon-shadow' is not a valid property name

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.066: Theme parsing error: gtk-contained.css:36:2: Missing name of pseudo-class

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.067: Theme parsing error: gtk-contained.css:49:21: Missing name of pseudo-class

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.070: Theme parsing error: gtk-contained.css:101:6: Missing name of pseudo-class

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.071: Theme parsing error: gtk-contained.css:103:8: Missing name of pseudo-class

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.072: Theme parsing error: gtk-contained.css:134:18: '-gtk-icon-shadow' is not a valid property name

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.073: Theme parsing error: gtk-contained.css:139:20: '-gtk-icon-shadow' is not a valid property name

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.079: Theme parsing error: gtk-contained.css:154:20: Missing name of pseudo-class

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.079: Theme parsing error: gtk-contained.css:162:12: 'min-height' is not a valid property name

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.080: Theme parsing error: gtk-contained.css:203:14: 'min-height' is not a valid property name

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.081: Theme parsing error: gtk-contained.css:211:13: Missing name of pseudo-class

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.081: Theme parsing error: gtk-contained.css:224:22: Missing name of pseudo-class

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.082: Theme parsing error: gtk-contained.css:262:13: Missing name of pseudo-class

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.082: Theme parsing error: gtk-contained.css:275:20: '-gtk-icon-shadow' is not a valid property name

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.082: Theme parsing error: gtk-contained.css:284:22: '-gtk-icon-shadow' is not a valid property name

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.083: Theme parsing error: gtk-contained.css:293:22: '-gtk-icon-shadow' is not a valid property name

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.083: Theme parsing error: gtk-contained.css:294:20: Missing name of pseudo-class

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.084: Theme parsing error: gtk-contained.css:324:36: Missing name of pseudo-class

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.084: Theme parsing error: gtk-contained.css:332:35: Missing name of pseudo-class

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.085: Theme parsing error: gtk-contained.css:332:35: Missing closing bracket for :not()

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.085: Theme parsing error: gtk-contained.css:337:37: Missing name of pseudo-class

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.085: Theme parsing error: gtk-contained.css:337:37: Missing closing bracket for :not()

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.085: Theme parsing error: gtk-contained.css:342:30: Missing name of pseudo-class

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.086: Theme parsing error: gtk-contained.css:352:58: Missing name of pseudo-class

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.087: Theme parsing error: gtk-contained.css:370:30: Missing name of pseudo-class

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.091: Theme parsing error: gtk-contained.css:403:12: 'min-height' is not a valid property name

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.092: Theme parsing error: gtk-contained.css:404:11: 'min-width' is not a valid property name

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.093: Theme parsing error: gtk-contained.css:424:20: '-gtk-icon-shadow' is not a valid property name

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.093: Theme parsing error: gtk-contained.css:440:20: '-gtk-icon-effect' is not a valid property name

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.094: Theme parsing error: gtk-contained.css:456:20: '-gtk-icon-shadow' is not a valid property name

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.094: Theme parsing error: gtk-contained.css:459:20: '-gtk-icon-effect' is not a valid property name
[... 634 lines cut ...]

[... cut ...]
(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.262: Theme parsing error: gtk-contained.css:4373:20: '-gtk-icon-shadow' is not a valid property name

(SolveSpace-98d9ac3-x86_64.AppImage:18450): Gtk-WARNING **: 20:07:47.262: Theme parsing error: gtk-contained.css:4378:22: '-gtk-icon-shadow' is not a valid property name
File /home/travis/build/probonopd/solvespace/src/platform/platform.cpp, line 241, function Join:
Assertion failed: component.find(SEPARATOR) == std::string::npos.
Message: Use the Path::Join(const Path &) overload to append an entire path.
Aborted
```

probonopd, 2019-04-24:
Needs further testing. 

```
File /home/travis/build/probonopd/solvespace/src/platform/platform.cpp, line 241, function Join:
Assertion failed: component.find(SEPARATOR) == std::string::npos.
Message: Use the Path::Join(const Path &) overload to append an entire path.
Aborted
```

seems to be coming from https://github.com/solvespace/solvespace/blob/fa6622903010eb82b07b16adf06d31d5ffcd6bb9/src/platform/platform.cpp#L241.

@whitequark can you give us some hint what may be going wrong here?

whitequark, 2019-04-24:
@probonopd Exactly what it says in the assertion text. Your PR was buggy and I fixed it in master already.

probonopd, 2019-04-24:
Sorry for my buggy PR and thanks for fixing!  👍 (You can guess that I am not a C++ expert by any stretch of the imagination.)

probonopd, 2019-04-24:
@Symbian9 a fixed AppImage for d103974 should appear in a few minutes on https://github.com/probonopd/solvespace/releases, would be great if you could give it a try again. Thanks!

Symbian9, 2019-04-25:
> would be great if you could give it a try again.

@probonopd, I just tested latest build & it still failed on start

<details>
    <summary>Terminal output</summary>

```
$ ./SolveSpace-d103974-x86_64.AppImage

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.712: Theme parsing error: gtk-contained.css:15:21: '-gtk-outline-radius' is not a valid property name

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.713: Theme parsing error: gtk-contained.css:16:28: '-gtk-secondary-caret-color' is not a valid property name

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.714: Theme parsing error: gtk-contained.css:28:20: '-gtk-icon-shadow' is not a valid property name

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.714: Theme parsing error: gtk-contained.css:36:2: Missing name of pseudo-class

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.715: Theme parsing error: gtk-contained.css:49:21: Missing name of pseudo-class

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.716: Theme parsing error: gtk-contained.css:101:6: Missing name of pseudo-class

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.717: Theme parsing error: gtk-contained.css:103:8: Missing name of pseudo-class

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.718: Theme parsing error: gtk-contained.css:134:18: '-gtk-icon-shadow' is not a valid property name

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.718: Theme parsing error: gtk-contained.css:139:20: '-gtk-icon-shadow' is not a valid property name

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.725: Theme parsing error: gtk-contained.css:154:20: Missing name of pseudo-class

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.726: Theme parsing error: gtk-contained.css:162:12: 'min-height' is not a valid property name

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.727: Theme parsing error: gtk-contained.css:203:14: 'min-height' is not a valid property name

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.727: Theme parsing error: gtk-contained.css:211:13: Missing name of pseudo-class

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.728: Theme parsing error: gtk-contained.css:224:22: Missing name of pseudo-class

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.728: Theme parsing error: gtk-contained.css:262:13: Missing name of pseudo-class

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.729: Theme parsing error: gtk-contained.css:275:20: '-gtk-icon-shadow' is not a valid property name

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.729: Theme parsing error: gtk-contained.css:284:22: '-gtk-icon-shadow' is not a valid property name

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.730: Theme parsing error: gtk-contained.css:293:22: '-gtk-icon-shadow' is not a valid property name

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.730: Theme parsing error: gtk-contained.css:294:20: Missing name of pseudo-class

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.731: Theme parsing error: gtk-contained.css:324:36: Missing name of pseudo-class

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.731: Theme parsing error: gtk-contained.css:332:35: Missing name of pseudo-class

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.731: Theme parsing error: gtk-contained.css:332:35: Missing closing bracket for :not()

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.731: Theme parsing error: gtk-contained.css:337:37: Missing name of pseudo-class

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.732: Theme parsing error: gtk-contained.css:337:37: Missing closing bracket for :not()

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.732: Theme parsing error: gtk-contained.css:342:30: Missing name of pseudo-class

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.732: Theme parsing error: gtk-contained.css:352:58: Missing name of pseudo-class

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.733: Theme parsing error: gtk-contained.css:370:30: Missing name of pseudo-class

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.737: Theme parsing error: gtk-contained.css:403:12: 'min-height' is not a valid property name

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.737: Theme parsing error: gtk-contained.css:404:11: 'min-width' is not a valid property name

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.738: Theme parsing error: gtk-contained.css:424:20: '-gtk-icon-shadow' is not a valid property name

[... 642 lines cut ...]
(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.892: Theme parsing error: gtk-contained.css:4194:12: 'min-height' is not a valid property name

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.893: Theme parsing error: gtk-contained.css:4211:29: Missing name of pseudo-class

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.893: Theme parsing error: gtk-contained.css:4217:11: 'min-width' is not a valid property name

[... cut ...]
(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.897: Theme parsing error: gtk-contained.css:4373:20: '-gtk-icon-shadow' is not a valid property name

(SolveSpace-d103974-x86_64.AppImage:2630): Gtk-WARNING **: 01:25:38.897: Theme parsing error: gtk-contained.css:4378:22: '-gtk-icon-shadow' is not a valid property name
SolveSpace!

File /home/travis/build/probonopd/solvespace/src/platform/platform.cpp, line 572, function LoadResource:
Assertion failed: ReadFile(ResourcePath(name), &cache[name]).
Message: Cannot read resource.
Aborted
```
</details>

probonopd, 2019-05-07:
I think the theme parsing stuff is not fatal, but the `Assertion failed: ReadFile(ResourcePath(name), &cache[name]).` is. Don't know why it is still trying to load a resource from a wrong path. Can you find out using `strace` which file is not found, and from where it is tried to be loaded? (`ENOENT` in `strace`)

whitequark, 2019-05-07:
@probonopd It crashes trying to find `/tmp/.mount_SolveSBylYgR/usr/share/solvespace/freedesktop/solvespace-48x48.png`.

whitequark, 2019-05-07:
@probonopd By the way your package is very wasteful on disk space. It packages `solvespace-cli`, which is about 60M large, and `libslvs`, about 4M, which can't be used at all.

probonopd, 2019-05-07:
Thanks for looking into it @whitequark .

> It packages solvespace-cli, which is about 60M large, and libslvs, about 4M, which can't be used at all.

So they are not needed and can be removed?

whitequark, 2019-05-07:
> So they are not needed and can be removed?

There's no way to use them from inside the AppImage.

probonopd, 2019-05-07:
> There's no way to use them from inside the AppImage.

We could easily provide for a way to use them from inside the AppImage
* If the user symlinks the AppImage to `solvespace-cli` and executes it via the symlink
* If the user executes the AppImage with `solvespace-cli` as the first parameter

This is currently not implemented but could easily be implemented if needed. But probably making a separate AppImage for the CLI tool is the better way to go.

probonopd, 2019-05-07:
@whitequark `/usr/share/solvespace/freedesktop/solvespace-48x48.png` is a rather unusual location. `/usr/share/icons/hicolor/48x48/apps/solvespace.png` would be the usual one. We can symlink it, no problem, just saying.

whitequark, 2019-05-07:
@probonopd 

> `/usr/share/solvespace/freedesktop/solvespace-48x48.png` is a rather unusual location.

It's explicitly installed there by the SolveSpace build system, because it is one image it is referring to internally. It also installs icons into `/usr/share/icons` separately.

I don't know why you started messing with it. Everything under `/usr/share/solvespace` should be left intact for the application to function correctly.

https://github.com/solvespace/solvespace/blob/e7b75f19c34c923780db776592b47152650d8f22/res/CMakeLists.txt#L127-L150

probonopd, 2019-05-07:
Yes, that's what I am doing now. I thought it was redundant, but I was obviously wrong.

probonopd, 2019-05-07:
Looks like we have something that launches now :-)

[image]

whitequark, 2019-05-07:
@probonopd No, it does not look good at all. You built with `-DENABLE_SANITIZERS=ON`, which is just completely wrong. Sanitizers destroy runtine performance and should never be deployed to users. Also, `-DCMAKE_BUILD_TYPE=RelWithDebInfo` does nothing but bloats the binaries, because you do not use `make install/strip`.

probonopd, 2019-05-07:
Thanks @whitequark. Please check again, are we getting there?

whitequark, 2019-05-07:
Yes, this is better. But the copy of GTK you are shipping is broken. E.g. the little arrow pointing to submenus is missing in your build:
[image]

probonopd, 2019-05-07:
As the AppImage does not ship a GTK theme nor GTK resources such as images, I think it is using what it finds on the host.

This is how it looks for me on Xubuntu 18.04:

[image]

whitequark, 2019-05-07:
In that case you should not ship the rest of GTK either and use what's on the host as well. This is exactly the kind of problem with AppImage that I have warned about before.

probonopd, 2019-05-07:
The problem with GTK is that even after it has reached a major version number (such as 3.0) they are making binary incompatible changes. If I delete `libgtk-3.so.0` from the AppDir, then I get `/usr/lib/x86_64-linux-gnu/libgtk-3.so.0: undefined symbol: gdk_display_get_monitor` on Ubuntu 18.04, and if I also delete `libgdk*` I get

> OpenGL context creation failed

So I think for now the current set of bundled libraries is at least functional, albeit with cosmetic glitches. If we find out which additional items we need to bundle, we can do that.

whitequark, 2019-05-07:
> The problem with GTK is that even after it has reached a major version number (such as 3.0) they are making binary incompatible changes.

In theory (see [this](https://github.com/solvespace/solvespace/issues/344#issuecomment-408624727) for details), after GTK has reached 3.22, it should be upwards ABI compatible. Are you building against 3.22 or later?

probonopd, 2019-05-07:
3.18.9, the one that is in xenial.

whitequark, 2019-05-07:
Wait. That doesn't add up. Specifically, the requirement for Xenial was because SolveSpace required new GtkGLArea features. However, in [this comment](https://github.com/solvespace/solvespace/issues/344#issuecomment-409667651), @ebassi said:

> a binary built with GTK+ < 3.22 is guaranteed to run against GTK+ ≥ 3.22

So, you should talk to the GTK people to figure out why that doesn't work. I don't think shipping an AppImage with broken UI theming is acceptable and if you do so I request you to prominently state that it is not supported by upstream.

probonopd, 2019-05-07:
I am not interested in doing anything that is not supported by upstream... which means we'll have to find a solution ;)

probonopd, 2019-05-07:
Let's check if my line of thinking is correct:
* The oldest currently supported Ubuntu LTS release is 16.04 (xenial)
* Assuming we want to target xenial and later (Ubuntu and other distributions), we can assume that those come with GTK >= 3.18.9, the one that is in xenial
* Given that "a binary built with GTK+ < 3.22 is guaranteed to run against GTK+ ≥ 3.22", we do not need to bundle GTK

Question: With "GTK" only `libgtk-3.so.0` is meant - or which _set of libraries_ exactly? How about gdk, atk, gtkmm?

whitequark, 2019-05-07:
That's a good question. Looking at my local build, these libraries *definitely* need to be bundled:
```
  NEEDED               libspnav.so.0
  NEEDED               libjson-c.so.3
  NEEDED               libgtkmm-3.0.so.1
  NEEDED               libatkmm-1.6.so.1
  NEEDED               libgdkmm-3.0.so.1
  NEEDED               libpangomm-1.4.so.1
  NEEDED               libglibmm-2.4.so.1
  NEEDED               libsigc-2.0.so.0
  NEEDED               libstdc++.so.6
```
We also have the following libraries, which are both our direct dependencies and dependencies of GTK...
```
  NEEDED               libfontconfig.so.1
  NEEDED               libexpat.so.1 ; through libfontconfig
  NEEDED               libuuid.so.1 ; through libfontconfig
  NEEDED               libfreetype.so.6
  NEEDED               libpng16.so.16
  NEEDED               libz.so.1
```
... and things get a bit weird there. All these libraries are guaranteed to be present on the host system, since GTK needs them, and GTK will load them into our address space. But one problematic case would be if GTK loads a version of one of those libraries with a *different soname*. In that case, if the library in question does not use symbol versioning (fontconfig, expat and freetype do not), it might result in a situation where GTK is linked against libfreetype.so.7, both it and libfreetype.so.6 are loaded into our process by the dynamic linker, and it's a coin flip as to which one ends up getting called.

To handle this, out of that list, we should only bundle the libraries that *do* use symbol versioning, namely:
```
  NEEDED               libpng16.so.16
  NEEDED               libz.so.1
```
so that if that case *does* arise, things break quickly and in an obvious way (missing library at startup) as opposed to slowly and in a complicated way (memory corruption or crash at runtime).

probonopd, 2019-05-07:
Oh, and an important one:

Does

> a binary built with GTK+ < 3.22 is guaranteed to run against GTK+ ≥ 3.22

also mean that

a binary built with GTK+ 3.18.9 is guaranteed to run against GTK+ ≥ 3.18.9 but <3.22?

ebassi, 2019-05-14:
Okay, there's *a lot* of ground to cover, here.

> So, you should talk to the GTK people to figure out why that doesn't work.

Theming is problematic.

CSS selectors were made part of the ABI or API in GTK 3.22; theming in general is not considered part of the API outside of the CSS selectors exposed by GTK widgets, which are part of the documentation and thus remain stable. The only theme that GTK guarantees to be working is Adwaita, because it's shipped by GTK itself.

For third party themes: it is possible to ship different versions of the theme for the releases of GTK that didn't have any stability guarantee for the CSS selectors; not every theme does that, though, so third party themes will likely break.

Additionally, GTK does not ship an icon theme, so if you're missing assets then you may be missing an icon theme with those assets from your bundle.

> Question: With "GTK" only libgtk-3.so.0 is meant - or which set of libraries exactly? How about gdk, atk, gtkmm? pango? harfbuzz?

GTK 3.x ships only two shared libraries:

 - libgdk-3.so.* (with the SONAME determined by the library version)
 - libgtk-3.so.* (with the SONAME determined by the library version)

anything else is a dependency of GTK; so if you used a specific version of ATK, Pango, Cairo, HarfBuzz, libX11, GNU libc, to compile GTK, then you need to ship those libraries as well, because GTK may be using a symbol added in Cairo 1.15, so you cannot ship Cairo 1.12.

Of course, if you built your application against GTK 3.18 **and all its dependencies** and then you upgrade to GTK 3.22 **and all its dependencies** then the application will continue to work—assuming none of the dependencies in question broken binary compatibility. Most of the dependencies of GTK are maintained by the same people, so they follow the same conventions; other dependencies are mostly stable/low level, so their ABI doesn't really change. Of course, this only goes in one direction. If, for instance, GTK 3.18 depends on Pango 1.40, and GTK 3.22 depends on Pango 1.44, it usually means that GTK 3.22 needs new symbols from Pango; this means that you can run your application built against a version of GTK with a *newer* version of GTK, but you cannot run your application built against a version of GTK with an *older* version of GTK. I know this is basic stuff, but I wanted to make sure we are all on the same page.

GTKmm is another thing entirely, and since the GTK team does not maintain bindings I honestly cannot say anything about them; the C++ bindings are, as far as I know, ABI stable, and since they provide API wrappers around GTK, then they have to follow the same convention of forward compatibility as GTK itself.

> a binary built with GTK+ 3.18.9 is guaranteed to run against GTK+ > 3.18.9 but < 3.22?

I don't understand the distinction you're trying to make.

A binary built against GTK 3.18.9:

 - is source compatible with all other micro releases of GTK 3.18; no new API is added, no API is deprecated
 - is API and ABI compatible with all *stable* minor releases of GTK: 3.18, 3.20, 3.22, 3.24, …

To sum up:

 - you can build against GTK 3.18 and run against GTK 3.22
 - third party theming is problematic for anything older than GTK 3.22 because CSS selectors and classes were not stable before that release

probonopd, 2019-05-14:
Thank you for the clarification @ebassi. I will try to find what this means for bundling vs. not bundling the concrete dependencies in question.

* We only care about running on systems that are _the same or newer than_ build system, we do not expect to be able to run on older systems than the build system
* Hence, we build on the oldest OS we care about (which for me is the oldest still-supported Ubuntu LTS release, which currently means we compile against GTK+ 3.18.9)
* We will have to bundle everything that we cannot expect to be present on all target systems
* We _have to expect that third party theming may be broken_ because despite having the major number "3", GTK+ 3.18.9 was not ABI stable yet

ebassi, 2019-05-14:
>     * We _have to expect that third party theming may be broken_ because despite having the major number "3", GTK+ 3.18.9 was not ABI stable yet

The **library** is ABI stable; the CSS selectors for third party themes were not part of the interface exposed to application developers, regardless of major or minor version.

probonopd, 2019-09-05:
How do we proceed from here? Make an AppImage that bundles *everything*?

whitequark, 2019-09-05:
No, follow https://github.com/solvespace/solvespace/issues/347#issuecomment-490266711.

probonopd, 2019-09-05:
Looking at https://github.com/AppImage/pkg2appimage/blob/master/excludelist we _should_ already be doing this but I am gonna check again.

probonopd, 2019-09-05:
Theming gets fixed if we do _not_ bundle libgdk-3.so.0 and libgtk-3.so.0 but then it will not run on non-Gtk systems anymore. Acceptable or not?

Symbian9, 2019-09-05:
> Theming gets fixed if we do not bundle libgdk-3.so.0 and libgtk-3.so.0 but then it will not run on non-Gtk systems anymore. Acceptable or not?

If bundle `libgdk-3.so.0` and `libgtk-3.so.0` would be resulted AppImage working on non-Gtk systems just without theming?

If so, then there as solution just provide two versions:

- `SolveSpace-*-Gtk-*-x86_64.AppImage` - for Gtk systems (with theming support);
- `SolveSpace-*-x86_64.AppImage` - for non-Gtk systems (without theming support).

baryluk, 2020-01-16:
I think you should bundle Gtk/Gdk, even if this means loosing theming support, so it works EVERYWHERE forever.

Please provide only single AppImage. The entire point of having it is just to JUST WORK anywhere. Adding two options will be confusing to people, and will be asking question all the time.

@Symbian9, the naming is confusing. The one with "Gtk" in its name, actually doesn't have Gtk, and the one without "Gtk" in its name, actually ships with gtk inside. Nono. Too confusing and unnecessary.

Plus if you provide the version without the Gtk, project dies, and you trie to run it 25 years from now to read some obscure slvs files, you are screwed, because you don't have access to the Gtk dependencies anymore.

Bundle everything.

As of theming, maybe the note at the total end of this guide: https://docs.appimage.org/packaging-guide/manual.html#bundling-gtk-libraries could help?

You can always make things work better in the future, and possibly have a detection script inside AppImage that tries to to use system gtk first, and if not then use its own, based on detection, trying to run it (and dynamic linker failing) or just user flags.

whitequark, 2020-01-16:
> You can always make things work better in the future, and possibly have a detection script inside AppImage that tries to to use system gtk first, and if not then use its own, based on detection, trying to run it (and dynamic linker failing) or just user flags.

That's not possible.

> Please provide only single AppImage. The entire point of having it is just to JUST WORK anywhere. Adding two options will be confusing to people, and will be asking question all the time.

As of today, we provide official Snap packages. If you insist on having just one distro-indepenent package like that, then there won't be an AppImage at all.

> I think you should bundle Gtk/Gdk, even if this means loosing theming support, so it works EVERYWHERE forever.

No software will work "forever" without some maintenance. For example, at some point desktop Linux will migrate to Wayland.

baryluk, 2020-01-16:
> As of today, we provide official Snap packages. If you insist on having just one distro-indepenent package like that, then there won't be an AppImage at all.

I am not sure I understand what you are saying. AppImage IS a distro-independent package. Properly packaged AppImage app will have no dependencies on the host system at all other than the kernel APIs and cross process APIs via sockets or shared memory. And OpenGL driver from the host if you want to use it.

whitequark, 2020-01-16:
> I was hearing that for the last 10 years.

When I decide to use sarcastic comments from nerds to decide on the roadmap instead of looking at the planned future directions of the supported platforms, I'll make sure to ask you for advice.

baryluk, 2020-01-16:
> > I was hearing that for the last 10 years.
> 
> When I decide to use sarcastic comments from nerds to decide on the roadmap instead of looking at the planned future directions of the supported platforms, I'll make sure to ask you for advice.

Even if majority of Linux desktop installations will migrate to Wayland, X11 will still be used for very very very long time.