solvespace/solvespace #356: "New Group -> Link Recent -> foo.slvs" fails with 'The linked file "" is not present."

vcaputo, 2018-09-28:
### System information

SolveSpace version: 3.0~4d1e1341

Operating system: Debian 9.5 amd64

### Expected behavior

What should have happened?

Linked the file

### Actual behavior

What actually happened?

Failed with nonsensical error.

### Additional information

If I instead go through "New Group -> Link / Assemble -> foo.slvs" it succeeds, so it's not a file-specific problem.  (the file in question doesn't reference other .slvs files either, it's a simple cylinder)

In summary:

"Link Recent" fails, showing a 'The linked file "" is not present' dialog, *twice* in quick succession.

A subsequent attempt at linking instead via "Link / Assemble" selecting the same file again shows the 'The linked file "" is not present" dialog, but only once, and succeeds at linking the file.

On a fresh start, going directly to "Link /Assemble" on the same file shows no such dialog and just works.

whitequark, 2018-09-28:
Can you post your `~/.config/solvespace/settings.json` file here?

vcaputo, 2018-09-28:
https://ruiner.halo.nu/~swivel/tmp/solvespace-settings.json

whitequark, 2018-09-28:
Thanks. I can't reproduce that. I know why you might be getting two dialogs in quick succession, but I believe I've fixed all of the bugs like that a while ago, and the code seems right to me.

Can you break in gdb at `LocateImportedFile` and show me the result of `info locals`?

vcaputo, 2018-09-28:
Ok, it looks like I made a mistake with asserting the file actually exists; it doesn't, one of the directories in the path had changed names recently.

However, there is still a bug with the dialog box showing "" for the file path, and there appears to be some kind of stuck state causing the dialog to recur even when loading the existing file.

@whitequark maybe you can reproduce what I'm seeing by renaming/reomving the file before opening it via "Link Recent"?

whitequark, 2019-05-24:
@vcaputo Yes. I can reproduce it now.

whitequark, 2019-05-24:
Fixed in eb7e12b.

closed 2019-05-24 by whitequark