solvespace/solvespace #358: Rotation/zoom changes are not rendered until window is resized

nazar-pc, 2018-09-29:
### System information

SolveSpace version: 3.0\~fb138f49 (from ppa) and also happens in 3.0\~88d59ea7 (compiled myself a while ago)

Operating system: Ubuntuj 18.10

### Expected behavior

Rotation/zoom changes should be rendered immediately

### Actual behavior

They are rendered only after SolveSpace's window resizing

### Additional information

I believe this is a regression I've got after upgrading from Mesa 18.1.5 to 18.2.1, which fixed one bug I had and introduced this one.

whitequark, 2018-09-29:
Indeed, I've been seeing these issues as well. I've spent many days poring over SolveSpace's OpenGL code and found absolutely nothing; half of my debugging tools do not indicate that I'm doing anything wrong at all, and the other perturb the environment just enough to make the bugs disappear. These are very frustrating.

nazar-pc, 2018-09-30:
@diggit are you using compositor? I have Qtile with Compton and no window decorations.
Also our GPU is a bit older, so I guess this issue might be caused for some optimizations for newer generations that are not applied to older ones.

diggit, 2018-09-30:
Usually, I do not use compositor. Now I tested solvespace with compton and xcompmgr using some default config and everything seemed ok. That probably points to HW specific problem or some special SW combination/settings.

whitequark, 2018-10-05:
Closing as upstream bug (see [comment](https://bugs.freedesktop.org/show_bug.cgi?id=108097#c3)).

whitequark, 2018-10-16:
Keeping this open since people are still hitting it...

mkyral, 2018-10-16:
I'm affected as well. On my first computer with nVidia graphics SolveSpace works, but on second with only Intel graphics not.

Both laptops have Gentoo Linux and KDE with compositor (kwin). On both SolveSpace compiled from sources. nVidia one compiled at 22:42:50 1.10.2018. The Intel only I need check tomorrow.

whitequark, 2018-10-16:
Yep. Unfortunately there is nothing I can do as this is an upstream bug. I'm sorry.

mkyral, 2018-10-17:
> Yep. Unfortunately there is nothing I can do as this is an upstream bug. I'm sorry.

Just tested a patch in upstream bug and it fixed SolveSpace behaviour on my i965.

PhilipAnderson, 2019-03-01:
Was this supposed to be fixed?
I have suddenly hit it after updating my Ubuntu 18.04.2, I am running mesa 18.2.2, and Solvespace compiled from source.

whitequark, 2019-03-01:
I have also hit this recently with NVidia graphics. Unfortunately still no idea what causes it. Sorry.

PhilipAnderson, 2019-03-01:
Sorry to keep bumping this old issue, but I just managed to get it working again by using updated drivers from this PPA: https://launchpad.net/~oibaf/+archive/ubuntu/graphics-drivers , hopefully it may help others too.

So now I can use Solvespace again :)

Thanks for your work on this great program.

philtipping, 2019-03-01:
This also worked for me on Lubuntu 18.04 on Dell Vostro laptop - can
finally start exploring v3 :)
Thanks guys for all the great work.
Phil

supertaz, 2019-06-07:
Thus far, I haven't hit this, except for the menu bug.  I'm using Ubuntu 18.04 and NVidia RTX cards.

whitequark, 2019-06-08:
I'm closing this because it's a fixed upstream bug I can't do anything about, but I'm also pinning it so that people can see it easily if they also have that problem.

closed 2019-06-08 by whitequark