solvespace/solvespace #365: Tooltips bug on Gentoo Linux

brothermechanic, 2018-11-09:
### System information
 i7 + gtx1060 kde5
SolveSpace version: (e.g. 3.0~3dd2fc00; go to Help → About...)
https://data.gpo.zugaina.org/booboo/media-gfx/solvespace/ git package 52c4178 commit
Operating system: (e.g. Debian testing)
Gentoo ~amd64 
### Expected behavior
?
What should have happened?
I want to see tooltips)
### Actual behavior
?
What actually happened?
Tooltips don't showings.
Please see my video screencast
[2018-11-09 20-57-00.zip](https://github.com/solvespace/solvespace/files/2566990/2018-11-09.20-57-00.zip)
### Additional information
Do you need my nvidia settings info? (other 3d soft works fine)
For bugs, please attach a savefile that shows the problematic behavior.
You can attach `.slvs` files by archiving them into a `.zip` first.

whitequark, 2018-11-09:
Thanks for the report. I believe this is a GTK issue, though it will need to be confirmed. What is your GTK theme?

mkyral, 2018-11-09:
Hi,
Im using Solvespace 3.0~258545a3 on Gentoo ~amd64, GT 740M (nvidia-drivers 410.73), Kde5 (platform 5.51, plasma 5.14.3), gtk theme Breeze-Dark and tooltips works.

brothermechanic, 2018-11-09:
hello,  mkyral 
my packages are same
gtk theme Breeze-Dark
do you saw my screencast in attachment?

ralvgh, 2018-11-12:
###  Possible Workaround

Having had the same problem (void linux, gtk+3-3.24.1, solvespace v3~8f2af5b2, various themes) the problem vanished when using the workaround from [this gtk issue](https://gitlab.gnome.org/GNOME/gtk/issues/1371): 

> cat ~/.config/gtk-3.0/settings.ini
> [Settings]
> gtk-cursor-theme-size=16

brothermechanic, 2018-11-12:
@ralvgh ! Works!

whitequark, 2018-11-12:
Resolution: upstream bug, no action needed. Keeping open until GTK issue is closed.

brothermechanic, 2018-11-12:
got to Qt!)))

whitequark, 2019-05-13:
According to the GTK bugtracker this is fixed on X11, which is really all we care about here.

closed 2019-05-13 by whitequark