solvespace/solvespace #367: DXF/DWG import silently fails

arvidbrodin, 2018-11-17:
### System information

SolveSpace version: 3.0~8f2af5b2
Operating system: Gentoo 2018-11-12

### Expected behavior
I'm trying to use CAD representations of ball bearings from [Schaeffler](https://medias.schaeffler.com/medias/en!hp.ec.br.cad/62*6206). For example [this FAG 6206 (zip)](https://github.com/solvespace/solvespace/files/2592363/80294414-37-inafag_6206_1r96qhnoprhgd7ga3gdtxq3j3.zip).

After unzipping the file and opening it with the Open file dialog reached by File->Import... (in a new Solvespace sketch), I expect a 3D model of the bearing to appear.

### Actual behavior
After selecting the file and clicking Open, the Open file dialog disappears and I am left with the same empty sketch as before. I even tried saving an empty sketch as an .slvs file and then saving again to another file after doing the import; the two files are identical according to 'diff -q'.

### Additional information
The same behaviour is seen with both of the 3D DWG formats as well as the 3D DXF format available on the site, and also when downloading files for another bearing (7206).

I tried to debug this a bit, and it appears that iface->addInsert() is never called (libdxfrw.cpp:2443). I thought that was a bit strange, but I'm not sure if it is significant.

whitequark, 2018-11-18:
Note that LibreCAD (which is the project of origin for _libdxfrw_) exhibits the exact same behavior, so it is unlikely that SolveSpace would be able to easily fix this bug.

Symbian9, 2019-09-29:
Just got same behavior with *SolveSpace* `3.0.0~5d78f993` when try import next DWG:

- https://github.com/GaelReinaudi/Aerobatics-Plane-CAD/blob/master/gael-plane-CAD.dwg