solvespace/solvespace #368: Crash while analyzing degrees of freedom

cbossut, 2018-11-26:
### System information

SolveSpace version: 2.3~7c1ca460.
Operating system: Windows 10

### Expected behavior

Show degrees of freedom ! Or at least a nice error :-)

Also, as a feature request (I can file it in another issue if you want, found advice 'bout that) :
I would like to see the degrees of freedom only of a selected group.
In that case, I need to find why on of my group still has DOF, but it crashes when analyzing a previous group ...

### Actual behavior

It show "generating group x/36" and at group 11, it stucks then crashes silently. Crashes as close by itself without warning.

### Additional information

So far, I've checked the tasks, and solvespace is using like 20% proc and 1G RAM before crashing.
I've tried to check "treat all dimansions as references" in the suspect group to make it easier to analyze, and it changed nothing.
By the way, I don't know if the groups are generated by order in the list or group number, so i'm not sure which group is guilty, but I bet on g00b more than on g00a which is eleventh.
g00b is really complex, but has 0 DOF, so I wonder if the solver is recalculating everything to check there's truly 0 DOF.
g00a is quite simplistic and has 1 DOF, only a point moving on a segment, no big reason to crash.
[Chapiteau vue de dessus.zip](https://github.com/solvespace/solvespace/files/2614831/Chapiteau.vue.de.dessus.zip)

whitequark, 2018-11-26:
Thanks for the report. Can you please recheck with a latest binary from [here](https://ci.appveyor.com/project/whitequark/solvespace/build/artifacts)?

whitequark, 2020-06-23:
I can reproduce the crash on current master. Maybe moving to Eigen will solve this, maybe not.

phkahler, 2023-08-09:
This shows DoF on most groups now without crashing.

closed 2023-08-09 by phkahler

ruevs, 2023-08-11:
Yes but `Analyze -> Show Unconstrained Points` (`Ctrl+Shift_F`) is veeeery slow and the fact that it does not crash is probably luck?