solvespace/solvespace #370: linpng-12 unmet build dependency on ubuntu 16.04

rshpeley, 2018-12-18:
### System information

SolveSpace version: latest github

Operating system: 
elementary OS 0.4.1 Loki
Built on "Ubuntu 16.04.5 LTS"
Linux 4.4.0-140-generic
GTK+ 3.18.9

### Expected behavior

Should have install libpng-dev from this command given in the SolveSpace README, Building for Linux...
```shell
sudo apt install git build-essential cmake zlib1g-dev libpng-dev \
                 libcairo2-dev libfreetype6-dev libjson-c-dev \
                 libfontconfig1-dev libgtkmm-3.0-dev libpangomm-1.4-dev \
                 libgl-dev libglu-dev libspnav-dev

```
### Actual behavior

Got the status and error...
```shell

sudo apt install git build-essential cmake zlib1g-dev libpng-dev \
                  libcairo2-dev libfreetype6-dev libjson-c-dev \
                  libfontconfig1-dev libgtkmm-3.0-dev libpangomm-1.4-dev \
                  libgl-dev libglu-dev libspnav-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libpng12-dev' instead of 'libpng-dev'
Note, selecting 'libgl1-mesa-dev' instead of 'libgl-dev'
Note, selecting 'libglu1-mesa-dev' instead of 'libglu-dev'
build-essential is already the newest version (12.1ubuntu2).
zlib1g-dev is already the newest version (1:1.2.8.dfsg-2ubuntu4.1).
zlib1g-dev set to manually installed.
git is already the newest version (1:2.20.1-0ppa1~ubuntu16.04.1).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help resolve the situation:

The following packages have unmet dependencies:
 libpng12-dev : Depends: libpng12-0 (= 1.2.54-1ubuntu1.1) but 1.2.54-5x0~ppa is to be installed
E: Unable to correct problems, you have held broken packages.

```
### Additional information

I'm running gimp 2.9.9 development release which may have specific requirements for libpng-dev. Any workarounds on this?

whitequark, 2018-12-18:
No idea, especially given that you haven't provided the list of package sources you use.

rshpeley, 2018-12-18:
from /etc/apt/sources.list

```shell

# deb cdrom:[elementary OS 0.4 _Loki_ - Stable amd64 (20160921)]/ xenial contrib main non-free

# See http://help.ubuntu.com/community/UpgradeNotes for how to upgrade to
# newer versions of the distribution.
deb http://ca.archive.ubuntu.com/ubuntu/ xenial main restricted
# deb-src http://ca.archive.ubuntu.com/ubuntu/ xenial main restricted

## Major bug fix updates produced after the final release of the
## distribution.
deb http://ca.archive.ubuntu.com/ubuntu/ xenial-updates main restricted
# deb-src http://ca.archive.ubuntu.com/ubuntu/ xenial-updates main restricted

## N.B. software from this repository is ENTIRELY UNSUPPORTED by the Ubuntu
## team, and may not be under a free licence. Please satisfy yourself as to
## your rights to use the software. Also, please note that software in
## universe WILL NOT receive any review or updates from the Ubuntu security
## team.
deb http://ca.archive.ubuntu.com/ubuntu/ xenial universe
# deb-src http://ca.archive.ubuntu.com/ubuntu/ xenial universe
deb http://ca.archive.ubuntu.com/ubuntu/ xenial-updates universe
# deb-src http://ca.archive.ubuntu.com/ubuntu/ xenial-updates universe

## N.B. software from this repository is ENTIRELY UNSUPPORTED by the Ubuntu 
## team, and may not be under a free licence. Please satisfy yourself as to 
## your rights to use the software. Also, please note that software in 
## multiverse WILL NOT receive any review or updates from the Ubuntu
## security team.
deb http://ca.archive.ubuntu.com/ubuntu/ xenial multiverse
# deb-src http://ca.archive.ubuntu.com/ubuntu/ xenial multiverse
deb http://ca.archive.ubuntu.com/ubuntu/ xenial-updates multiverse
# deb-src http://ca.archive.ubuntu.com/ubuntu/ xenial-updates multiverse

## N.B. software from this repository may not have been tested as
## extensively as that contained in the main release, although it includes
## newer versions of some applications which may provide useful features.
## Also, please note that software in backports WILL NOT receive any review
## or updates from the Ubuntu security team.
deb http://ca.archive.ubuntu.com/ubuntu/ xenial-backports main restricted universe multiverse
# deb-src http://ca.archive.ubuntu.com/ubuntu/ xenial-backports main restricted universe multiverse

## Uncomment the following two lines to add software from Canonical's
## 'partner' repository.
## This software is not part of Ubuntu, but is offered by Canonical and the
## respective vendors as a service to Ubuntu users.
# deb http://archive.canonical.com/ubuntu xenial partner
# deb-src http://archive.canonical.com/ubuntu xenial partner

deb http://security.ubuntu.com/ubuntu xenial-security main restricted
# deb-src http://security.ubuntu.com/ubuntu xenial-security main restricted
deb http://security.ubuntu.com/ubuntu xenial-security universe
# deb-src http://security.ubuntu.com/ubuntu xenial-security universe
deb http://security.ubuntu.com/ubuntu xenial-security multiverse
# deb-src http://security.ubuntu.com/ubuntu xenial-security multiverse
deb https://dl.winehq.org/wine-builds/ubuntu/ xenial main
# deb-src https://dl.winehq.org/wine-builds/ubuntu/ xenial main

```

whitequark, 2018-12-18:
There's also sources.list.d. And I'll need to see where `1.2.54-5x0~ppa` comes from.

rshpeley, 2018-12-18:
from sources.list.d...

```shell

ls /etc/apt/sources.list.d
appcenter.list                                        libreoffice-ubuntu-ppa-xenial.list.save
appcenter.list.save                                   mozillateam-ubuntu-thunderbird-next-xenial.list
docker.list                                           mozillateam-ubuntu-thunderbird-next-xenial.list.save
docker.list.save                                      nodesource.list.save
elementary.list                                       opera-stable.list
elementary.list.save                                  opera-stable.list.save
git-core-ubuntu-ppa-xenial.list                       otto-kesselgulasch-ubuntu-gimp-edge-xenial.list
git-core-ubuntu-ppa-xenial.list.save                  otto-kesselgulasch-ubuntu-gimp-edge-xenial.list.save
google-chrome.list                                    patches.list
google-chrome.list.save                               patches.list.save
inkscape_dev-ubuntu-stable-xenial.list                sublime-text.list
inkscape_dev-ubuntu-stable-xenial.list.save           sublime-text.list.save
jonathonf-ubuntu-texlive-xenial.list                  vscode.list
jonathonf-ubuntu-texlive-xenial.list.save             vscode.list.save
js-reynaud-ubuntu-kicad-dev-nightly-xenial.list       wireshark-dev-ubuntu-stable-xenial.list
js-reynaud-ubuntu-kicad-dev-nightly-xenial.list.save  wireshark-dev-ubuntu-stable-xenial.list.save
libreoffice-ubuntu-ppa-xenial.list

```
and the individual sources, ignoring the .save (except nodesource which shouldn't matter)

```shell

cat /etc/apt/sources.list.d/appcenter.list
deb http://packages.elementary.io/appcenter xenial main
cat /etc/apt/sources.list.d/docker.list
deb [arch=amd64] https://download.docker.com/linux/ubuntu xenial edge
cat /etc/apt/sources.list.d/elementary.list
deb http://ppa.launchpad.net/elementary-os/stable/ubuntu xenial main
deb-src http://ppa.launchpad.net/elementary-os/stable/ubuntu xenial main
cat /etc/apt/sources.list.d/git-core-ubuntu-ppa-xenial.list
deb http://ppa.launchpad.net/git-core/ppa/ubuntu xenial main
cat /etc/apt/sources.list.d/google-chrome.list
deb [arch=amd64] http://dl.google.com/linux/chrome/deb/ stable main
cat /etc/apt/sources.list.d/inkscape_dev-ubuntu-stable-xenial.list
deb http://ppa.launchpad.net/inkscape.dev/stable/ubuntu xenial main
cat /etc/apt/sources.list.d/jonathonf-ubuntu-texlive-xenial.list
deb http://ppa.launchpad.net/jonathonf/texlive/ubuntu xenial main
cat /etc/apt/sources.list.d/js-reynaud-ubuntu-kicad-dev-nightly-xenial.list
deb http://ppa.launchpad.net/js-reynaud/kicad-dev-nightly/ubuntu xenial main
cat /etc/apt/sources.list.d/libreoffice-ubuntu-ppa-xenial.list
deb http://ppa.launchpad.net/libreoffice/ppa/ubuntu xenial main
cat /etc/apt/sources.list.d/mozillateam-ubuntu-thunderbird-next-xenial.list
deb http://ppa.launchpad.net/mozillateam/thunderbird-next/ubuntu xenial main
cat /etc/apt/sources.list.d/nodesource.list.save
deb https://deb.nodesource.com/node_9.x xenial main
deb-src https://deb.nodesource.com/node_9.x xenial main
cat /etc/apt/sources.list.d/opera-stable.list
deb https://deb.opera.com/opera-stable/ stable non-free #Opera Browser (final releases)
cat /etc/apt/sources.list.d/otto-kesselgulasch-ubuntu-gimp-edge-xenial.list
deb http://ppa.launchpad.net/otto-kesselgulasch/gimp-edge/ubuntu xenial main
cat /etc/apt/sources.list.d/patches.list
deb http://ppa.launchpad.net/elementary-os/os-patches/ubuntu xenial main
deb-src http://ppa.launchpad.net/elementary-os/os-patches/ubuntu xenial main
cat /etc/apt/sources.list.d/sublime-text.list
deb https://download.sublimetext.com/ apt/stable/
cat /etc/apt/sources.list.d/vscode.list
deb [arch=amd64] http://packages.microsoft.com/repos/vscode stable main
cat /etc/apt/sources.list.d/wireshark-dev-ubuntu-stable-xenial.list
deb http://ppa.launchpad.net/wireshark-dev/stable/ubuntu xenial main

```

Still working on `1.2.54-5x0~ppa`.

rshpeley, 2018-12-18:
`1.2.54-5x0~ppa` looks to be from the gimp development version maintained by otto-kesselgulasch. 

Found these links,
<https://launchpad.net/~otto-kesselgulasch/+archive/ubuntu/gimp-edge/+build/11878284>
<https://launchpad.net/~otto-kesselgulasch/+archive/ubuntu/gimp-edge/+sourcepub/7410499/+listing-archive-extra>

whitequark, 2018-12-18:
Try installing -dev package directly from there.

rshpeley, 2018-12-18:
hmmm... that first link was for arm64

still digging...

rshpeley, 2018-12-18:
ok, so I installed this ppa (which is probably the one you meant),
<https://launchpad.net/~otto-kesselgulasch/+archive/ubuntu/gimp-edge/+files/libpng12-dev_1.2.54-5x0~ppa_amd64.deb>

...then ran the install which has completed. 

I'd like to leave this question open until I complete the SolveSpace install and see if the ppa broke anything on Gimp. Thanks @whitequark !

rshpeley, 2018-12-18:
SolveSpace built and installed ok, but won't start from the gui. When I run from the command line I get,

```

$ solvespace
solvespace: error while loading shared libraries: libpng16.so.16: cannot open shared object file: No such file or directory

```

Which seems to be related. I installed `libpng12-dev` and solvespace wants `libpng16`.

rshpeley, 2018-12-18:
And gimp 2.9.9 continues to work with `libpng16-16` installed.

To summarise, the install error,

```

The following packages have unmet dependencies:
 libpng12-dev : Depends: libpng12-0 (= 1.2.54-1ubuntu1.1) but 1.2.54-5x0~ppa is to be installed
E: Unable to correct problems, you have held broken packages.

```

needed the ppa at,

<https://launchpad.net/~otto-kesselgulasch/+archive/ubuntu/gimp-edge/+files/libpng12-dev_1.2.54-5x0~ppa_amd64.deb>

and solvespace needed the `libpng16-16` lib which was installed with,

`sudo apt install libpng16-16`

closing this question...

closed 2018-12-18 by rshpeley