solvespace/solvespace #373: HighDPI works incorrect on Win10_64bit with 200% scaling

konsultaner, 2019-01-13:
### System information

SolveSpace version: 2.3~7c1ca460

Operating system: Win10 64bit

### Expected behavior

I should see the workspace in HighDPI and my mousepointer should have no offset.

### Actual behavior

HighDPI works incorrect on Win10_64bit with 200% scaling.

### Additional information

[image]

whitequark, 2019-01-13:
There is no High-DPI support in 2.x versions. High-DPI is only present in the 3.x versions; you can find nightly builds [here](https://ci.appveyor.com/project/whitequark/solvespace).

konsultaner, 2019-01-13:
now it works :)

closed 2019-01-13 by konsultaner