solvespace/solvespace #374: Fresh compiled binary for v3.0 does not work on voidlinux.

imrn, 2019-01-31:
2019-01-31. I've checked out and compiled the source on VoidLinux.
Running solvespace binary gives this message:

  SolveSpace!
  glGetError() == 0x501
  glGetError() == 0x501
  glGetError() == 0x501
  No provider of glGetShaderiv found.  Requires one of:
    Desktop OpenGL 2.0
    OpenGL ES 2.0
  Aborted

Partial glxinfo is here:
OpenGL ES profile version string: OpenGL ES 2.0 Mesa 18.3.2
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 1.0.16

I've manually downloaded, extracted debian package
solvespace_2.3+repack1-3+b1_amd64.deb. It works fine
with this trick: Symlink libjson-c.so.3 to libjson-c.so.4.

imrn, 2019-01-31:
+ solvespace-cli 3.0 runs fine.

whitequark, 2019-01-31:
Are you using X11 or Wayland? What graphics card do you use?

imrn, 2019-01-31:
X11. Intel i915.
CPU0: Intel(R) Atom(TM) CPU N570   @ 1.66GHz (family: 0x6, model: 0x1c, stepping: 0xa)

Relevant sections of Xorg.log.0:
```
X.Org X Server 1.20.3
X Protocol Version 11, Revision 0
[ 12019.226] Build Operating System: Linux Void
[ 12019.229] Current Operating System: Linux void 4.20.3-46 #1 SMP PREEMPT Sat Jan 19 16:46:45 +03 2019 x86_64
[ 12019.257] Build Date: 27 October 2018  12:31:17PM
[ 12019.296] (II) Module ABI versions:
[ 12019.296] 	X.Org ANSI C Emulation: 0.4
[ 12019.296] 	X.Org Video Driver: 24.0
[ 12019.296] 	X.Org XInput driver : 24.1
[ 12019.296] 	X.Org Server Extension : 10.0
[ 12019.297] (II) xfree86: Adding drm device (/dev/dri/card0)
[ 12019.317] (--) PCI:*(0@0:2:0) 8086:a011:1043:83ac rev 2, Mem @ 0xf7e00000/524288, 0xd0000000/268435456, 0xf7d00000/1048576, I/O @ 0x0000dc00/8, BIOS @ 0x????????/131072
[ 12019.317] (--) PCI: (0@0:2:1) 8086:a012:1043:83ac rev 2, Mem @ 0xf7e80000/524288
[ 12019.317] (II) LoadModule: "glx"
[ 12019.318] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[ 12019.335] (II) Module glx: vendor="X.Org Foundation"
[ 12019.335] 	compiled for 1.20.3, module version = 1.0.0
[ 12019.335] 	ABI class: X.Org Server Extension, version 10.0
[ 12019.335] (==) Matched intel as autoconfigured driver 0
[ 12019.336] (==) Matched modesetting as autoconfigured driver 1
[ 12019.336] (==) Matched fbdev as autoconfigured driver 2
[ 12019.336] (==) Matched vesa as autoconfigured driver 3
[ 12019.336] (==) Assigned the driver to the xf86ConfigLayout
[ 12019.336] (II) LoadModule: "intel"
[ 12019.336] (II) Loading /usr/lib/xorg/modules/drivers/intel_drv.so
[ 12019.337] (II) Module intel: vendor="X.Org Foundation"
[ 12019.337] 	compiled for 1.20.1, module version = 2.99.917
[ 12019.337] 	Module class: X.Org Video Driver
[ 12019.337] 	ABI class: X.Org Video Driver, version 24.0
[ 12019.337] (II) LoadModule: "modesetting"
[ 12019.337] (II) Loading /usr/lib/xorg/modules/drivers/modesetting_drv.so
[ 12019.355] (II) Module modesetting: vendor="X.Org Foundation"
[ 12019.355] 	compiled for 1.20.3, module version = 1.20.3
[ 12019.355] 	Module class: X.Org Video Driver
[ 12019.355] 	ABI class: X.Org Video Driver, version 24.0
[ 12019.357] (II) intel: Driver for Intel(R) Integrated Graphics Chipsets:
	i810, i810-dc100, i810e, i815, i830M, 845G, 854, 852GM/855GM, 865G,
	915G, E7221 (i915), 915GM, 945G, 945GM, 945GME, Pineview GM,
	Pineview G, 965G, G35, 965Q, 946GZ, 965GM, 965GME/GLE, G33, Q35, Q33,
	GM45, 4 Series, G45/G43, Q45/Q43, G41, B43
[ 12019.359] (II) intel: Driver for Intel(R) HD Graphics
[ 12019.359] (II) intel: Driver for Intel(R) Iris(TM) Graphics
[ 12019.359] (II) intel: Driver for Intel(R) Iris(TM) Pro Graphics
[ 12019.359] (II) modesetting: Driver for Modesetting Kernel Drivers: kms
[ 12019.359] (--) using VT number 7
[ 12019.379] (II) intel(0): Using Kernel Mode Setting driver: i915, version 1.6.0 20180921
[ 12019.379] (II) intel(0): SNA compiled from 2.99.917-847-g25c9a2fc
[ 12019.395] (WW) Falling back to old probe method for modesetting
[ 12019.396] (WW) VGA arbiter: cannot open kernel arbiter, no multi-card support
[ 12019.396] (--) intel(0): Integrated Graphics Chipset: Intel(R) Pineview GM
[ 12019.396] (--) intel(0): CPU: x86-64, sse2, sse3, ssse3; using a maximum of 2 threads
[ 12019.396] (II) intel(0): Creating default Display subsection in Screen section
	"Default Screen Section" for depth/fbbpp 24/32
[ 12019.397] (==) intel(0): Depth 24, (--) framebuffer bpp 32
[ 12019.397] (==) intel(0): RGB weight 888
[ 12019.397] (==) intel(0): Default visual is TrueColor
[ 12019.397] (II) intel(0): Output LVDS1 has no monitor section
[ 12019.398] (**) intel(0): Found backlight control interface intel_backlight (type 'raw') for output LVDS1
[ 12019.398] (II) intel(0): Enabled output LVDS1
[ 12019.398] (II) intel(0): Output VGA1 has no monitor section
[ 12019.398] (II) intel(0): Enabled output VGA1
[ 12019.399] (--) intel(0): Using a maximum size of 256x256 for hardware cursors
[ 12019.399] (II) intel(0): Output VIRTUAL1 has no monitor section
[ 12019.399] (II) intel(0): Enabled output VIRTUAL1
[ 12019.399] (--) intel(0): Output LVDS1 using initial mode 1024x600 on pipe 1
[... 36 lines cut ...]
[ 12019.416] (II) Initializing extension XFIXES
[ 12019.417] (II) Initializing extension RENDER
[ 12019.417] (II) Initializing extension RANDR
[ 12019.419] (II) Initializing extension COMPOSITE
[ 12019.419] (II) Initializing extension DAMAGE
[ 12019.420] (II) Initializing extension MIT-SCREEN-SAVER
[ 12019.420] (II) Initializing extension DOUBLE-BUFFER
[ 12019.421] (II) Initializing extension RECORD
[ 12019.422] (II) Initializing extension DPMS
[ 12019.422] (II) Initializing extension Present
[ 12019.423] (II) Initializing extension DRI3
[ 12019.423] (II) Initializing extension X-Resource
[ 12019.424] (II) Initializing extension XVideo
[ 12019.424] (II) Initializing extension XVideo-MotionCompensation
[ 12019.425] (II) Initializing extension GLX
[ 12019.447] (II) AIGLX: Loaded and initialized i915
[ 12019.447] (II) GLX: Initialized DRI2 GL provider for screen 0
[ 12019.447] (II) Initializing extension XFree86-VidModeExtension
[ 12019.448] (II) Initializing extension XFree86-DGA
[ 12019.449] (II) Initializing extension XFree86-DRI
[ 12019.449] (II) Initializing extension DRI2
[ 12019.455] (II) intel(0): switch to mode 1024x600@60.0 on LVDS1 using pipe 1, position (0, 0), rotation normal, reflection none
[ 12019.455] (II) intel(0): Setting screen physical size to 270 x 158

```

whitequark, 2019-01-31:
OK. I think you need to install GLX. It is not normal that a desktop X11 installation only has EGL. I do not think this is a SolveSpace issue; the message seems to be printed by some libGL dispatch mechanism on your system.

closed 2019-01-31 by whitequark

imrn, 2019-01-31:
I guess it is installed. Full glxinfo output:

```
name of display: :0.0
display: :0  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
server glx extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_no_error, 
    GLX_ARB_create_context_profile, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_libglvnd, GLX_EXT_no_config_context, 
    GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, GLX_EXT_visual_rating, 
    GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, GLX_OML_swap_method, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
client glx extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_profile, GLX_ARB_create_context_robustness, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, 
    GLX_ARB_get_proc_address, GLX_ARB_multisample, GLX_EXT_buffer_age, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, 
    GLX_EXT_visual_rating, GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
    GLX_MESA_multithread_makecurrent, GLX_MESA_query_renderer, 
    GLX_MESA_swap_control, GLX_OML_swap_method, GLX_OML_sync_control, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
GLX version: 1.4
GLX extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_profile, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, 
    GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, 
    GLX_EXT_visual_rating, GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
    GLX_MESA_query_renderer, GLX_MESA_swap_control, GLX_OML_swap_method, 
    GLX_OML_sync_control, GLX_SGIS_multisample, GLX_SGIX_fbconfig, 
    GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Intel Open Source Technology Center (0x8086)
    Device: Mesa DRI Intel(R) Pineview M  (0xa011)
    Version: 18.3.2
    Accelerated: yes
    Video memory: 384MB
    Unified memory: yes
    Preferred profile: compat (0x2)
    Max core profile version: 0.0
    Max compat profile version: 1.4
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 2.0
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) Pineview M 
OpenGL version string: 1.4 Mesa 18.3.2
OpenGL extensions:
    GL_3DFX_texture_compression_FXT1, GL_AMD_shader_trinary_minmax, 
    GL_ANGLE_texture_compression_dxt3, GL_ANGLE_texture_compression_dxt5, 
    GL_APPLE_object_purgeable, GL_APPLE_packed_pixels, 
    GL_ARB_ES2_compatibility, GL_ARB_clear_buffer_object, 
    GL_ARB_compressed_texture_pixel_storage, GL_ARB_copy_buffer, 
[... 66 lines cut ...]
    GL_OES_draw_elements_base_vertex, GL_OES_element_index_uint, 
    GL_OES_fbo_render_mipmap, GL_OES_get_program_binary, GL_OES_mapbuffer, 
    GL_OES_packed_depth_stencil, GL_OES_required_internalformat, 
    GL_OES_rgb8_rgba8, GL_OES_stencil8, GL_OES_surfaceless_context, 
    GL_OES_texture_3D, GL_OES_texture_border_clamp, GL_OES_texture_npot, 
    GL_OES_vertex_array_object

29 GLX Visuals
    visual  x   bf lv rg d st  colorbuffer  sr ax dp st accumbuffer  ms  cav
  id dep cl sp  sz l  ci b ro  r  g  b  a F gb bf th cl  r  g  b  a ns b eat
----------------------------------------------------------------------------
0x020 24 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
... blah blah

41 GLXFBConfigs:
    visual  x   bf lv rg d st  colorbuffer  sr ax dp st accumbuffer  ms  cav
  id dep cl sp  sz l  ci b ro  r  g  b  a F gb bf th cl  r  g  b  a ns b eat
----------------------------------------------------------------------------
0x05c  0 tc  0  16  0 r  y .   5  6  5  0 .  .  0  0  0  0  0  0  0  0 0 None
... blah blah

```
 

```

whitequark, 2019-01-31:
What is your libGL pointing to?

imrn, 2019-01-31:
libGL.so.1.2.0

imrn, 2019-01-31:
More specifically
libGL.so -> libGL.so.1-> libGL.so.1.2.0

whitequark, 2019-01-31:
no, I mean, what implementation is it? it seems like your libGL only provides EGL functions, not desktop GL. the message implies it's some sort of libGL multiplexer.

whitequark, 2019-01-31:
That's strange. I suggest asking about this in Void Linux support channels as I don't think I can help you any further.

imrn, 2019-01-31:
Thanks. VoidLinux package request is submitted and it refers to
this issue.

https://github.com/void-linux/void-packages/issues/8037

I realized that other distros all have solvespave v2.3 packages.

Does anyone have any pointers to a v3.0 (git version) package on
any distro that can be an submitted to VoidLinux as an example?

whitequark, 2019-01-31:
NixOS maybe has one?

bobertlo, 2019-01-31:
Popping back from void-packages, this is a stable package repo. I don't think they would package the 3.0 git release. I am happy to look at packaging 2.3, but for solverspace/libdxfrw I'd need a release to pull and build in tree for the package. Void really does not do git releases.