solvespace/solvespace #375: Solvespace API

imrn, 2019-01-31:
Can solvespace expose its internal modelling functions as an
API so that it makes programmatic CAD model generation possible?

For example https://docs.blender.org/api/current/ enables
headless usage. It is a large api and uses python.
(There is also Openscad.)

I do not think it is a good idea to embed python in solvespace.
But just stripping solvespace's GUI functions and exposing (and
documenting) modeling functions as cpp library will enable many
interesting options.

Infact, exposing constraints library is already a similar move.
Is it also possible for modelling functions?

whitequark, 2019-01-31:
No. SolveSpace's internal modeling functions assume that they are always called with correct arguments. If this is not the case, the entire application (or anything that embeds SolveSpace) will abort.

Doing what you want such that it is actually usable would require a complete restructure of the entire application.

imrn, 2019-01-31:
Can it be possible in the future version?
Infact, I opened it as a Feature Request.

I see mixed manual and programmed modelling
as the logical direction for the evolution of CAD systems.

whitequark, 2019-01-31:
This is an obvious improvement, but there are two issues.

 1. It would only become possible after there is a Python API (because there are more constraints--SolveSpace's API would have to play nice in other applications, as opposed to playing nice with itself).
 2. This is an enormous amount of work. Likely more than a full rewrite from scratch.

As there is no one at the moment who can invest several person-years of work into refactoring SolveSpace, you should not expect this to happen any time soon.

closed 2019-01-31 by whitequark

imrn, 2019-01-31:
I did NOT request a Python API. It was just an example.
c++ is fine.

imrn, 2019-01-31:
Besides, if anything which plays nice with outside c++ code exists,
then it will become trivial to build any wrapper around it.

whitequark, 2019-01-31:
>  I did NOT request a Python API. It was just an example.

A Python or Lua API is something that has long been planned.

imrn, 2019-01-31:
> A Python or Lua API is something that has long been planned.

I'd appreciate it. However, I'd personally avoid an additional
runtime in between. I think, layered approach would be better.
That is exposing C++ first.

After that, providing python, Lua, whatever... layer becomes trivial.

whitequark, 2019-01-31:
Unfortunately, converting a standalone C++ application into an usable embeddable library is a task far more complex than the already extremely complex task of adding error handling to external APIs. As such I do not see it as a priority.

imrn, 2019-01-31:
> Unfortunately, converting a standalone C++ application into an usable
> embeddable library is a task far more complex than the already extremely
> complex task of adding error handling to external APIs. As such I do not see
> it as a priority.

That's understandable. But I'd love the see this issue as "Open"
even with a low priority.

whitequark, 2019-01-31:
As I currently see it, there's essentially no chance this will happen. There's no point in making the issue tracker a wishlist of items that will not happen.