solvespace/solvespace #378: All solid edges shown as occluded

Symbian9, 2019-02-09:
### System information

SolveSpace version: 3.0~10123738

Operating system: Debian 9.x Stretch

### Expected behavior

Only really occluded edges should be shown as hidden

### Actual behavior

All solid edges shown as occluded

### Additional information

[image]

[image]

[image]

Evil-Spirit, 2019-02-10:
@whitequark probably here is the problem caused by https://github.com/solvespace/solvespace/commit/e383b7fba8fbad23ac370ea5731a87b7eacdf6c8

whitequark, 2019-02-10:
@Evil-Spirit No, it was a problem as far as I can remember, I just never paid attention to it. It renders the same way if I revert e383b7f.

phkahler, 2020-09-14:
On my system it looks like the occluded lines are drawn by drawing all lines (in whichever style) without Z-buffer, and then drawing all of them again in normal style with the Z-buffer enabled. It looks like some kind of fighting between the two line styles rather than only using the right one for each edge. Is that what's happening? @Evil-Spirit ?

Evil-Spirit, 2020-09-16:
Probably, I can blame glDepthRangef follwing after this. I suppose for hidden lines it should be performed in a different way.

Evil-Spirit, 2020-09-16:
Also, depth write probably should be switched off  (glDepthMask(GL_FALSE)) for non-occluded edges.

phkahler, 2020-09-16:
@Evil-Spirit its behaving like depth testing is turned off for occluded edges. And yes, depth write should probably be off for all edges.

phkahler, 2020-09-16:
The problem has to be that there are multiple areas in the code affecting this. I can go in Group::Draw(Canvas *canvas) and comment out the entire section enclosed by  if(SS.GW.showEdges) {  ....  } and I'm still able to use the toggle buttons to produce most of the options including showing the occluded lines. The bug is because a lot of this section of code is redundant in some way - I just haven't figured out what's stepping on it.

phkahler, 2020-09-17:
OK, commenting out everyting but:
```
    GenerateDisplayItems();
    DrawMesh(DrawMeshAs::DEFAULT, canvas);
```
It's still drawing almost all edges (no silhouettes) and allowing 3 options for them with zero calls to DrawOutlines. This is also enabled/disabled via the "show all / hide all" links in the text window. I thought those were only supposed to toggle entities and the toolbar buttons determined which edges are visible and what style.