solvespace/solvespace #383: solvespace hangs trying to open a corrupt file

pmarches, 2019-02-22:
### System information

SolveSpace version: (e.g. 3.0~3dd2fc00; go to Help → About...)
SolveSpace version 2.3~.

Operating system: (e.g. Debian testing)
Ubuntu

### Expected behavior

What should have happened?
I should be able to open the file. 

### Actual behavior

What actually happened?
I opened the file, changed the radius of a tube, then saved. Solvespace hung on the save. I killed the process. Tried re-opening the same file but solvespace now hangs with high CPU usage on the file open. I have to kill the process.

stdout on the file open operation:
pmarches@pop-os:~/projects/sunsplash-cad/ss$ solvespace davits-tubing.slvs 
didn't converge
have 2461.920 507.874 1927.538
want 2461.920 507.874 1927.538
distance = 1.33051e-08
didn't converge
have 2461.920 0.000 1571.920
want 2461.920 0.000 1571.920
distance = 8.45835e-08
didn't converge
have 2461.920 0.000 1571.920
want 2461.920 0.000 1571.920
distance = 4.48967e-08
didn't converge
have 2418.080 0.000 1571.920
want 2418.080 0.000 1571.920
distance = 8.45834e-08
didn't converge
have 2418.080 0.000 1571.920
want 2418.080 0.000 1571.920
distance = 4.4897e-08
didn't converge
have 2418.080 507.874 1883.697
want 2418.080 507.874 1883.697
distance = 1.3305e-08
didn't converge
have 2418.080 507.874 1883.697
want 2418.080 507.874 1883.697
distance = 4.48968e-08
didn't converge
have 2418.080 0.000 1528.080
want 2418.080 0.000 1528.080
distance = 7.79714e-07
didn't converge
have 2461.920 -0.000 1528.080
want 2461.920 0.000 1528.080
distance = 4.81295e-08

### Additional information

For bugs, please attach a savefile that shows the problematic behavior.
You can attach `.slvs` files by archiving them into a `.zip` first.

[davits-tubing.zip](https://github.com/solvespace/solvespace/files/2893881/davits-tubing.zip)

whitequark, 2019-02-22:
What is your chord tolerance set to?

pmarches, 2019-02-22:
I am using the default setting of 0.5%. I tried increasing it to 50% but I still am unable to open the file.

whitequark, 2019-02-22:
OK. SolveSpace from the master branch can open this file with 0.5% chord tolerance, so I believe this is already fixed. Let me know if that doesn't work for you.

closed 2019-02-22 by whitequark

pmarches, 2019-02-22:
Thank you, the git version 9d1601eea9039011c16c435b26a680fb4bc35c34 worked for me. Thanks for the support!