solvespace/solvespace #388: Crash on close

starsolaris, 2019-03-20:
### System information

SolveSpace version: 3.0~83a0a517; (build from https://ci.appveyor.com/project/whitequark/solvespace/build/artifacts)

Operating system: Windows 10

### Expected behavior

SolveSpace should close

### Actual behavior

SolveSpace crashed with error message:
File C:\projects\solvespace\src\platform\guiwin.cpp, line 459, function SolveSpace::Platform::MenuBarImplWin32: Win32 API call failed: DestroyMenu(hMenuBar).
Error: Неверный дескриптор меню. 
("Invalid Menu Handle" in English)

whitequark, 2019-03-20:
Looks like:

> If the specified window is a parent or owner window, DestroyWindow automatically destroys the associated child or owned windows when it destroys the parent or owner window. The function first destroys child or owned windows, and then it destroys the parent or owner window.

So there's probably a sequencing issue between destroying the window and its menu bar.

whitequark, 2019-05-13:
Fixed in a460cf4.

closed 2019-05-13 by whitequark