solvespace/solvespace #390: Option 'Make faces selectable with mouse' doesn't stay disabled

philtipping, 2019-04-04:
### System information

SolveSpace version: 3.0~46528bfc

Operating system: lubuntu 18.04.2 LTS

### Expected behavior

'Make faces selectable with mouse' option should stay disabled until user enables it.
Note this is a presumption so may not be a bug.

### Actual behavior

Option enables itself when group is re-activated.

### Additional information

Recreate problem as follows:-
sketch a rectangle
extrude sketch
disable the option
click home
activate previous sketch-in-plane group - option remains disabled as expected
re-activate extrude group - option becomes enabled

whitequark, 2019-04-04:
That's deliberate:
https://github.com/solvespace/solvespace/blob/a7b2f2899944cd267767caa96c0da995b38bb203/src/group.cpp#L351-L360

philtipping, 2019-04-04:
Ok, if it's intentional, that's fine.
Not sure of the github protocol so do I close this or do you?
Many thanks for the quick response.

closed 2019-04-04 by whitequark