solvespace/solvespace #397: Not all solid edges exported for 2D VIEW if "Draw occluded lines"

Symbian9, 2019-04-25:
### System information

SolveSpace version: (e.g. 3.0~3dd2fc00; go to Help → About...)

Operating system: (e.g. Debian testing)

### Expected behavior

What should have happened?

### Actual behavior

What actually happened?

### Additional information

For bugs, please attach a savefile that shows the problematic behavior.
You can attach `.slvs` files by archiving them into a `.zip` first.

phkahler, 2020-09-08:
Can not reproduce. It looks like the silhouette edges are missing in the @Symbian9 image. Are they possibly there but drawn in white? (does changing the background color work?

BTW I tried all 3 variations of the occluded line toolbar button and they all worked for me. I did notice that the hover-text tells what it will switch to after you press it, now what it is currently set to. Doesn't matter because all 3 options produced the same edges in the .pdf as what I had on screen.

Can you attach a file that exhibits the problem?

phkahler, 2020-09-22:
Closing as not reproducible.

closed 2020-09-22 by phkahler