solvespace/solvespace #398: Cannot open solvespace.dmg on OSX

fredOnGitHub, 2019-04-30:
### System information : Mac Mojave 10.14.4

SolveSpace version:  solvespace.dmg  

Operating system: Mac

### Expected behavior

What should have happened?

### Actual behavior

Cannot open solvespace.dmg on OSX

### Additional information

whitequark, 2019-04-30:
SolveSpace is no longer supported on macOS as it got increasingly hostile to OpenGL applications.

thmrtnz, 2019-05-22:
It's quite bizarre.
I can run SolveSpace 2.3 in macOS 10.14.5 given I delete the packaged dylibs first.

fredOnGitHub, 2019-06-06:
> It's quite bizarre.
> I can run SolveSpace 2.3 in macOS 10.14.5 given I delete the packaged dylibs first.
> This from a freshly downloaded dmg from the releases section here, Github.

Hello,
Thank YOU for your helping;
Yes ! On macOS 10.14.5 i have deleted (in blue)
1. libfreetype.dylib
2. libpng.dylib
3. libz.dylib
from tha solvespace's package 

[image]
in the bind
[image]

and after leaving the package

and clicking on the application logo 
[image]

 it works

I have to work much with this version on Mac to see if it's ok

Thanks

tko, 2019-10-30:
From the looks of it, each of `libz`, `libpng`, and `libfreetype` is frozen to a (relatively old) vendored version. https://github.com/solvespace/solvespace/blob/5d78f993cea3749b6644e0a2f9ddf89c9754e377/CMakeLists.txt#L134-L137

As the vendored versions aren't working anymore, it might be better to remove the forced versioning, or update the versions. (Generally speaking, I'm not familiar with macos app distribution.)

vespakoen, 2019-11-28:
!Testers needed!

I have worked on Catalina support, and some mac related fixes.

I also made updated the build config so new dmg's can easily be built.

I have signed (and notarized) it with my developer certificate.

Would love to get feedback on possible installation problems.

https://github.com/vespakoen/solvespace/releases

Note that I tagged it 2.4 and that this not an official release.

whitequark, 2019-11-28:
Please do not make an one-sided choice of version numbers to distribute builds of. There will be no 2.x releases after 2.3, and the branch you have built is a development build from the 3.x series. Calling it something like 3.0.vespakoen would have been much better.

vespakoen, 2019-11-28:
Ok I'll change that.

I had to tag it with something to get the file out of travis, and that was the best I could come up with in that moment.

altaic, 2019-11-29:
> **! Testers needed !**
> 
> I have worked on Catalina support, and some macOS related fixes.
> 
> I also updated the CI config so new installers can easily be built and released in the future.
> 
> I have signed (and notarized) SolveSpace with my own personal developer certificate.
> 
> This should install without any problems on macOS and i'd like to verify if that's really the case.
> 
> https://github.com/vespakoen/solvespace/releases
> 
> Note that I tagged it ~2.4~ 3.0-vespakoen and that this is **NOT** an official release.
> 
> This PR adds the new CI config: #517
> 
> And the 3.0-vespakoen release is based on the latest commit on master ([0472b9e](https://github.com/solvespace/solvespace/commit/0472b9ea603003d59fd885d499ff176ca0bf7890))

Works for me on macOS 10.12.6!

On first opening (after copying it to `/Applications`), Solvespace did display an error "Couldn't read from file '/-psn_0_35709452'" though it has not reappeared since. I'm guessing it was due to detritus from my previous Solvespace install.

vespakoen, 2019-11-29:
Thanks for the report!
Interestingly I remember seeing a similar error popup in first load.

I didn't find psn anywhere in the source code though so I am not sure where it comes from.

Glad the dmg / app is working though

whitequark, 2019-11-29:
Could the psn thing be related to macOS opening on startup what it thinks the files that were opened the previous time?

vespakoen, 2019-11-29:
I don't think so, it looks like it could be loading a font or autosave
file, I tried to debug it just now but I cannot get it to trigger anymore.
will create a new user account and try it over there...

stiff, 2020-04-13:
Runs on 10.15, with /-psn complaints. Maybe it's related to this question https://stackoverflow.com/questions/10242115/os-x-strange-psn-command-line-parameter-when-launched-from-finder ?

Also for some reason locale doesn't match the system one, but this is minor issue.

gon, 2020-05-05:
+1 Thanks @vespakoen, great build indeed!

tthrift, 2020-05-05:
@vespakoen Thanks!  :)
The 3.0-vespakoen dmg seemed to work fine on macos 10.15.4 (19E287).
Upon first launch only, solvespace did show an error dialog with the message 
" Couldn't read from file '/-psn_0_204850' "
This dialog seems consistent with what others have seen.  
Otherwise, solvespace seems to be working well so far.

whitequark, 2020-05-10:
@stiff Thanks for investigating. I've opened #602.

vespakoen, 2020-10-19:
Good news!

I just installed the macOS Big Sur (build 10) beta and SolveSpace seems to be running just fine!

[image]

vespakoen, 2020-10-19:
I also made a new pre-release for macOS from the latest master branch which you can get over here: https://github.com/vespakoen/solvespace/releases/tag/3.0.0-alpha.4

Feedback & testing is appreciated!

ruevs, 2020-10-19:
I am curious what does it report here?
https://user-images.githubusercontent.com/6709544/90786031-5fce8b00-e303-11ea-8fe9-39b73a03e3c5.png

vespakoen, 2020-10-19:
This is what I got:

[image]

(It's on a MacBookPro14,1)

phkahler, 2020-10-19:
I'll try it on my old Macbook Air. I've never installed XCode or done any development on it other than Python, so my environment is not polluted ;-)

ruevs, 2020-10-19:
So is this Angle using Metal? Or is OpenGL still there in the  Big Sur beta? Is there a way to know? (Just curiosity on my part.)

vespakoen, 2020-10-19:
I am not sure, but I'll look into that and report back.

phkahler, 2020-10-19:
I'm on Sierra 10.12.6 and it seem to work OK. Is it built as debug or release? With OpenMP or LTO? It does seem slow but this mac is slow compared to my desktop anyway.

phkahler, 2020-10-19:
@ruevs Looks like OpenGL is still there:
https://developer.apple.com/forums/thread/650427

@vespakoen  can we build Universal Binaries to run on Apple silicon?

vespakoen, 2020-10-19:
It is a Debug build without OpenMP (I will try to enable that soon, I've already added it to OpenCamLib's CMake in the past)
And I will look into the Universal binaries as well.

phkahler, 2020-10-19:
The option is already there just add  -DENABLE_OPENMP=yes  and  -DENABLE_LTO=yes  to you cmake. LTO is probably a bad idea for debug builds that happen when we update a PR because they take forever.

vespakoen, 2020-10-26:
There is a new edge build for macOS which from now on will always be up to date with whats on the master branch.
OpenMP is also included in the macOS build now, and I would love to hear if it's still working for everyone.

https://github.com/solvespace/solvespace/releases/download/edge/SolveSpace.dmg

stiff, 2020-10-26:
Seems to be working good 👍

phkahler, 2020-10-29:
@vespakoen  I think it's finally time to close this issue. Thank you so much for all you've done!

tko, 2020-10-29:
Can confirm. edge release runs fine for me. (macos Mojave)

ruevs, 2020-10-29:
We should close it when 3.0 is released and downloadable from the official site.

ruevs, 2021-04-07:
@phkahler Added this to the 3.0 milestone so that we remember to close it and remove it from the pinned issues.

closed 2021-04-17 by phkahler