solvespace/solvespace #400: Feature Request: Use FileChooserNative

iicurtis, 2019-05-08:
### System information

SolveSpace version: 3.0~e7b75f19

Operating system: Arch Linux 5.0.13

### Expected behavior

It would be nice if Solvespace used the native file chooser dialog for other desktop environments (such as KDE). 

### Actual behavior

The default gtk dialog is used. The default gtk dialog leaves much to be desired, and is often painful to use. (i.e. by default, typing doesn't rename the file, but rather opens up a search dialog)

### Additional information

There is a fix for this: GTK has introduced [`Gtk::FileChooserNative`](https://developer.gnome.org/gtkmm/stable/classGtk_1_1FileChooserNative.html). Unfortunately, although it is supposed to be nearly a drop-in replacement, in this instance it is not. The Gtk team decided to make the default constructors protected. Already this conflicts with how `Gtk::FileChooserDialog` is being used. The alternative, `Gtk::FileChooserNative::create()` returns the world's worst smart pointer `Glib::RefPtr` which cannot be de-referenced.

Current state in src/platform/guigtk.cpp:

```c++
FileDialogRef CreateOpenFileDialog(WindowRef parentWindow) {
    Gtk::Window &gtkParent = std::static_pointer_cast<WindowImplGtk>(parentWindow)->gtkWindow;
    Gtk::FileChooserDialog gtkDialog(gtkParent, C_("title", "Open File"),
                                     Gtk::FILE_CHOOSER_ACTION_OPEN);
    gtkDialog.add_button(C_("button", "_Cancel"), Gtk::RESPONSE_CANCEL);
    gtkDialog.add_button(C_("button", "_Open"), Gtk::RESPONSE_OK);
    gtkDialog.set_default_response(Gtk::RESPONSE_OK);
    return std::make_shared<FileDialogImplGtk>(std::move(gtkDialog));

}
```

whitequark, 2019-05-08:
Thanks, I wasn't aware that FileChooserNative was a thing.

whitequark, 2019-05-08:
Well, I've implemented this. Unfortunately it is one of those FreeDesktop features that look sensible but do not actually work in practice and are impossible to debug. I have absolutely no idea how to get it to work, so now you're on your own.

<details>
<summary>patch</summary>

```patch
diff --git a/src/CMakeLists.txt b/src/CMakeLists.txt
index 874e88d..91a01c4 100644
--- a/src/CMakeLists.txt
+++ b/src/CMakeLists.txt
@@ -13,6 +13,14 @@ endif()
 
 set(HAVE_SPACEWARE ${SPACEWARE_FOUND})
 
+if(NOT WIN32 OR APPLE)
+    if(GTKMM_gtkmm-3.0_VERSION VERSION_LESS "3.24.0")
+        set(HAVE_GTK_FILECHOOSERNATIVE 0)
+    else()
+        set(HAVE_GTK_FILECHOOSERNATIVE 1)
+    endif()
+endif()
+
 configure_file(${CMAKE_CURRENT_SOURCE_DIR}/config.h.in
                ${CMAKE_CURRENT_BINARY_DIR}/config.h)
 
diff --git a/src/config.h.in b/src/config.h.in
index 2f210d7..475b973 100644
--- a/src/config.h.in
+++ b/src/config.h.in
@@ -16,4 +16,7 @@
 #cmakedefine HAVE_BACKTRACE
 #define BACKTRACE_HEADER @BACKTRACE_HEADER@
 
+/* If we use GTK, can we use the native file chooser? */
+#cmakedefine HAVE_GTK_FILECHOOSERNATIVE
+
 #endif
diff --git a/src/platform/guigtk.cpp b/src/platform/guigtk.cpp
index 8f57afd..d8aeb24 100644
--- a/src/platform/guigtk.cpp
+++ b/src/platform/guigtk.cpp
@@ -15,6 +15,10 @@
 #include <gtkmm/cssprovider.h>
 #include <gtkmm/entry.h>
 #include <gtkmm/filechooserdialog.h>
+#define HAVE_GTK_FILECHOOSERNATIVE
+#if defined(HAVE_GTK_FILECHOOSERNATIVE)
+#   include <gtkmm/filechoosernative.h>
+#endif
 #include <gtkmm/fixed.h>
 #include <gtkmm/glarea.h>
 #include <gtkmm/main.h>
@@ -1187,32 +1191,27 @@ MessageDialogRef CreateMessageDialog(WindowRef parentWindow) {
 // File dialogs
 //-----------------------------------------------------------------------------
 
-class FileDialogImplGtk final : public FileDialog {
+class FileDialogImplGtk : public FileDialog {
 public:
-    Gtk::FileChooserDialog      gtkDialog;
+    Gtk::FileChooser           *gtkChooser;
     std::vector<std::string>    extensions;
 
-    FileDialogImplGtk(Gtk::FileChooserDialog &&dialog)
-        : gtkDialog(std::move(dialog))
-    {
-        gtkDialog.property_filter().signal_changed().
+    void InitFileChooser(Gtk::FileChooser &chooser) {
+        gtkChooser = &chooser;
+        gtkChooser->property_filter().signal_changed().
[... 159 lines cut ...]
-    Gtk::FileChooserDialog gtkDialog(gtkParent, C_("title", "Open File"),
-                                     Gtk::FILE_CHOOSER_ACTION_OPEN);
-    gtkDialog.add_button(C_("button", "_Cancel"), Gtk::RESPONSE_CANCEL);
-    gtkDialog.add_button(C_("button", "_Open"), Gtk::RESPONSE_OK);
-    gtkDialog.set_default_response(Gtk::RESPONSE_OK);
-    return std::make_shared<FileDialogImplGtk>(std::move(gtkDialog));
-
+    return std::make_shared<FILE_DIALOG_IMPL>(gtkParent, /*isSave=*/false);
 }
 
 FileDialogRef CreateSaveFileDialog(WindowRef parentWindow) {
     Gtk::Window &gtkParent = std::static_pointer_cast<WindowImplGtk>(parentWindow)->gtkWindow;
-    Gtk::FileChooserDialog gtkDialog(gtkParent, C_("title", "Save File"),
-                                     Gtk::FILE_CHOOSER_ACTION_SAVE);
-    gtkDialog.set_do_overwrite_confirmation(true);
-    gtkDialog.add_button(C_("button", "_Cancel"), Gtk::RESPONSE_CANCEL);
-    gtkDialog.add_button(C_("button", "_Save"), Gtk::RESPONSE_OK);
-    gtkDialog.set_default_response(Gtk::RESPONSE_OK);
-    return std::make_shared<FileDialogImplGtk>(std::move(gtkDialog));
+    return std::make_shared<FILE_DIALOG_IMPL>(gtkParent, /*isSave=*/true);
 }
 
 //-----------------------------------------------------------------------------
```
</details>

iicurtis, 2019-05-08:
Huh, well your patch looks pretty similar to mine. Mine segfaults but it does bring up the native dialog.

I think this is going to change at some point, but until then, you need to have the environment variable `GTK_USE_PORTAL=1`

I'm running it with `GTK_USE_PORTAL=1 ./bin/solvespace`

whitequark, 2019-05-08:
OK, it works with that environment variable set.

whitequark, 2019-05-08:
Fixed in bc3e09e.

closed 2019-05-08 by whitequark

whitequark, 2019-05-11:
> The alternative, `Gtk::FileChooserNative::create()` returns the world's worst smart pointer `Glib::RefPtr` which cannot be de-referenced.

I just found out that gtkmm 4 dispenses with that and uses std::shared_ptr. https://www.murrayc.com/permalink/2017/04/20/gtkmm-4-progress/