solvespace/solvespace #405: Regression: Hard to drag constraint labels

rpavlik, 2019-05-15:
### System information

SolveSpace version:  3.0~201e15e6

Operating system: Debian testing/buster

### Expected behavior
When I hover over a newly-added constraint label (like a distance or an angle) and it turns color, I should be able to click and drag to move. (And could, in 2.3)

### Actual behavior

I typically can't, at least not without zooming in or out. I think this is worse when the constraint happens to be near the origin, but not 100% sure. It's been this way on master for some time - since my earlier PR.

### Additional information
Here's the file I used in the screencast below - add an angle constraint to the tip of the slot sketch group. Sometimes it works, sometimes it acts up.

[sample.zip](https://github.com/solvespace/solvespace/files/3183960/sample.zip)

Screencast showing two attempts: the first one, I was able to zoom a bit and get it to drag eventually. The second clip, I couldn't get it to drag no matter what I did.

[slvs trouble dragging constraints.zip](https://github.com/solvespace/solvespace/files/3183971/slvs.trouble.dragging.constraints.zip)

whitequark, 2019-05-15:
Is this a duplicate of #161 ?

rpavlik, 2019-05-15:
Hmm. Possibly - wonder if the fact that I can't move the origin is why clicking and dragging on a near-origin constraint causes failure. I'll pay attention and see if I can figure it out better.

whitequark, 2019-05-24:
Actually, I think it's a duplicate of #174.

rpavlik, 2019-05-24:
ah, yeah, that matches better, Sorry for missing the dupe.

closed 2019-05-24 by whitequark