solvespace/solvespace #406: Regression: Strange tool tips

rpavlik, 2019-05-15:
### System information

SolveSpace version:  3.0~201e15e6

Operating system: Debian testing/buster

### Expected behavior

I don't remember what the tooltips were in 2.3, but I know they probably weren't random tool descriptions.

### Actual behavior

Hovering something that might produce a tooltip in a sketch will display some other tool's tooltip. Often, the same "other" tooltip text will be used for an extended period of time. Unfortunately I can't figure out how the text is chosen or when exactly it changes.

### Additional information

Screencast of me showing it happen, resizing the winodow to bring the tools into view, showing it happen again (now the text is different), then using a tool (which doesn't apparently change the text)
[slvs weird tooltips.zip](https://github.com/solvespace/solvespace/files/3183986/slvs.weird.tooltips.zip)

whitequark, 2019-05-15:
Is this a duplicate of #365 ?

rpavlik, 2019-05-15:
Nope, the tooltips for the actual tools look fine.  Just fired up 2.3 on the same system and realized that the places that are triggering tooltips in the screencast didn't actually trigger tooltips in the past, just a color change from hover.

I'm using a standard gnome-shell on xorg config with not much special configuration-wise.

rpavlik, 2019-05-15:
Turns out that you don't actually even need to hover anything - just "hovering"  the empty blackness of the background for a few moments is enough to trigger a tooltip. I presumably just don't pause long enough with my mouse not on an entity to notice this.

whitequark, 2019-05-15:
Yeah, I switched to using native tooltips and this happened. It seems to only be the case with old GTK versions, which one do you have?

rpavlik, 2019-05-15:
3.24.5 as packaged by debian in buster.

```
≻ apt-cache policy libgtk-3-0
libgtk-3-0:
  Installed: 3.24.5-1
  Candidate: 3.24.5-1
  Version table:
 *** 3.24.5-1 500
        500 http://httpredir.debian.org/debian buster/main amd64 Packages
        100 /var/lib/dpkg/status
```

whitequark, 2019-05-15:
That's... bizarre, as I have the exact same version. I'm going to have to find some workaround, I guess.

rpavlik, 2019-05-15:
Are you on Wayland vs xorg perhaps?

whitequark, 2019-05-15:
I'm on xorg as well

whitequark, 2019-05-23:
Hopefully fixed in 3296474c.

rpavlik, 2019-05-23:
Yep, fixed. Thanks!

closed 2019-05-23 by rpavlik