solvespace/solvespace #412: Property Browser window stays on top after switching focus

nabijaczleweli, 2019-05-18:
### System information

SolveSpace version: 3.0~62aba398

Operating system: Windows 10

### Expected behavior

The Property Browser window	is obscured alongside the sketch window after changing focus to a different window.

### Actual behavior

The Property Browser window stays on top of whatever new window was selected. Clicking on any part of the window focuses it and brings back the sketch window.

### Additional information

Works as expected on 2.3.

[Screenshot](//user-images.githubusercontent.com/6709544/57976067-227c9e80-79d8-11e9-9025-b5efac359667.png)

whitequark, 2019-05-20:
I believe the cause is somewhere along these lines:
https://github.com/solvespace/solvespace/blob/f43954cc29059600023b2fbb2d0a25776ae7edd1/src/platform/guiwin.cpp#L541-L559

The relevant code in 2.3 is here:
https://github.com/solvespace/solvespace/blob/b1d87bf284b32e875c8edba592113e691ea10bcd/src/win32/w32main.cpp#L1253-L1319

There's a fair bit of difference between them. Do you think you could try playing with the `WS_*` bitfields and see if it helps? There's a bunch of different cases to be handled here with regards to focus handling, and evidently I missed some. Note that some of the `WS_*` values include the other, like `WS_OVERLAPPEDWINDOW`.

Here is the fairly confusing [Microsoft doc](https://docs.microsoft.com/en-us/windows/desktop/winmsg/window-styles) for window styles.

nabijaczleweli, 2019-05-20:
I managed to arrive at the correct behaviour by applying the following:

```diff
diff --git a/src/platform/guiwin.cpp b/src/platform/guiwin.cpp
index 73da2da..ec6bfca 100644
--- a/src/platform/guiwin.cpp
+++ b/src/platform/guiwin.cpp
@@ -545,7 +545,7 @@ public:
                 break;

             case Window::Kind::TOOL:
-                style |= WS_POPUPWINDOW|WS_CAPTION;
+                style |= (/*WS_POPUP | */WS_BORDER | WS_SYSMENU)|WS_CAPTION|WS_CHILD;
                 break;
         }
         sscheck(hWindow = CreateWindowExW(0, L"SolveSpace", L"", style,
```

However, the window now looks ~~like absolute garbage~~ pretty bad:

[image]

Unfortunately, I haven't managed to make it look platable nor as-was in 2.3 (not for lack of slamming my head against a proverbial wall) with any combination of more or less sensible options.

Also, a few questions that popped up while I was making this:
  1. Do you consider warnings bugs? Because this file has two, so dunno what I should do about them, and
  2. Scrolling in the Property Browser selects the face that's below it (didn't happen in 2.3), is this intended?

whitequark, 2019-05-20:
> However, the window now looks ~like absolute garbage~ pretty bad

Have you tried this? I think the property browser isn't intended to have a sysmenu, and that might make it look better:

```
style |= WS_BORDER|WS_CAPTION|WS_CHILD;
```

> and, perhaps more importantly, appears permanently disabled

That's actually intended, or at least that was the 2.x behavior, IIRC. This should have the effect of forwarding key bindings to the parent window.

Can you try disabling the following code and seeing if key bindings (e.g. `d`) are still forwarded?

https://github.com/solvespace/solvespace/blob/f43954cc29059600023b2fbb2d0a25776ae7edd1/src/platform/guiwin.cpp#L945-L952

> Scrolling in the Property Browser selects the face that's below it (didn't happen in 2.3), is this intended?

Here is the part of the code that deals with scroll events:

https://github.com/solvespace/solvespace/blob/f43954cc29059600023b2fbb2d0a25776ae7edd1/src/platform/guiwin.cpp#L863-L878

I'm not entirely sure from your description what went wrong here.

> Do you consider warnings bugs? Because this file has two, so dunno what I should do about them

Depends. Some warnings are useful, some are not, and compilers constantly add new warnings. You could post them here if you wanted.

nabijaczleweli, 2019-05-20:
The `WS_BORDER|WS_CAPTION|WS_CHILD` combo does look marginally better:
[image]

In both 2.3 as well as unpatched `master`, the Property Browser window can be focused, which provides a visual indicator, that it's interactable:
[image]

Commenting out the `else` block has no effect on `d` being forwarded in both `WS_POPUPWINDOW|WS_CAPTION` and `WS_BORDER|WS_CAPTION|WS_CHILD` window styles.

Another thing I saw in taking those screenshots is that in the `WS_CHILD`-containing variant the Property Browser window is constrained to the sketch window and cannot extend beyond it:
[image]

[Here](https://github.com/solvespace/solvespace/files/3199964/2019-05-20.23-44-19.zip)'s a video illustrating the issue: the scrolling code works fine, but the as the mouse is over the Property Browser window and scrolling, if there's any selectable face in the sketch window at the mouse's position, it will be highlighted, until the mouse is moved.

<details>
<summary>And here's the warnings</summary>

```
[1/2] Building CXX object src/CMakeFiles/solvespace.dir/platform/guiwin.cpp.obj
../src/platform/guiwin.cpp: In function 'UINT SolveSpace::Platform::ssGetDpiForWindow(HWND)':
../src/platform/guiwin.cpp:70:78: warning: cast between incompatible function types from 'FARPROC' {aka 'long long int (*)()'} to 'SolveSpace::Platform::LPFNGETDPIFORWINDOW' {aka 'unsigned int (*)(HWND__*)'} [-Wcast-function-type]
             GetProcAddress(GetModuleHandleW(L"user32.dll"), "GetDpiForWindow");
                                                                              ^
../src/platform/guiwin.cpp: In function 'BOOL SolveSpace::Platform::ssAdjustWindowRectExForDpi(LPRECT, DWORD, BOOL, DWORD, UINT)':
../src/platform/guiwin.cpp:93:87: warning: cast between incompatible function types from 'FARPROC' {aka 'long long int (*)()'} to 'SolveSpace::Platform::LPFNADJUSTWINDOWRECTEXFORDPI' {aka 'int (*)(tagRECT*, long unsigned int, int, long unsigned int, unsigned int)'} [-Wcast-function-type]
             GetProcAddress(GetModuleHandleW(L"user32.dll"), "AdjustWindowRectExForDpi");
```

</details>

whitequark, 2019-05-20:
I fixed the warning in be0dc7e2c.

whitequark, 2019-05-20:
Now that I look closer at the window staying on top, well, there's code that does that:

https://github.com/solvespace/solvespace/blob/be0dc7e2cb43d8105a6b5d5565719d678f50700b/src/platform/guiwin.cpp#L562-L565

What if you comment it out and rely on parent-child relationship?

nabijaczleweli, 2019-05-20:
That did it, the window abides as expected!

whitequark, 2019-05-20:
As a Windows user, which set of `WS_*` flags do you think SolveSpace should use?

whitequark, 2019-05-20:
> [Here](https://github.com/solvespace/solvespace/files/3199964/2019-05-20.23-44-19.zip)'s a video illustrating the issue: the scrolling code works fine, but the as the mouse is over the Property [...]

Regarding this bug, I think I know what causes it, as well as why I can't reproduce it. Can you open another issue so it doesn't get lost?

nabijaczleweli, 2019-05-20:
`WS_POPUPWINDOW|WS_CAPTION` are actually perfect, now that it doesn't linger!

whitequark, 2019-05-20:
To recheck, do I understand it correctly that without that code, the property browser window is still on top of the main window at all times, but now it is no longer on top of other windows when the main window is not in focus?

nabijaczleweli, 2019-05-20:
Yes

whitequark, 2019-05-20:
Fixed in 5dbe090.

closed 2019-05-20 by whitequark

nabijaczleweli, 2019-05-20:
Superb, thank you!