solvespace/solvespace #414: clang-format config?

rpavlik, 2019-05-20:
In working on the code, I tend to accidentally not follow the style guide because it's a little unique (mostly the "no spaces in `if(condition)`, etc" item)  If this style is still the desired one and important, would be great if someone could make a clang-format (or other formatter) config file that enforced it.

whitequark, 2019-05-20:
I generally agree, but I'm not sure how clang-format deals with partial reformats. As been discussed before, bulk formatting changes are a no-go, and if clang-format can only do one file at a time its usefulness seems limited here.

rpavlik, 2019-05-20:
I tend to use git clang-format to just reformat the changes currently staged, for projects where I can't format the whole thing.

whitequark, 2019-05-20:
Whoa, I had no idea that existed. I'm completely in favor of having a `.clang-format` file then and suggesting its use in guidelines. I'm not sure when I will get around it, so if you feel like making one, go ahead please.

closed 2019-05-23 by whitequark