solvespace/solvespace #415: Image Export generates black image

nabijaczleweli, 2019-05-20:
### System information

SolveSpace version: 3.0~62aba398

Operating system: Windows 10

### Expected behavior

The File -> Export Image… menu item, after confirming location, exports the visible geometry.

### Actual behavior

The output file has the correct size, but contains only black.

### Additional information

Works on 2.3.

Doesn't depend on the contents of the sketch.

whitequark, 2019-05-20:
The bug is likely related to this function:
https://github.com/solvespace/solvespace/blob/c18deb2d811deaf40ede9a3771e1753192990f96/src/render/rendergl3.cpp#L681-L687

On Windows, ANGLE is used, which might have some issue with `glReadPixels`. That said, I'm pretty sure I tested it before and it worked...

nabijaczleweli, 2019-05-20:
Following the lead from [this SO answer](https://stackoverflow.com/a/45700864/2851815), I applied the following patch:

```diff
diff --git a/src/render/rendergl3.cpp b/src/render/rendergl3.cpp
index 167ce03..cc21130 100644
--- a/src/render/rendergl3.cpp
+++ b/src/render/rendergl3.cpp
@@ -5,6 +5,7 @@
 //-----------------------------------------------------------------------------
 #include "solvespace.h"
 #include "gl3shader.h"
+#include <fstream>

 namespace SolveSpace {

@@ -679,10 +680,25 @@ void OpenGl3Renderer::Clear() {
 }

 std::shared_ptr<Pixmap> OpenGl3Renderer::ReadFrame() {
+    std::ofstream err("t:/solvespace-err.log");
+    err << std::hex;
+
     std::shared_ptr<Pixmap> pixmap =
         Pixmap::Create(Pixmap::Format::RGB, (size_t)camera.width, (size_t)camera.height);
+
+    err << "pregl\n";
+    for(GLenum e; (e = glGetError()) != GL_NO_ERROR;) {
+        err << e << '\n';
+    }
+
     glReadPixels(0, 0, (int)camera.width, (int)camera.height,
                  GL_RGB, GL_UNSIGNED_BYTE, &pixmap->data[0]);
+
+    err << "postgl\n";
+    for(GLenum e; (e = glGetError()) != GL_NO_ERROR;) {
+        err << e << '\n';
+    }
+
     return pixmap;
 }
```

And, after Exporting Image, the contents of `t:/solvespace-err.log` are

```
pregl
postgl
502
```

Which [this](https://www.khronos.org/opengl/wiki/OpenGL_Error#Meaning_of_errors) says is `GL_INVALID_OPERATION`.

Sadly, I don't really do GL et al. so I don't know where to follow with this.

whitequark, 2019-05-20:
From the [OpenGL ES 2 doc](http://docs.gl/es2/glReadPixels):
```
GL_INVALID_OPERATION is generated if type is GL_UNSIGNED_SHORT_5_6_5 and format is not GL_RGB.

GL_INVALID_OPERATION is generated if type is GL_UNSIGNED_SHORT_4_4_4_4 or GL_UNSIGNED_SHORT_5_5_5_1 and format is not GL_RGBA.

GL_INVALID_OPERATION is generated if format and type are neither GL_RGBA and GL_UNSIGNED_BYTE, respectively, nor the format/type pair returned by querying GL_IMPLEMENTATION_COLOR_READ_FORMAT and GL_IMPLEMENTATION_COLOR_READ_TYPE.
```

whitequark, 2019-05-20:
I believe this is fixed in 11c5cdc7b0. Can you check?

nabijaczleweli, 2019-05-20:
And so it is! Thank you!

[image]

closed 2019-05-20 by nabijaczleweli

whitequark, 2019-05-20:
Very nice!