solvespace/solvespace #417: Face under mouse highlighted when scrolling Property Browser window

nabijaczleweli, 2019-05-20:
### System information

SolveSpace version: 3.0~be0dc7e2

Operating system: Windows 10

### Expected behavior

A face that's hidden under the Property Browser window doesn't get highlighted.

### Actual behavior

Faces covered by Property Browser window get highlighted if the mouse is over them while scrolling that window.

### Additional information

[Video](https://github.com/solvespace/solvespace/files/3199964/2019-05-20.23-44-19.zip)

whitequark, 2019-05-20:
So what I think is happening here is that events are forwarded to the property browser window correctly (AFAICT its window in your video is not in focus), but then are also handled by the main window as well. This is handled here:

https://github.com/solvespace/solvespace/blob/be0dc7e2cb43d8105a6b5d5565719d678f50700b/src/platform/guiwin.cpp#L869-L884

nabijaczleweli, 2019-05-20:
Can confirm, this doesn't happen if the Property Browser window is focused.

whitequark, 2019-05-20:
What if you apply this?
```patch
diff --git a/src/platform/guiwin.cpp b/src/platform/guiwin.cpp
index 0366880..a39e351 100644
--- a/src/platform/guiwin.cpp
+++ b/src/platform/guiwin.cpp
@@ -822,6 +822,7 @@ public:
                 event.shiftDown   = (wParam & MK_SHIFT) != 0;
                 event.controlDown = (wParam & MK_CONTROL) != 0;
 
+                bool consumed = false;
                 switch(msg) {
                     case WM_LBUTTONDOWN:
                         event.button = MouseEvent::Button::LEFT;
@@ -872,6 +873,7 @@ public:
                         sscheck(hWindowUnderMouse = WindowFromPoint(pt));
                         if(hWindowUnderMouse && hWindowUnderMouse != h) {
                             SendMessageW(hWindowUnderMouse, msg, wParam, lParam);
+                            consumed = true;
                             break;
                         }
 
@@ -903,7 +905,7 @@ public:
                     }
                 }
 
-                if(window->onMouseEvent) {
+                if(!consumed && window->onMouseEvent) {
                     window->onMouseEvent(event);
                 }
                 break;
```

nabijaczleweli, 2019-05-20:
That did it!

whitequark, 2019-05-20:
Fixed in 7f75148.

closed 2019-05-20 by whitequark