solvespace/solvespace #421: FPS counter shows LONG_MIN if renderTime.count() is 0

nabijaczleweli, 2019-05-21:
### System information

SolveSpace version: 3.0~39c34809

Operating system: Windows 10

### Expected behavior

The counter checks if the count is zero and prevents casting an `inf` to a `long`.

### Actual behavior

The FPS counter overflows `long` argument and displays it.

### Additional information

[image]

whitequark, 2019-05-21:
What about this?
```patch
diff --git a/src/draw.cpp b/src/draw.cpp
index 3f13b5d..b9ddcb4 100644
--- a/src/draw.cpp
+++ b/src/draw.cpp
@@ -888,7 +888,7 @@ void GraphicsWindow::Paint() {
 
     // Also display an fps counter.
     RgbaColor renderTimeColor;
-    if(1000 / renderTime.count() < 60) {
+    if(renderTime.count() > 16.67) {
         // We aim for a steady 60fps; draw the counter in red when we're slower.
         renderTimeColor = { 255, 0, 0, 255 };
     } else {
@@ -896,7 +896,7 @@ void GraphicsWindow::Paint() {
     }
     uiCanvas.DrawBitmapText(ssprintf("rendered in %ld ms (%ld 1/s)",
                                      (long)renderTime.count(),
-                                     (long)(1000/renderTime.count())),
+                                     (long)(1000 / std::max(0.1, renderTime.count()))),
                             5, 5, renderTimeColor);
 
     canvas->FlushFrame();
```

nabijaczleweli, 2019-05-21:
Yep, that does it!

[image]

whitequark, 2019-05-21:
Fixed in 9500487a3.

closed 2019-05-21 by whitequark