solvespace/solvespace #422: Add point on line segment

Harvie, 2019-05-22:
When i set constraint for point to stay on line, it can move past the two points defining the line like this:

[image]

I think it would be usefull to have constraint that can limit point movement only between these two points, eg.:

[image]

Evil-Spirit, 2019-05-23:
This will be able with ranged constraints merged. PointOnLine has param and it can be constrained to be 0..1.

phkahler, 2019-06-20:
Can this be applied to point-on-spline then?

whitequark, 2019-07-17:
@phkahler I think the problem is the discontiguity between two parts of a spline.

Evil-Spirit, 2019-07-17:
There is no problem to make pt-on-spline if spline is split on cubic bezier segments, but in case of SolveSpace where we have one primitive for several splines (up to 15 pts) we should invent some special derivative-function made of segments. My suggestion is to remove the current spline entity and substitute it with
1) Cubic spline made of 4 pts
2) Tangent of second order derivative
So, we can support old files in the way where we substitute one big spline with small ones and applying special (G2/C2 derivatives smooth) tangency after loading.

phkahler, 2019-07-17:
I was only thinking of the cubic spline with 4 points. I had forgotten that those are created with the same tool as the many-point splines and may be considered "the same thing" when they're actually sort-of different. I know the more complex version is actually composed of many instances of the simple version internally, but it still begs the question of how to transition a constrained point from one segment to the next. It wouldn't seem right to allow it for one curve and not another when they are created with the same tool.

I'd recommend doing point-on-line-segment first. Then the code will be in place to limit the range of parameters (Revolve angle may want that too).

Evil-Spirit, 2019-07-19:
@phkahler  it is already implemented and needs some UI/cosmetic fixes https://github.com/Evil-Spirit/solvespace-master/commit/7cb2b8fb4afd8fbd4bfeb5cc1a6ad7c9f4c7075a

Thovthe, 2022-04-15:
3.2 milestone? Is this behaviour toggleable?

ruevs, 2022-04-15:
Maybe 4.0 together with #77 .

liuxilu, 2023-06-01:
Workaround: A chord is always <= the diameter.

Harvie, 2024-09-12:
Hello, any news on this?

ruevs, 2024-09-12:
Went there are "news" they will probably come from #77.

Probably by merging and improving some form of https://github.com/Evil-Spirit/solvespace-master/commit/7cb2b8fb4afd8fbd4bfeb5cc1a6ad7c9f4c7075a

https://github.com/Evil-Spirit/solvespace-master/commit/7cb2b8fb4afd8fbd4bfeb5cc1a6ad7c9f4c7075a