solvespace/solvespace #426: Feature - New Group: Spiral (flat spiral)

Symbian9, 2019-05-23:
### Expected behavior

Add "Spiral" tool for creating flat spirals:

- spiral of Cornu;
- hairspring (balance spring);
- etc.

It could be realized in two ways:

1. as "Sketch" tool: 1) draw spiral in plane; 2) then extrude it (cons - spirals profile always would rectangular)
2. as "New Group" tool: 1) draw profile; 2) create/rotate spiral.

Second variant is preferred.

Or as option such as "Helix" for "Lathe" tool: https://github.com/solvespace/solvespace/issues/389

### Additional information

[image]

whitequark, 2019-05-23:
I don't think SolveSpace should gain a generic planar spiral generator. Make a spiral elsewhere and export a DXF, then extrude; that would be far more efficient.

closed 2019-05-23 by whitequark

Evil-Spirit, 2019-06-12:
NoteCAD have "function" entity to create such curves like this
catenary a
[image]
rch
cornu spiral based on fresnel integrals
[image]

Symbian9, 2019-06-12:
> NoteCAD have "function"

Would be happy to see this in SolveSpace too.

Evil-Spirit, 2019-06-13:
> Would be happy to see this in SolveSpace too.

SolveSpace's solver has born to has freature like this.

Symbian9, 2019-06-13:
> SolveSpace's solver has born to has feature like this.

But would be it possible draw Euler spirals and extrude along Euler spiral in SolveSpace? (spiros, clothoids, or Cornu spirals)

Evil-Spirit, 2019-06-13:
I don't know. If anyone will implement it. NURBS approximation of such functions can be rather complicated task. NURBS actually not an advantage for SolveSpace at the moment. Too many people want to help but can't because of buggy (from one side) and too complicated (from the other side) of NURBS backend. I think SolveSpace can't beat commercial software, so not necessary to try do that. Instead, SolveSpace can utilize some proven OpenSource geometric libraries like CGAL which provide slow but really robust booleans based on Nef Polyhedra algorithms. So, for realtime geometric computations, SolveSpace can use fast BSP and for export - slow but good quality CGAL. If this step will be performed, any of  geometric functions like extrusion along path will be much simplier in implementation and probability of happening of such implemetation will increase.

Evil-Spirit, 2019-06-13:
For the record. I have implemented helical revolve for NoteCAD in 2 evernings. That's because I have no any NURBS.

whitequark, 2019-06-13:
@Evil-Spirit If there is no NURBS, there is no STEP export that machine shops can take. That would make SolveSpace mostly useless for me. Maybe I could still use it for 3D printing (not that I do a lot of it), but I will need another CAD for any serious machining work, so why bother maintaining it?

Evil-Spirit, 2019-06-14:
@whitequark I understand.

stefano2734, 2024-02-07:
See Levien PHD for better bezier curves also
Inkscape spiro curves are from Levien.

https://levien.com/spiro/
https://levien.com/phd/thesis.pdf

ruevs, 2024-02-08:
> See Levien PHD for better bezier curves also Inkscape spiro curves are from Levien. https://levien.com/spiro/ https://levien.com/phd/thesis.pdf see page 152 in pdf (135 in text) for optimized bezier curves.

This would be relevant is one day we decide to implement spirals or curves based on "any fuction". My detailed response is here https://github.com/solvespace/solvespace/issues/1447#issuecomment-1934077135